bluez/bluez #998: LE Audio unstable with Intel AX210 controller and SoundPeats Air4 Pro earbuds

ilyats, 2024-10-29:
Trying to use LE Audio with Intel AX210 controller and SoundPeats Air4 Pro earbuds, the recording does not work at all, and the playback is very unstable, with constant disappearance of sound either in both earbuds or only in the right one, disconnects and reconnects, and crashes in bluetoothd. Here is output of sample run of bluetoothd built from latest git sources. 
[bluetoothd.log](https://github.com/user-attachments/files/17552597/bluetoothd.log)
I have previously [reported](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4368) this problem to pipewire people

Vudentz, 2024-10-29:
@ilyats not a crash:

```
bluetoothd[609898]: Terminating
bluetoothd[609898]: src/adapter.c:adapter_shutdown() 
bluetoothd[609898]: src/adapter.c:adapter_set_power_state() on-disabling
```

This is the handling of SIGINT/SIGTERM so something is killing the bluetoothd process. Also it looks like the connection is timing out:

`bluetoothd[609898]: src/device.c:att_disconnected_cb() Connection timed out (110)`

Anyway perhaps this is due to the setting pipewire has choosen, try collecting the HCI trace with btmon so we can have a better look of what is going on.

ilyats, 2024-11-01:
@Vudentz I used to have crashes in bluez before, I couldn't reproduce them recently. I did a new attempt with latest bluez and pipewire/wireplumber git sources, there were sound disappearances and several disconnects and reconnects. The result of `btmon -w btmon.dump` is 
[BTMON.DMP](https://github.com/user-attachments/files/17596803/BTMON.DMP)
and output of `bluetoothd -n -d -f ...` is 
[bluetooth.log](https://github.com/user-attachments/files/17596810/bluetooth.log)

ilyats, 2024-11-04:
[More tries](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4368#note_2642053)

ilyats, 2024-11-23:
New[ test results](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4368#note_2671837)

ilyats, 2024-12-23:
I was testing it again with latest git sources and having disabled some high-quality modes, as pipewore people suggested. At one point I got input and no output, input seemed to work OK. Then, apparently when I tried to disconnect and reconnect device hoping to get output, I got a crash in bluetoothd. 
[pipewire.log](https://github.com/user-attachments/files/18230512/pipewire.log)
hen I tried to disconnect and reconnect device, I got a crash in bluetoothd. 
[bluetoothd.log](https://github.com/user-attachments/files/18230497/bluetoothd.log)
[BTMON.LOG](https://github.com/user-attachments/files/18230498/BTMON.LOG)
[pipewire.log](https://github.com/user-attachments/files/18230532/pipewire.log)
[wireplumber.log](https://github.com/user-attachments/files/18230526/wireplumber.log)

ilyats, 2025-01-13:
The buds manufacturer has made a firmware update, now it does not seem to have le audio at all
```Device 98:80:BB:40:FB:F4 (public)
	Name: SOUNDPEATS Air4 Pro
	Alias: SOUNDPEATS Air4 Pro
	Paired: yes
	Bonded: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00001100-d102-11e1-9b23-00025b00a5a5)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (0000eb10-d102-11e1-9b23-00025b00a5a5)
	UUID: Vendor specific           (0000eb16-d102-11e1-9b23-00025b00a5a5)
	UUID: Qualcomm Technologies I.. (0000fd92-0000-1000-8000-00805f9b34fb)
	RSSI: 0xffffffbe (-66)
	AdvertisingFlags:
  1a                                               .               
	Battery Percentage: 0x00 (0)
```

closed 2025-07-17 by github-actions[bot]