bluez/bluez #1002: MOONDROP Ultrasonic TWS LE Audio doesn't work

shdwchn10, 2024-11-03:
I can connect to both LE earbuds successfully (in GNOME Settings and bluetoothctl), but audio device doesn't appear in pipewire. I reported about this issue [in the pipewire repo first](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4348) but I was redirected to the bluez repo, because device fails before PipeWire is involved.

- PipeWire version (`pipewire --version`): pipewire-1.2.5-1.20241014git34bca4a.fc41.x86_64 (built from master)
- Distribution and distribution version (`PRETTY_NAME` from `/etc/os-release`): Fedora Linux 41.20241013.0 (Silverblue Prerelease)
- Desktop Environment: GNOME 47
- Kernel version (`uname -r`): 6.11.3-300.fc41.x86_64
- BlueZ version (`bluetoothctl --version`): bluez-20241014.bd7d49d-1.fc41.x86_64 (built from master)
- `lsusb`: Bus 001 Device 003: ID 8087:0032 Intel Corp. AX210 Bluetooth
- Bluetooth devices:

```
❯ bluetoothctl devices
Device XX:XX:XX:XX:XX:83 Moondrop-LEA
Device XX:XX:XX:XX:XX:64 Moondrop-LEA
Device XX:XX:XX:XX:XX:D6 MOONDROP Ultrasonic # Non-LE
```

pw-dump.log, bluetoothd.log, pipewire-bluez.log and dump.btsnoop: https://drive.proton.me/urls/E506KG8TYR#lmLRJqmf8T4v

Vudentz, 2024-11-04:
Seems to be due to malformed GATT database:

```
Oct 14 05:51:44 ga1ahad bluetoothd[1855]: [0x5d53e50f0a10] src/shared/gatt-client.c:discover_incl_cb() handle: 0x0085, start: 0x0071, end: 0x0080,uuid: 00001843-0000-1000-8000-00805f9b34fb
...
Oct 14 05:51:44 ga1ahad bluetoothd[1855]: [0x5d53e50f0a10] src/shared/gatt-client.c:init_complete() Failed to initialize gatt-client

```

This is quite strange as it means there are no ranges to discover but that only happens if we fails at some point but there is no logging of that, well except if the code you are running is not really the latest?

shdwchn10, 2024-11-04:
I built bluez from bd7d49d commit (this was latest commit when issue was reported in pipewire repo). I will try to build new packages from master (664a5d3) now

everything411, 2025-01-01:
exactly the same problem on MOONDROP Robin's Earphones, Ubuntu 24.04, Intel AX210

everything411, 2025-01-04:
By adding more logs in bluez src, I found that bluez failed in a `bt_gatt_discover_characteristics` call at https://github.com/bluez/bluez/blob/dfb1ffdc95a00bc06d81a75c11ab5ad2e24d37bf/src/shared/gatt-client.c#L682C26-L682C58 , where, in the `discover_chrcs_cb` callback, `success` is `false` and `att_ecode` is `0`. The `bt_gatt_discover_characteristics` call is complex and I failed to find out exactly in which function it failed.
@Vudentz I'm not familiar with bluez and bluetooth specifications. Any suggestions on further debugging?

daniel-fs, 2025-02-25:
Has there been any update on this issue? Same symptoms with Moondrop Robin's Earphones and also the Intel AX210 card. Bluez version 5.79, Pipewire 1.2.7, kernel 6.13. The Robin's connect in LE mode to my computer, but no audio device appears to actually have audio output.

I can provide any logs that might be helpful to solve this, just let me know which and how to get them.

ValeZAA, 2025-03-28:
Here on Windows 11 initial pairing works (LE on toggle in settings), but then it keep connecting and disconnecting. Probably a bug in Intel Killer / BE200.

closed 2025-07-17 by github-actions[bot]

everything411, 2026-10-06:
I've confirmed that the real reason for this issue is #2224