bluez/bluez #1003: AVDTP reconfiguration fails on Samsung HW-C450 soundbar when setting codec to anything other than default ( default is AAC, even switch to SBC fails)

Leopard1907, 2024-11-03:
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4375

Issue was first reported here, after gathering bt traffic dumps Pipewire devs redirected me here.

[trafficdump.txt](https://github.com/user-attachments/files/17612134/trafficdump.txt)

While reporting that issue, i learned some BT chips are fake and causes issues so i checked an mine is a genuine one.

https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4375#note_2635572

When this said device is connected to an Android phone, it can successfully switch between codecs AAC and SBC without fail.

Vudentz, 2024-11-04:
Is this perhaps fixed by: https://patchwork.kernel.org/project/bluetooth/list/?series=905703?

pv, 2024-11-04:
That series was for a different issue -- I don't think there is a race in this case.

The btmon dump looks like:
<details>

```
bluetoothd[650]: < ACL Data TX: Hand.. flags 0x00 dlen 7  #811 [hci0] 17.221884
      Channel: 256 len 3 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Suspend (0x09) Command (0x00) type 0x00 label 0 nosp 0
        ACP SEID: 1
> ACL Data RX: Handle 75 flags 0x02 dlen 6                #813 [hci0] 17.249797
      Channel: 64 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Suspend (0x09) Response Accept (0x02) type 0x00 label 0 nosp 0
bluetoothd[650]: < ACL Data TX: Hand.. flags 0x00 dlen 7  #814 [hci0] 17.250505
      Channel: 256 len 3 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Close (0x08) Command (0x00) type 0x00 label 1 nosp 0
        ACP SEID: 1
> ACL Data RX: Handle 75 flags 0x02 dlen 6                #816 [hci0] 17.266450
      Channel: 64 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Close (0x08) Response Accept (0x02) type 0x00 label 1 nosp 0
< ACL Data TX: Handle 75 flags 0x00 dlen 12               #817 [hci0] 17.266476
      L2CAP: Disconnection Request (0x06) ident 14 len 4
        Destination CID: 257
        Source CID: 66
< ACL Data TX: Handle 75 flags 0x00 dlen 12               #818 [hci0] 17.266525
      L2CAP: Disconnection Request (0x06) ident 15 len 4
        Destination CID: 259
        Source CID: 68
> ACL Data RX: Handle 75 flags 0x02 dlen 12               #821 [hci0] 17.293124
      L2CAP: Disconnection Response (0x07) ident 14 len 4
        Destination CID: 257
        Source CID: 66
> ACL Data RX: Handle 75 flags 0x02 dlen 12               #822 [hci0] 17.293125
      L2CAP: Disconnection Request (0x06) ident 206 len 4
        Destination CID: 64
        Source CID: 256
> ACL Data RX: Handle 75 flags 0x02 dlen 12               #823 [hci0] 17.293126
      L2CAP: Disconnection Response (0x07) ident 15 len 4
        Destination CID: 259
        Source CID: 68
< ACL Data TX: Handle 75 flags 0x00 dlen 12               #824 [hci0] 17.293147
      L2CAP: Disconnection Response (0x07) ident 206 len 4
        Destination CID: 64
        Source CID: 256
> ACL Data RX: Handle 75 flags 0x02 dlen 12               #826 [hci0] 17.313122
      L2CAP: Disconnection Request (0x06) ident 207 len 4
        Destination CID: 65
        Source CID: 255
< ACL Data TX: Handle 75 flags 0x00 dlen 12               #827 [hci0] 17.313133
      L2CAP: Disconnection Response (0x07) ident 207 len 4
        Destination CID: 65
        Source CID: 255
> HCI Event: Disconnect Complete (0x05) plen 4            #829 [hci0] 17.475420
        Status: Success (0x00)
        Handle: 75 Address: 68:52:10:28:F9:43 (MCS Logic)
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8     {0x0001} [hci0] 17.475427
        BR/EDR Address: 68:52:10:28:F9:43 (MCS Logic)
        Reason: Connection terminated by remote host (0x03)
```

</details>
So it seems the device accepts our AVDTP Suspend and  AVDTP Close, but then closes L2CAP channels and disconnects.

Not sure if there's something we can do here. Android is probably doing something different when you change the codec.

Vudentz, 2024-11-04:
> That series was for a different issue -- I don't think there is a race in this case.
> 
> The btmon dump looks like:
> 
> So it seems the device accepts our AVDTP Suspend and AVDTP Close, but then closes L2CAP channels and disconnects, instead of going IDLE.
> 
> Not sure if there's something we can do here. Android is probably doing something different when you change the codec.

Weird, well I guess we need the logs from Android then to attempt to do something similar. Btw, has there been any reports of SBC not working lately, I was just trying with my XM4 but it looks like something is up with SBC as there is no audio, Ive checked the volume is fine and both SBC and SBC-dual presets don't seem to produce any audio but AAC works fine, but there is no configuration errors or anything of sorts, it is just muted for some reason.

Leopard1907, 2024-11-04:
I retrieved btsnoop_hci.log from Android but it is an unreadable mess when opened in kate at least. Does it need some sort edid decode like thing to be readable?

pv, 2024-11-04:
>  Btw, has there been any reports of SBC not working lately,

No, first I hear of that --- if you mean Pipewire, I don't think there have been any substantial A2DP-related changes in a long time.

Leopard1907, 2024-11-04:
> Weird, well I guess we need the logs from Android then to attempt to do something similar. Btw, has there been any reports of SBC not working lately, I was just trying with my XM4 but it looks like [...]

Here are the logs from Android.

[btsnoop_hci.log](https://github.com/user-attachments/files/17625003/btsnoop_hci.log)

[txt form.txt](https://github.com/user-attachments/files/17625019/txt.form.txt)

Txt form one is : Opened via wireshark, exported as txt.

Vudentz, 2024-11-05:
> Here are the logs from Android.
>
> [btsnoop_hci.log](https://github.com/user-attachments/files/17625003/btsnoop_hci.log)
>
> [txt [...]

Looks like the only different is that we do Suspend + Close vs Android does Close directly, that said our SetConfiguration seems to do Close directly so I assume the Suspend is actually Pipewire releasing the transport before attempting to reconfiguring, which imo is the right thing to do, but perhaps the triggers a 'clean' disconnect logic on the remote end causing this problem.

pv, 2024-11-05:
In the Android `btsnoop_hci.log` I don't see AVDTP Start, so that there's no Suspend makes sense. Probably audio was not playing.

@Leopard1907 can you get a log with codec changed while audio is playing?

Leopard1907, 2024-11-05:
Yes i can in a moment, i thought not playing anything would be equal because when switch failed on Linux i was not playing anything either.

pv, 2024-11-05:
Another difference is that Android sends SetConfiguration right away after CLOSE_RSP, whereas BlueZ waits first. You can try building bluez from sources (you can run bluetoothd from build directory), with change
```diff
diff --git a/profiles/audio/a2dp.c b/profiles/audio/a2dp.c
index 37332abca..4eb31602f 100644
--- a/profiles/audio/a2dp.c
+++ b/profiles/audio/a2dp.c
@@ -1398,7 +1398,7 @@ static bool setup_reconfigure(struct a2dp_setup *setup)
 
        DBG("%p", setup);
 
-       setup->id = g_timeout_add(RECONFIGURE_TIMEOUT, a2dp_reconfigure, setup);
+       setup->id = g_idle_add(a2dp_reconfigure, setup);
 
        setup->reconfigure = FALSE;
 
```

Leopard1907, 2024-11-05:
New logs:

Switches from AAC to SBC multiple times while media playing.

[btsnoop_hci.log](https://github.com/user-attachments/files/17637365/btsnoop_hci.log)
[new log.txt](https://github.com/user-attachments/files/17637366/new.log.txt)

pv, 2024-11-05:
Android appears to do there Suspend -> Suspend_Rsp -> Close -> Close_Rsp -> L2CAP disconnect + SetConfiguration, so seems like same sequence as in the BlueZ dump above. Don't have other ideas than the timing difference above...

Leopard1907, 2024-11-05:
> Another difference is that Android sends SetConfiguration right away after CLOSE_RSP, whereas BlueZ waits first (and in the above dumps the device disconnects before bluez gets to it). You can try building bluez from sources (you can run bluetoothd from build directory), with change
> 
> ```diff
> diff --git a/profiles/audio/a2dp.c b/profiles/audio/a2dp.c
> index 37332abca..4eb31602f 100644
> --- a/profiles/audio/a2dp.c
> +++ b/profiles/audio/a2dp.c
> @@ -1398,7 +1398,7 @@ static bool setup_reconfigure(struct a2dp_setup *setup)
>  
>         DBG("%p", setup);
>  
> -       setup->id = g_timeout_add(RECONFIGURE_TIMEOUT, a2dp_reconfigure, setup);
> +       setup->id = g_idle_add(a2dp_reconfigure, setup);
>  
>         setup->reconfigure = FALSE;
>  
> ```

So things i have to do are:

`git clone git://git.kernel.org/pub/scm/bluetooth/bluez.git`

`cd bluez`

`apply diff`

`./bootstrap-configure`

`make`

And then run `bluetoothd` ?

pv, 2024-11-05:
You need to stop the systme bluetoothd instance first, `systemct stop bluetooth.service`. For me systemd wants to start it after I stop it, so you may need to also do `systemctl mask bluetooth.service` (and unmask after you've done testing).

Leopard1907, 2024-11-05:
Thanks for the pointers, juggling with those was required indeed.

So please check if my a2dp.c file is correct:

[a2dp.tar.gz](https://github.com/user-attachments/files/17638136/a2dp.tar.gz)

If yes, it is correct; then unfortunately it still fails:

Ran as `sudo ./src/bluetoothd -n -d -f ./src/main.conf`:

[trying.txt](https://github.com/user-attachments/files/17638143/trying.txt)

Vudentz, 2024-11-05:
> Thanks for the pointers, juggling with those was required indeed.
>
> So please check if my a2dp.c file is correct:
>
> [a2dp.tar.gz](https://github.com/user-attachments/files/17638136/a2dp.tar.gz)
>
> If [...]

Perhaps it is because we disconnect AVRCP in the process:

```
bluetoothd[1012750]: src/service.c:change_state() 0x5333350: device 94:DB:56:F7:F2:88 profile a2dp-sink state changed: connected -> disconnected (0)
bluetoothd[1012750]: plugins/policy.c:policy_disconnect() /org/bluez/hci0/dev_94_DB_56_F7_F2_88 profile avrcp-controller
bluetoothd[1012750]: src/service.c:change_state() 0x53337c0: device 94:DB:56:F7:F2:88 profile avrcp-controller state changed: connected -> disconnecting (0)
bluetoothd[1012750]: profiles/audio/avrcp.c:avrcp_disconnect() path /org/bluez/hci0/dev_94_DB_56_F7_F2_88
bluetoothd[1012750]: src/service.c:change_state() 0x5334170: device 94:DB:56:F7:F2:88 profile audio-avrcp-target state changed: connected -> disconnected (0)
bluetoothd[1012750]: src/service.c:change_state() 0x53337c0: device 94:DB:56:F7:F2:88 profile avrcp-controller state changed: disconnecting -> disconnected (0)
```

Perhaps we need a new state to indicate to the likes of sink.c/source.c that the service is not to be considered disconnected.

Vudentz, 2024-11-05:
@Leopard1907 try with the following changes:

https://gist.github.com/Vudentz/4cf33625753a4bc8805b8f8b7f1af5cd

Leopard1907, 2024-11-05:
> @Leopard1907 try with the following changes:
> 
> https://gist.github.com/Vudentz/4cf33625753a4bc8805b8f8b7f1af5cd

Should i get rid of earlier changes in a2dp or keep them still?

pv, 2024-11-05:
Discard the other a2dp.c changes, if they didn't help they're not needed.

Leopard1907, 2024-11-05:
With latest changes it seems it was succesful.

I was able to switch between aac, sbc and sbc-xq without fail. Also tried it when media was playing and it still worked.

[succesful i think.txt](https://github.com/user-attachments/files/17638913/succesful.i.think.txt)