bluez/bluez #1007: avrcp remote volume control failing with release 5.79

borine, 2024-11-07:
In commit 8baabd2f5, AVRCP EVENT_VOLUME_CHANGED is excluded from the target session supported_events when the remote device has SetAbsoluteVolume capability. This is contrary to the Bluetooth SIG AVRCP specification v1.5.1, which states in Section 4.5 that EVENT_VOLUME_CHANGED is mandatory for a TG with Category 2 feature support.

I suspect this commit is the reason my iphone is no longer able to remotely control the volume of my Bluez audio sink device when using Bluez 5.79. Modifying `target_init()` in `profiles/audio/avrcp.c` such that EVENT_VOLUME_CHANGED is always included in the session supported_events restores remote volume function to my iphone.

borine, 2024-11-08:
Reading the source code again, this actually just looks like a simple typo or copy&paste error. The following patch fixes it:
```diff
diff --git a/profiles/audio/avrcp.c b/profiles/audio/avrcp.c
index b5e6548e6..9fe8f55e5 100644
--- a/profiles/audio/avrcp.c
+++ b/profiles/audio/avrcp.c
@@ -4268,7 +4268,7 @@ static void target_init(struct avrcp *session)
        if (target->version < 0x0104)
                return;
 
-       if (!avrcp_volume_supported(target))
+       if (avrcp_volume_supported(target))
                session->supported_events |=
                                (1 << AVRCP_EVENT_VOLUME_CHANGED);
```

borine, 2024-11-08:
> do you care to send a proper fix to the mailing list?

Sorry, I am unable to submit to the mailing list; but hopefully the original author(s) of commit 8baabd2 will be able/willing to submit a correction?

closed 2024-11-08 by BluezTestBot

Vudentz, 2024-12-02:
> That is unfortunate to hear and see. I was not made aware that [my patch was applied when I sent it](https://lore.kernel.org/linux-bluetooth/20241005214321.84250-3-marijn.suijten@somainline.org/), nor that it was completely modified to no longer look anywhere close to what I originally sent.
> 
> @Vudentz when you do such a thing, please at least indicate so in the commit message or on the mailing list that it's no longer authentic to what the submitter sent.
> 
> It also looks like half of the change was committed/squashed into a different commit: [ba98be0](https://github.com/bluez/bluez/commit/ba98be0356215dd3fdfb0cd40d7bf4cde4d7359a)

Ive had consolidated some of the checks as part of avrcp_volume_supported, but apparently I messed something up and left part of the changes behind which was only pushed in the above change, anyway this has been fixed and we can't really rework the commits at this point so Im afraid the only real lesson is that we need more testing.

MarijnS95, 2024-12-02:
Thanks. The second lesson or "request" I wanted to get across is that I would have been totally okay to rework the patches and send a `v5` with it. Or otherwise have correct attribution to the change, i.e. some trailers like:

```

[Luiz: consolidate some checks into a new avrcp_volume_supported() function]

```

As is standard in the kernel :grin:. Nothing to be done about it now, but I'd appreciate something like that in the future :pray:

alex-robbins, 2025-02-28:
Would the maintainers please consider cutting a release that includes this fix? 

(As a user of a stack that includes bluez, it was frustrating to spend time debugging this, only to find that it was a known issue, for which a fix was committed 3 months ago but never released.)

Vudentz, 2025-02-28:
> Would the maintainers please consider cutting a release that includes this fix?
> 
> (As a user of a stack that includes bluez, it was frustrating to spend time debugging this, only to find that it was a known issue, for which a fix was committed 3 months ago but never released.)

Yeah, we are on it, should be out soon.

alex-robbins, 2025-03-15:
I almost don't want to leave this comment because I don't want to bother you any more than I already have, but I wanted to pop back in to say that I'm actually *not* personally affected by this bug, and I misunderstood it completely.

Re-reading the original comment, it seems that this bug is about AVRCP remote volume control when using bluez as an audio *sink*. My casual skimming of the patches and commits referred to in this thread (along with the title) lead me to believe that AVRCP volume control was broken in general, e.g. for bluetooth headphones connected to a host running bluez 1.79. (That was the kind of breakage I was seeing, and it's due to a bug in the way my specific distro uses wireplumber. It has nothing to do with bluez at all.)

So, sorry for the noise. If someone reads this who was confused in the same way that I was, then at least this comment will have been worth it. (After all, this thread was the top Google result for "bluez avrcp hardware volume control", which is how I ended up here while trying to diagnose my problem.)

Lastly, as long as I've got you: Thanks for bluez. For me, it has (as it turns out) always been rock solid.