bluez/bluez #1008: How to change Phy_1M, 2M and Coded programmatically in python bluez

premanandaembed, 2024-11-08:
1. I'm getting Local name, advertisement type and data but not getting any idea how to set Phy_1M and 2M and coded 
2. Need help on increase Tx power using same how to achieve?

premanandaembed, 2024-11-12:
there is no reply or no help i want some help on this please help me

xihua13104, 2025-02-17:
[https://github.com/weliem/bluez_inc](url)
Here is an open source project that may be able to help you. Although it is not implemented in Python, you can refer to it.
To enable Phy_1M, 2M and Coded, here is the patch:
```c
---
 binc/advertisement.c | 23 +++++++++++++++++++++++
 binc/advertisement.h | 11 +++++++++++
 2 files changed, 34 insertions(+)

diff --git a/binc/advertisement.c b/binc/advertisement.c
index 6a6959f..c3c9a98 100644
--- a/binc/advertisement.c
+++ b/binc/advertisement.c
@@ -39,6 +39,13 @@ struct binc_advertisement {
     guint32 max_interval;
     guint16 appearance;
     gboolean general_discoverable;
+    SecondaryChannel secondary_channel;
+};
+
+static char *secondary_channel_str[BINC_SC_MAX] = {
+    "1M",
+    "2M",
+    "Coded"
 };
 
 static void add_manufacturer_data(gpointer key, gpointer value, gpointer userdata) {
@@ -108,6 +115,8 @@ GVariant *advertisement_get_property(GDBusConnection *connection,
         ret = g_variant_new_uint16(advertisement->appearance);
     } else if (g_str_equal(property_name, "Discoverable")) {
         ret = g_variant_new_boolean(advertisement->general_discoverable);
+    } else if (g_str_equal(property_name, "SecondaryChannel")) {
+        ret = g_variant_new_string(secondary_channel_str[advertisement->secondary_channel]);
     }
     return ret;
 }
@@ -145,6 +154,7 @@ void binc_advertisement_register(Advertisement *advertisement, const Adapter *ad
             "       <property name='MaxInterval' type='u' access='read'/>"
             "       <property name='Appearance' type='q' access='read'/>"
             "       <property name='Discoverable' type='b' access='read'/>"
+            "       <property name='SecondaryChannel' type='s' access='read'/>"
             "   </interface>"
             "</node>";
 
@@ -308,3 +318,16 @@ void binc_advertisement_set_general_discoverable(Advertisement *advertisement, g
     advertisement->general_discoverable = general_discoverable;
 }
 
+void binc_advertisement_set_secondary_channel(Advertisement *advertisement, SecondaryChannel secondary_channel) {
+    g_assert(advertisement != NULL);
+    g_assert(secondary_channel < BINC_SC_MAX);
+
+    advertisement->secondary_channel = secondary_channel;
+}
+
+SecondaryChannel binc_advertisement_get_secondary_channel(Advertisement *advertisement) {
+    g_assert(advertisement != NULL);
+
+    return advertisement->secondary_channel;
+}
+
diff --git a/binc/advertisement.h b/binc/advertisement.h
index d71248b..ffc80ae 100644
--- a/binc/advertisement.h
+++ b/binc/advertisement.h
@@ -31,6 +31,13 @@
 extern "C" {
 #endif
 
+typedef enum {
+    BINC_SC_1M = 0, //default
+    BINC_SC_2M,
+    BINC_SC_CODED,
+    BINC_SC_MAX
+} SecondaryChannel;
+
 Advertisement *binc_advertisement_create(void);
 
 void binc_advertisement_free(Advertisement *advertisement);
@@ -59,6 +66,10 @@ guint16 binc_advertisement_get_appearance(Advertisement *advertisement);
 
 void binc_advertisement_set_general_discoverable(Advertisement *advertisement, gboolean general_discoverable);
 
+void binc_advertisement_set_secondary_channel(Advertisement *advertisement, SecondaryChannel secondary_channel);
+
+SecondaryChannel binc_advertisement_get_secondary_channel(Advertisement *advertisement);
+
 #ifdef __cplusplus
 }
 #endif
-- 
2.42.0
```

closed 2025-07-17 by github-actions[bot]