bluez/bluez #1013: Bluez 5.78 input issue Switch Pro controller

Azyrion, 2024-11-10:
Hello everybody, as the title says I have issues with my switch controllers over Bluetooth. I have the issue where my input just doesn't come through for a period of time (3-6 sec) before suddenly having some of the inputs I tried during that time coming through seemingly all at once. Sometimes it also seems like it disconnects for fraction of time and is keeping the last input the controller had at the time. But there is no notification/sound and the "Gamecontroller" page in system settings (KDE) doesn't show it disappearing though. But it shows when input are "stuck" or just not getting through at all.
Playing through usb is fine but that's not my preferred way to play.

Tried it with 3 different controllers: 1 original Switch Pro controller and 2 3rd party that can be connected as switch pro controller but also as Xbox Wireless Controllers. It's only the when they are in switch mode that the issue appears in the 3rd party controllers. Connecting as Xbox controllers shows no problems.

OS: Nobara 40
Kernel: 6.11.7
Bluez: 5.78
CPU: Ryzen 7 5800X3D
RAM: 32GB
MoBo: MSI X570 Tomahawk Wifi
GPU: RX 6900 XT

Azyrion, 2024-11-15:
[Bluetooth.txt](https://github.com/user-attachments/files/17782328/Bluetooth.txt)
tried to capture traffic on wireshark no idea if it helps....if not could someone tell me what data would help and how to log them?

Vudentz, 2024-11-19:
@Azyrion try using btmon -p debug -w <filename>

Azyrion, 2024-11-20:
> @Azyrion try using btmon -p debug -w

thanks a lot!
hopefully you guys can see smth in this:
[hcidump.log](https://github.com/user-attachments/files/17826728/hcidump.log)

Vudentz, 2024-11-20:
> > @Azyrion try using btmon -p debug -w
> 
> thanks a lot! hopefully you guys can see smth in this: [hcidump.log](https://github.com/user-attachments/files/17826728/hcidump.log)

Don't seem to have any valuable information, the only thing I see is:

```
> ACL Data RX: Handle 256 flags 0x02 dlen 54                                                                            #5604 [hci0] 88.823267
      Channel: 65 len 50 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 30 7b 80 00 00 00 00 f8 7f 00 f8 7f 0c 4b ff  .0{...........K.
        76 00 7b 10 0b 00 44 00 fd ff 4d ff 76 00 74 10  v.{...D...M.v.t.
        0b 00 43 00 fd ff 54 ff 77 00 71 10 0a 00 44 00  ..C...T.w.q...D.
        fc ff
```

This is probably a connection that wasn't capture in the logs, did you actually run btmon while attempting to connect?

Azyrion, 2024-11-20:
> Don't seem to have any valuable information, the only thing I see is:
>
> ```
>       Channel: 65 len 50 [PSM 0 mode Basic (0x00)] {chan 65535}
>         a1 30 7b 80 00 00 00 00 f8 7f 00 f8 7f 0c 4b ff [...]

No the controller was already connected but I did try again making sure of it:
[hcidump.log](https://github.com/user-attachments/files/17834119/hcidump.log)

Azyrion, 2024-11-20:
Manage to catch a really bad case were the issue went on for nearly a minute or so. And it does seem to be a connection issue imo. The parentheses appeared after the controller briefly disappeared from the menu:
[image]
Here the new log
[hcidump3.log](https://github.com/user-attachments/files/17837532/hcidump3.log)

Azyrion, 2024-11-22:
> > Manage to catch a really bad case were the issue went on for nearly a minute or so. And it does seem to be a connection issue imo. The parentheses appeared after the controller briefly disappeared from the menu: ![kcm_gamecontroller issue2](https://private-user-images.githubusercontent.com/29234550/388286027-68dca862-9490-4936-b1ca-7d4a4f8154d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxNDI5ODAsIm5iZiI6MTczMjE0MjY4MCwicGF0aCI6Ii8yOTIzNDU1MC8zODgyODYwMjctNjhkY2E4NjItOTQ5MC00OTM2LWIxY2EtN2Q0YTRmODE1NGQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIwVDIyNDQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2Yzg2NzRjODFhNjk3NWFjZjRmY2M1NzA0ZmNmNTM0YTQ0MTAwZjIwODI2ZWQ0YTI4ZWQwODI1ZjMwZTYyZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UIE_iWtg-UaGJ0jaxoQ0DrnXUM_PgaJWDcGaBC2NuYE) Here the new log [hcidump3.log](https://github.com/user-attachments/files/17837532/hcidump3.log)
> 
> Totally useless again, if you don't start capturing the logs _before_ you connect it won't be able to decode the traffic.

Sorry read this as if I shouldn't have started loggin before connecting. This time I started the log before I connected my controller:
[hcidump.log](https://github.com/user-attachments/files/17856661/hcidump.log)

Azyrion, 2024-11-29:
Is the data I provided still not useful?

Azyrion, 2025-02-01:
Is this still being looked at or not at all?