bluez/bluez #1014: BT Pair Fails

maheshtalewad, 2024-11-11:
[image]
[dmesg_fail_5.txt](https://github.com/user-attachments/files/17697873/dmesg_fail_5.txt)
[fail_btmon.txt](https://github.com/user-attachments/files/17697883/fail_btmon.txt)

maheshtalewad, 2024-11-11:
Below commit is causing this issue:
https://patches.linaro.org/project/linux-bluetooth/patch/20240923165116.1540227-1-luiz.dentz@gmail.com/
+       if (conn && hci_dev_test_flag(hdev, HCI_MGMT))
+               mgmt_device_connected(hdev, conn, NULL, 0)

Vudentz, 2024-11-11:
@maheshtalewad please check if https://patchwork.kernel.org/project/bluetooth/patch/20241108162608.262188-1-luiz.dentz@gmail.com/ fixes it.

maheshtalewad, 2024-11-12:
> @maheshtalewad please check if https://patchwork.kernel.org/project/bluetooth/patch/20241108162608.262188-1-luiz.dentz@gmail.com/ fixes it.

Hi Luiz,
This patch fixed the issue.

Thanks and regards,
Mahesh Vithal Talewad

closed 2024-11-18 by Vudentz

joakim-tjernlund, 2024-11-21:
Should this fix be in kernel 6.6.x too? It is not ATM