bluez/bluez #1016: Service Class in main.conf always defaults to 0x6c0000 -- and (seems to) register unneeded UUIDS

abqjln, 2024-11-11:
I am attempting to eliminate unwanted UUIDS apparently registered automatically resulting from the default Service Class of 0x6c0000, using RPis 3,4,5 using v5.78.

Is there any way to eliminate the default Service Class 0x6c0000 or at least stop it from registering the associated UUIDs? Thanks.

**********
My application should have Class = 0x800900 (Service=Info, Major=Health, Minor=Misc), but bluez always installs 0x6c0000 (Service Classes=Rendering, Capturing, Audio, Telephony). Wireshark shows failed attempts to use the auto-installed UUIDS associated with the latter.

bluetoothctl show
        Class: 0x006c0000 (7077888)
...
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
        UUID: Audio Input Control       (00001843-0000-1000-8000-00805f9b34fb)
        UUID: Volume Offset Control     (00001845-0000-1000-8000-00805f9b34fb)
        UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
...
the client/server then attempt to use these UUIDS, resulting in unnecessary traffic as it will always fail

If I change main.conf
Class = 0x000000 (comment says this is default)
sudo systemctl daemon-reload && sudo systemctl restart bluetooth
I get the same 0x6c0000

If I change main.conf
Class = 0x800900 (Info, health, undefined)
It changes to
Class = 0x6c0900 (it picks up the Major=Health, Minor=undefined but keeps the Service Class 6c0, which I think triggers the associated UUIDS. The comment in main.conf does say it only major and minor class bits are considered. But why the default 0x6c0xxx?

I have also tried eliminating the hostname plugin but don't see any difference.

/lib/systemd/system/bluetooth.service
    ExecStart=/usr/libexec/bluetooth/bluetoothd -P battery --noplugin=hostname

Thanks.

Vudentz, 2024-11-11:
> If I change main.conf Class = 0x800900 (Info, health, undefined) It changes to Class = 0x6c0900 (it picks up the Major=Health, Minor=undefined but keeps the Service Class 6c0, which I think triggers [...]

These extra services might be the result of PulseAudio/Pipewire registering audio Endpoint for these UUIDS, in order to avoid that you can disable their plugins e.g -P bap, a2dp, avrcp, hfp, etc.

abqjln, 2024-11-11:
Thanks, you were correct. Purging PulseAudio and Pipewire (not using audio) eliminated the 0c6cxxx masking, and eliminated these UUIDs

UUID: Handsfree Audio Gateway (0000111f-0000-1000-8000-00805f9b34fb)
UUID: Handsfree (0000111e-0000-1000-8000-00805f9b34fb)
UUID: Audio Source (0000110a-0000-1000-8000-00805f9b34fb)
UUID: Audio Sink (0000110b-0000-1000-8000-00805f9b34fb)

But these remain. Does bluez have any way of identifying who is registering these other UUIDs? (You kinda just knew the above answer) Is there a list of plugins that are being installed?

UUID: A/V Remote Control (0000110e-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
UUID: Broadcast Audio Scan (0000184f-0000-1000-8000-00805f9b34fb)
UUID: Volume Control (00001844-0000-1000-8000-00805f9b34fb)
UUID: Audio Input Control (00001843-0000-1000-8000-00805f9b34fb)
UUID: Volume Offset Control (00001845-0000-1000-8000-00805f9b34fb)
UUID: Microphone Control (0000184d-0000-1000-8000-00805f9b34fb)

closed 2024-12-21 by abqjln