bluez/bluez #1017: handsfree

HinaHyugaHime, 2024-11-12:
```❯ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Tue 2024-11-12 00:41:12 CST; 44min ago
 Invocation: 4102e8fe6f184379ba222cf57ead0dc8
       Docs: man:bluetoothd(8)
   Main PID: 24865 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38271)
     Memory: 1.1M (peak: 1.9M)
        CPU: 103ms
     CGroup: /system.slice/bluetooth.service
             └─24865 /usr/lib/bluetooth/bluetoothd

Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/faststream
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/faststream_duplex
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/opus_05
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/opus_05
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Nov 12 00:41:12 arch bluetoothd[24865]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Nov 12 00:41:43 arch bluetoothd[24865]: profiles/audio/avdtp.c:cancel_request() Start: Operation canceled (125)
Nov 12 00:41:43 arch bluetoothd[24865]: profiles/audio/a2dp.c:a2dp_suspend() SEP in bad state for suspend
Nov 12 00:55:22 arch bluetoothd[24865]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connecte
d (107)
```
also get a tunnel sounding audio

Vudentz, 2024-11-13:
> ```
> ● bluetooth.service - Bluetooth service
>      Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
>      Active: active (running) since Tue 2024-11-12 00:41:12 [...]

What version?

HinaHyugaHime, 2024-11-13:
5.71-2

pv, 2024-11-23:
I think this is the issue fixed by a14d59725ef35d91b8504770046c701d0a2efe07 so should be fixed in next release

closed 2025-07-17 by github-actions[bot]