bluez/bluez #1019: handsfree audio output doesn't work with Bose headphones after upgrade to bluez 5.79

rptaylor, 2024-11-13:
Versions 5.78 and earlier were working fine:
```
2024-09-11T20:19:02-0700 SUBDEBUG Upgrade: bluez-5.78-1.fc40.x86_64
2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-libs-5.78-1.fc40.x86_64
2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-obexd-5.78-1.fc40.x86_64
2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-cups-5.78-1.fc40.x86_64
```

Yesterday I upgraded to the new version:
```
2024-11-12T09:36:03-0800 SUBDEBUG Upgrade: bluez-5.79-1.fc40.x86_64
2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-libs-5.79-1.fc40.x86_64
2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-obexd-5.79-1.fc40.x86_64
2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-cups-5.79-1.fc40.x86_64
```

The headphones pair and regular headset mode works for playing music, but in the handsfree mode for video conferences there is no audio output.

The only error messages I see are:
```
Nov 13 12:34:33 desktop wireplumber[6490]: spa.bluez5.native: RFCOMM receive command but modem not available: AT+BTRH?
Nov 13 12:34:34 desktop wireplumber[6490]: spa.bluez5.native: RFCOMM receive command but modem not available: AT+XEVENT=Bose SoundLink,158
```

Headset is Bose QuietComfort 35 Series 2.

Vudentz, 2024-11-14:
> Versions 5.78 and earlier were working fine:
> 
> ```
> 2024-09-11T20:19:02-0700 SUBDEBUG Upgrade: bluez-5.78-1.fc40.x86_64
> 2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-libs-5.78-1.fc40.x86_64
> 2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-obexd-5.78-1.fc40.x86_64
> 2024-09-11T20:19:03-0700 SUBDEBUG Upgrade: bluez-cups-5.78-1.fc40.x86_64
> ```
> 
> Yesterday I upgraded to the new version:
> 
> ```
> 2024-11-12T09:36:03-0800 SUBDEBUG Upgrade: bluez-5.79-1.fc40.x86_64
> 2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-libs-5.79-1.fc40.x86_64
> 2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-obexd-5.79-1.fc40.x86_64
> 2024-11-12T09:36:04-0800 SUBDEBUG Upgrade: bluez-cups-5.79-1.fc40.x86_64
> ```
> 
> The headphones pair and regular headset mode works for playing music, but in the handsfree mode for video conferences there is no audio output.
> 
> The only error messages I see are:
> 
> ```
> Nov 13 12:34:33 desktop wireplumber[6490]: spa.bluez5.native: RFCOMM receive command but modem not available: AT+BTRH?
> Nov 13 12:34:34 desktop wireplumber[6490]: spa.bluez5.native: RFCOMM receive command but modem not available: AT+XEVENT=Bose SoundLink,158
> ```
> 
> Headset is Bose QuietComfort 35 Series 2. There is a workaround I had to apply in /etc/bluetooth/main.conf which had been required for it to work with these headphones so far:
> 
> ```
> ControllerMode = bredr
> ```

Hmm, it might be worth checking with pipewire to see if there is something changed there, anyway could you please add the HCI trace using btmon -p debug -w <filename>?

rptaylor, 2024-11-15:
Thanks for looking @Vudentz !
I checked pipewire* and wireplumber* packages; they did not change recently. 
I emailed the file to the address in your github profile.
Here is what I did:

1. Start btmon
2. Turn on and peer headphones
3. Start a test video call with Google Meet
4. Try to play a test sound in the speakers, it did not work. (But apparently other people could hear the audio from my mic when I tried this yesterday).
5. Disconnect from the Google Meet call.   At this point strangely the btmon output suddenly scrolled by extremely quickly, rapidly producing lots of output in only a few seconds.
6. Turn off headphones
7. Stop btmon

Vudentz, 2024-11-18:
> Thanks for looking @Vudentz ! I checked pipewire* and wireplumber* packages installed on my system; they did not change recently. I emailed the file to the address in your github profile. Here is what I did:
> 
>     1. Start btmon
> 
>     2. Turn on and peer headphones
> 
>     3. Start a test video call with Google Meet
> 
>     4. Try to play a test sound in the speakers, it did not work. (But apparently other people could hear the audio from my mic when I tried this yesterday).
> 
>     5. Disconnect from the Google Meet call.   At this point strangely the btmon output suddenly scrolled by extremely quickly, rapidly producing lots of output in only a few seconds.
> 
>     6. Turn off headphones
> 
>     7. Stop btmon

Please attach the file here.

rptaylor, 2024-11-18:
Strange, I didn't change anything else or upgrade any other packages but now it started working normally again....

Wish I knew what happened but glad it's working. Thanks for looking anyway!

closed 2024-11-18 by rptaylor