bluez/bluez #1020: Sony WH-1000XM5 and WF-1000XM5 cannot pair

pivonroll, 2024-11-14:
I am trying to pair Sony WH-1000XM5 (headphones) and WF-1000XM5 (earbuds).
It seems like Sony products are cancelling the pairing process.
I also tried to use password: 0000, found [here](https://helpguide.sony.net/mdr/wh1000xm5/v1/en/contents/TP1000534541.html), but no luck.
I did not tried to use password for WF-1000XM5.

I can pair Bose QuietComfort buds without problem.

Hardware:
Laptop: Lenovo Yoga 720 15IKB
Bluetooth device: Asus BT500 (usb bluetooth).
I also tried to use builtin bluetooth but no luck. Same issue with Asus BT500 and builtin bluetooth.

### System and Software:
OS: Arch Linux 
Kernel: 6.11.7
Bluez: 5.79-1
Pipewire: 1:1.2.6-1
pipewire-audio: 1:1.2.6-1
pipewire-pulse: 1:1.2.6-1
libpipewire: 1:1.2.6-1
wireplumber: 0.5.6-1
libpipewire: 1:1.2.6-1

Services Running:
bluetooth.service (root)

Enabled pipewire and wireplumber with:
```
systemctl --user --now enable pipewire pipewire-pulse wireplumber
```

### Logs:
<details><summary>btmon</summary>
<p>
btmon[13482]: @ RAW Close: btmon                                                                                                                    {0x0002} 1035.864459
> HCI Event: Command Status (0x0f) plen 4                                                                                                        #247 [hci0] 1035.867987
      Read Remote Supported Features (0x01|0x001b) ncmd 2
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                      #248 [hci0] 1035.871940
        Handle: 4 Address: AC:80:0A:4C:08:75 (Sony Corporation)
        Max slots: 5
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                       #249 [hci0] 1035.874977
        Status: Success (0x00)
        Handle: 4 Address: AC:80:0A:4C:08:75 (Sony Corporation)
        Features: 0xaf 0xfe 0x0d 0xfe 0xd8 0xbf 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          u-law log synchronous data
          A-law log synchronous data
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          Extended SCO link (EV3 packets)
          AFH capable peripheral
          AFH classification peripheral
[... 120 lines cut ...]
Nov 14 14:54:01 radovan wireplumber[2716]: spa.audioconvert: 0x638f1db82e00: (0 suppressed) out of buffers on port 0 3
Nov 14 14:54:01 radovan wireplumber[2716]: spa.audioconvert: 0x638f1db82e00: (46 suppressed) out of buffers on port 0 3
Nov 14 14:54:01 radovan wireplumber[2716]: spa.bluez5: Failed to release transport /org/bluez/hci0/dev_60_AB_D2_9B_67_00/sep3/fd0: Method "Release" with signature "" on>

● pipewire.service - PipeWire Multimedia Service
     Loaded: loaded (/usr/lib/systemd/user/pipewire.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-11-14 10:22:52 CET; 6h ago
 Invocation: c1f0f33b43364bc19c45e1b39c77486c
TriggeredBy: ● pipewire.socket
   Main PID: 2715 (pipewire)
      Tasks: 3 (limit: 18914)
     Memory: 11M (peak: 12.4M)
        CPU: 258ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire.service
             └─2715 /usr/bin/pipewire

Nov 14 10:22:52 radovan systemd[1617]: Started PipeWire Multimedia Service.
Nov 14 10:32:27 radovan pipewire[2715]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-54) graph xrun not-triggered (0 suppressed)
Nov 14 10:32:27 radovan pipewire[2715]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-54) xrun state:0x7146bcac5008 pending:1/1 s:553128883629 a:553128899960 f:5>
Nov 14 10:32:29 radovan pipewire[2715]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-54) graph xrun not-triggered (0 suppressed)
Nov 14 10:32:29 radovan pipewire[2715]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-54) xrun state:0x7146bcac5008 pending:1/1 s:597773409498 a:597773507949 f:5>
Nov 14 10:32:29 radovan pipewire[2715]: pw.node: (alsa_input.pci-0000_00_1f.3.analog-stereo-55) graph xrun not-triggered (0 suppressed)
Nov 14 10:32:29 radovan pipewire[2715]: pw.node: (alsa_input.pci-0000_00_1f.3.analog-stereo-55) xrun state:0x7146bcac4008 pending:1/1 s:597774868454 a:597774999277 f:59>
</p>
</details>

CodeThomnics, 2024-11-17:
I have the same issue with my WH-1000XM5

Vudentz, 2024-11-18:
Pairing is being cancel:

```
bluetoothd[488]: @ MGMT Command: Cancel Pair Device (0x001a) plen 7                                                                          {0x0001} [hci0] 1060.976585
        BR/EDR Address: AC:80:0A:4C:08:75 (Sony Corporation)
```

Enable bluetoothd debug by passing -d and then run btmon -p debug so it can collect the daemon logs together with the HCI so we can see the full trace.

Vudentz, 2024-11-19:
Something is calling disconnect:

`bluetoothd[134627]: = [:1.645:method_call] > org.bluez.Device1.Disconnect [#15]`

Now who is the process of (D-Bus):1.645 and why it is calling Disconnect?

Are you using gnome to pair? Have you tried with bluetoothctl?

pivonroll, 2024-11-20:
> Now who is the process of (D-Bus):1.645 and why it is calling Disconnect?

Sorry, I do not understand the question. Could you elaborate a bit?

> Are you using gnome to pair? Have you tried with bluetoothctl?

I tried with `bluetoothctl` without any success.
I have not tried gnome-pair. Will let you know how it went.

Vudentz, 2024-11-20:
> > Now who is the process of (D-Bus):1.645 and why it is calling Disconnect?
> 
> Sorry, I do not understand the question. Could you elaborate a bit?

There is a process interfering with the pairing, it is calling org.bluez.Device1.Disconnect which causes the pairing to be aborted.

> > Are you using gnome to pair? Have you tried with bluetoothctl?
> 
> I tried with `bluetoothctl` without any success. I have not tried `gnome-pair`. Will let you know how it went.

pivonroll, 2024-11-20:
WOW,
So I just retried to connect using `bluetoothctl` (out of curiosity):
```
# start bluetoothcl
$ bluetoothctl

[bluetoothctl] # scan on
... just to get the address for WH-1000XM5

[bluetoothctl] # connect 80:99:E7:0B:1E:82
bluetooth]# connect 80:99:E7:0B:1E:82
Attempting to connect to 80:99:E7:0B:1E:82
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 Connected: yes
[WH-1000XM5]# [CHG] Device 41:45:BA:26:72:BD RSSI: 0xffffff9c (-100)
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 00000000-deca-fade-deca-deafdecacaff
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000110d-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000110f-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 00001131-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 45c93e07-d90d-4b93-a9db-91e5dd734e35
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 764cbf0d-bbcb-438f-a8bb-6b92759d6053
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 81c2e72a-0591-443e-a1ff-05f988593351
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 931c7e8a-540f-4686-b798-e8df0a2ad9f7
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 956c7b26-d49a-4ba8-b03f-b17d393cb6e2
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: 9b26d8c0-a8ed-440b-95b0-c4714a518bcc
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: df21fe2c-2515-4fdb-8886-f12c4d67927c
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: f76acb00-7cab-495f-bb1a-e664598fd77f
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: f7a96061-a1b3-40de-aff0-e78ec45a151e
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 UUIDs: f8d1fbe4-7966-4334-8024-ff96c9330e15
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 ServicesResolved: yes
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 Bonded: yes
[WH-1000XM5]# [CHG] Device 80:99:E7:0B:1E:82 Paired: yes
[WH-1000XM5]# [NEW] Endpoint /org/bluez/hci0/dev_80_99_E7_0B_1E_82/sep1 
[WH-1000XM5]# [NEW] Endpoint /org/bluez/hci0/dev_80_99_E7_0B_1E_82/sep2 
[WH-1000XM5]# [NEW] Transport /org/bluez/hci0/dev_80_99_E7_0B_1E_82/sep2/fd0 
[WH-1000XM5]# [CHG] Transport /org/bluez/hci0/dev_80_99_E7_0B_1E_82/sep2/fd0 Delay: 0x06a4 (1700)
[WH-1000XM5]# Connection successful

Heard pairing sound cue from headphones.
```

Played youtube video. Sound works!

The only thing which I did was to globaly disable pipewire, wireplumber and pipewire-pulse.
```
sudo systemctl disable --global pipewire-pulse wireplumber pipewire
```
but it makes no sense.

I also did:
```
systemctl disable --user pipewire.socket 
```
which makes even less sense.

I am really wondering if this was just an unfortunate coincidence.

cojosh, 2024-11-20:
I have also been experiencing the same issue for a while. It seemed that pairing didn't work after installing a firmware update from sony (may be a pure coincidence as I also updated my distros packages at the same time) .  Disabling pipewire and trying to pair via `bluetoothctl` did not solve the issue for me.

# System
```
Distro: Fedora 41
Kernel: linux 6.11.8-300.fc41.x86_64
Bluez : bluez-5.79-1.fc41.x86_64
```

# Logs

## Bluetoothctl

```
[bluetooth]# pair AC:80:0A:1C:05:E9
Attempting to pair with AC:80:0A:1C:05:E9
[bluetooth]# hci0 device_flags_changed: AC:80:0A:1C:05:E9 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 AC:80:0A:1C:05:E9 type BR/EDR connected eir_len 0
[WH-1000XM5]# [CHG] Device AC:80:0A:1C:05:E9 Connected: yes
[WH-1000XM5]# Failed to pair: org.bluez.Error.AuthenticationTimeout
```

## Btmon
```
Bluetooth monitor ver 5.79
btmon[8149]: = Note: Linux version 6.11.8-300.fc41.x86_64 (x86_64)                                                                                                                                                                   0.487248
btmon[8149]: = Note: Bluetooth subsystem version 2.22                                                                                                                                                                                0.487250
= New Index: 10:6F:D9:DC:D3:86 (Primary,USB,hci0)                                                                                                                                                                             [hci0] 0.487251
= Open Index: 10:6F:D9:DC:D3:86                                                                                                                                                                                               [hci0] 0.487251
= Index Info: 10:6F:D9:DC:D3:86 (MediaTek, Inc.)                                                                                                                                                                              [hci0] 0.487252
bluetoothctl[8067]: @ MGMT Open: bluetoothctl (privileged) version 1.23                                                                                                                                                     {0x0002} 0.487253
bluetoothd[1212]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                                                                                                                         {0x0001} 0.487254
bluetoothd[1212]: @ MGMT Command: Add Device (0x0033) plen 8                                                                                                                                                         {0x0001} [hci0] 1.953773
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Added (0x001a) plen 8                                                                                                                                                                           {0x0002} [hci0] 1.953786
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                                                                                                                                                  {0x0002} [hci0] 1.953789
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Supported Flags: 0x00000001
          Remote Wakeup
        Current Flags: 0x00000000
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                                                                                                                                                  {0x0001} [hci0] 1.953789
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Supported Flags: 0x00000001
          Remote Wakeup
        Current Flags: 0x00000000
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                                                                                      {0x0001} [hci0] 1.953792
      Add Device (0x0033) plen 7
        Status: Success (0x00)
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
bluetoothd[1212]: @ MGMT Command: Pair Device (0x0019) plen 8                                                                                                                                                        {0x0001} [hci0] 1.953829
        BR/EDR Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Capability: KeyboardDisplay (0x04)
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                                                                                                                                     #1 [hci0] 1.953848
        Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow peripheral (0x01)
[... cut ...]
          Simultaneous LE and BR/EDR (Host)
> HCI Event: Mode Change (0x14) plen 6                                                                                                                                                                                   #13 [hci0] 17.789225
        Status: Success (0x00)
        Handle: 51 Address: AC:80:0A:1C:05:E9 (Sony Corporation)
        Mode: Sniff (0x02)
        Interval: 500.000 msec (0x0320)
bluetoothd[1212]: = Pair device failed: Timeout (0x08)                                  
```

Vudentz, 2024-11-20:
> I have also been experiencing the same issue for a while. It seemed that pairing didn't work after installing a firmware update from sony (may be a pure coincidence as I also updated my distros [...]

Looks unrelated, this one times out, which perhaps is because it entering sniff mode or something, anyway I tried my XM4 on fc41 and it works just fine so I wonder if this is controller related, your seems to be mediatek's any chance to test with something different?

cojosh, 2024-12-02:
@Vudentz Thanks for your quick reply. I tested it with an old laptop I had, updating it to the newest fedora release and indeed pairing worked without issue. After updating my ditros packages on my current machine which included a pipewire update to `1.2.7` I am no longer experiencing any issues.

ValeZAA, 2025-03-29:
Issue can be closed.

closed 2025-05-28 by pivonroll