bluez/bluez #1021: Bluez Peripheral fails to recognize paired iPhone

kdewald, 2024-11-15:
Hi,

I'm running into an issue in the following setup:
- A Raspberry Pi running Ubuntu 22.04/24.04, acting as a Peripheral.
- Application exposes a characteristic with the `secure-notify` flag.
- Connections are initiated via an iPhone.

The issue I see is that passkey entry is requested at every connection, despite bonding already in place from a prior connection. I'm highlighting the portions of the btmon logs that I think are relevant and attaching the full output.

First connection:
```
@ MGMT Event: Passkey Notify (0x0017) plen 12       [hci0] 19.186025
        LE Address: 6A:FD:5C:B7:C1:E7 (Resolvable)
        Passkey: 0x000b28f5 (731381)
        Entered: 0

> ACL Data RX: Handle 64 flags 0x02 dlen 21         #534 [hci0] 27.429609
      SMP: Identity Information (0x08) len 16
        Identity resolving key: 6707fa39a8ad3990878821064ace0fcd

> ACL Data RX: Handle 64 flags 0x02 dlen 12         #535 [hci0] 27.429952
      SMP: Identity Address Information (0x09) len 7
        Address: F8:E9:4E:6E:25:74 (Apple, Inc.)

@ MGMT Event: New Identity Resolv.. (0x0018) plen 30   [hci0] 27.429990
        Store hint: Yes (0x01)
        Random address: 6A:FD:5C:B7:C1:E7 (Resolvable)
        LE Address: F8:E9:4E:6E:25:74 (Apple, Inc.)
        Key: 6707fa39a8ad3990878821064ace0fcd
```

Second connection:
```
> ACL Data RX: Handle 64 flags 0x02 dlen 11         #351 [hci0] 65.978925
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses, CT2 (0x2d)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey LinkKey (0x0b)
        Responder key distribution: EncKey IdKey LinkKey (0x0b)

< ACL Data TX: Handle 64 flags 0x00 dlen 11         #352 [hci0] 65.979016
      SMP: Pairing Response (0x02) len 6
        IO capability: DisplayOnly (0x00)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, MITM, SC, No Keypresses, CT2 (0x2d)

@ MGMT Event: Passkey Notify (0x0017) plen 12       [hci0] 66.074942
        LE Address: F8:E9:4E:6E:25:74 (Apple, Inc.)
        Passkey: 0x000620f4 (401652)
        Entered: 0

> ACL Data RX: Handle 64 flags 0x02 dlen 21         #534 [hci0] 74.528509
      SMP: Identity Information (0x08) len 16
        Identity resolving key: 6707fa39a8ad3990878821064ace0fcd

> ACL Data RX: Handle 64 flags 0x02 dlen 12         #535 [hci0] 74.528850
      SMP: Identity Address Information (0x09) len 7
        Address: F8:E9:4E:6E:25:74 (Apple, Inc.)

@ MGMT Event: New Identity Resolv.. (0x0018) plen 30   [hci0] 74.528888
        Store hint: Yes (0x01)
        Random address: 00:00:00:00:00:00 (Non-Resolvable)
        LE Address: F8:E9:4E:6E:25:74 (Apple, Inc.)
        Key: 6707fa39a8ad3990878821064ace0fcd
```

My interpretation of this is that the iPhone is requesting to pair again, despite the bonding information existing. Is that correct?

What else could I missing? Please don't hesitate to request more information if necessary.

[bluez-5.72-incoming-btmon.log](https://github.com/user-attachments/files/17781062/bluez-5.72-incoming-btmon.log)

Vudentz, 2024-11-15:
Well I guess you will have to ask the Apple folks what is going on, perhaps they don't expect us to enable encryption or something, or maybe it doesn't like our Security Request:

```
< ACL Data TX: Handle 64 flags 0x00 dlen 6                #335 [hci0] 65.618887
      SMP: Security Request (0x0b) len 1
        Authentication requirement: Bonding, MITM, SC, No Keypresses, CT2 (0x2d)
```

Vudentz, 2024-11-15:
> Well I guess you will have to ask the Apple folks what is going on, perhaps they don't expect us to enable encryption or something, or maybe it doesn't like our Security Request:
>
> ```
> < ACL Data TX: [...]

I checked our interpretation of this command and we have a key we just start the encryption, which seems to be what follows:

```
> HCI Event: LE Meta Event (0x3e) plen 13                 #339 [hci0] 65.708257
      LE Long Term Key Request (0x05)
        Handle: 64 Address: 49:7F:07:8D:92:E6 (Resolvable)
          Identity type: Random (0x01)
          Identity: F8:E9:4E:6E:25:74 (Static)
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
< HCI Command: LE Long Term Key... (0x08|0x001a) plen 18  #340 [hci0] 65.708318
        Handle: 64 Address: 49:7F:07:8D:92:E6 (Resolvable)
          Identity type: Random (0x01)
          Identity: F8:E9:4E:6E:25:74 (Static)
        Long term key: 0bb1f46ec618136c6e458bf74fc44d7e
> HCI Event: Command Complete (0x0e) plen 6               #341 [hci0] 65.708666
      LE Long Term Key Request Reply (0x08|0x001a) ncmd 1
        Status: Success (0x00)
        Handle: 64 Address: 49:7F:07:8D:92:E6 (Resolvable)
          Identity type: Random (0x01)
          Identity: F8:E9:4E:6E:25:74 (Static)
> HCI Event: Encryption Change (0x08) plen 4              #342 [hci0] 65.858368
        Status: Success (0x00)
        Handle: 64 Address: 49:7F:07:8D:92:E6 (Resolvable)
          Identity type: Random (0x01)
          Identity: F8:E9:4E:6E:25:74 (Static)
        Encryption: Enabled with AES-CCM (0x01
```

After this the link in to be considered encrypted, so I don't know why iOS is sending a Pairing Request after this, the weird thing is that it proceed as normal with GATT operations.

kdewald, 2024-11-19:
@Vudentz I've got some additional information on this issue.

We ran `bluetoothd -d` and looked at the logs of the same experiment, and noticed the following lines after a disconnection:

```
bluetoothd[2023]: src/adapter.c:dev_disconnected() Device F8:E9:4E:6E:25:74 disconnected, reason 3
bluetoothd[2023]: src/adapter.c:adapter_remove_connection()
bluetoothd[2023]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[2023]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr F8:E9:4E:6E:25:74 type 1 status 0xe
bluetoothd[2023]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[2023]: src/device.c:device_bonding_failed() status 14
```

Could this be the related to the repeated pairing attempts? I've attached all logs for reference.

[bluetoothd-log.txt](https://github.com/user-attachments/files/17822699/bluetoothd-log.txt)

kdewald, 2024-11-20:
> The bonding failure is the result of the disconnection, not the other way around.

Yes, but I think you're missing part of the picture. The sequence of events was as follows:

- Bluez acting as Peripheral.
- iPhone initiates connection, pairing passkey is prompted from the user.
- User enters correct passkey. *Bluez reports the device as bonded*.
- Bluez peripheral starts streaming data, iPhone sees correct data.
- iPhone disconnects.
- Bluetoothd reports pairing failure.

After the disconnection happens is where we see the bonding failure, as seen in these extended logs:
```
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[2023]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000c
bluetoothd[2023]: src/adapter.c:dev_disconnected() Device F8:E9:4E:6E:25:74 disconnected, reason 3
bluetoothd[2023]: src/adapter.c:adapter_remove_connection()
bluetoothd[2023]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[2023]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr F8:E9:4E:6E:25:74 type 1 status 0xe
bluetoothd[2023]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[2023]: src/device.c:device_bonding_failed() status 14
```

I'm attaching the btmon logs as well for this same connection: [btmon-log.log](https://github.com/user-attachments/files/17834886/btmon-log.log)

I insist that there is definitely something wrong going on here, could you please have a deeper look?

kdewald, 2024-12-06:
@Vudentz sorry for the delay, I've got one update:

We realized that this problem is only happening on the Raspberry Pi 4 (which uses a CYW43455 chipset) but not on other devices with different adapters.

Could this be a driver/kernel issue? Any idea how to track it down?

closed 2025-07-16 by github-actions[bot]