bluez/bluez #1025: Cannot pair with Samsung Q990B soundbar

Electry, 2024-11-19:
I'm trying to pair and connect to a Samsung Q990B soundbar, but unfortunately I don't seem to be able to.

Hardware: Raspberry Pi Zero 2 W
OS: Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l
BlueZ: 5.79, built from source

After finding the device address via `scan on` using `bluetoothctl`, and attempting to pair with it, I'm getting org.bluez.Error.AuthenticationCanceled:

```
[bluetooth]# pair 1C:86:9A:12:34:56
Attempting to pair with 1C:86:9A:12:34:56
[bluetooth]# hci0 device_flags_changed: 1C:86:9A:12:34:56 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 1C:86:9A:12:34:56 type BR/EDR connect failed (status 0x07, No Resources)
[bluetooth]# Failed to pair: org.bluez.Error.AuthenticationCanceled
```

btmon

```
bluetoothctl[667]: @ MGMT Open: bluetoothctl (privileged) version 1.22                                {0x0002} 0.437807
bluetoothd[499]: @ MGMT Open: bluetoothd (privileged) version 1.22                                    {0x0001} 0.437810
bluetoothd[499]: @ MGMT Command: Add Device (0x0033) plen 8                                    {0x0001} [hci0] 3.488029
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Added (0x001a) plen 8                                                     {0x0002} [hci0] 3.488066
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                            {0x0002} [hci0] 3.488081
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Supported Flags: 0x00000001
          Remote Wakeup
        Current Flags: 0x00000000
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                            {0x0001} [hci0] 3.488081
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Supported Flags: 0x00000001
          Remote Wakeup
        Current Flags: 0x00000000
@ MGMT Event: Command Complete (0x0001) plen 10                                                {0x0001} [hci0] 3.488098
      Add Device (0x0033) plen 7
        Status: Success (0x00)
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
bluetoothd[499]: @ MGMT Command: Pair Device (0x0019) plen 8                                   {0x0001} [hci0] 3.488344
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Capability: KeyboardDisplay (0x04)
< HCI Command: Create Connection (0x01|0x0005) plen 13                                               #1 [hci0] 3.488452
        Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow peripheral (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                            #2 [hci0] 3.489696
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11                                                         #3 [hci0] 4.405984
        Status: Connection Rejected due to Limited Resources (0x0d)
        Handle: 12
        Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                   {0x0002} [hci0] 4.406034
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Status: No Resources (0x07)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                   {0x0001} [hci0] 4.406034
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Status: No Resources (0x07)
@ MGMT Event: Command Complete (0x0001) plen 10                                                {0x0001} [hci0] 4.406067
      Pair Device (0x0019) plen 7
        Status: No Resources (0x07)
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
bluetoothd[499]: @ MGMT Command: Remove Device (0x0034) plen 7                                 {0x0001} [hci0] 4.436345
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
@ MGMT Event: Device Removed (0x001b) plen 7                                                   {0x0002} [hci0] 4.453288
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
@ MGMT Event: Command Complete (0x0001) plen 10                                                {0x0001} [hci0] 4.453318
      Remove Device (0x0034) plen 7
        Status: Success (0x00)
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
```
(address altered for privacy)

I am able to pair and connect to the soundbar using my phone or windows PC just fine.

Is there anything else I can try?

Vudentz, 2024-11-19:
Actually the device is rejecting the connection:

```
> HCI Event: Connect Complete (0x03) plen 11                                                         #3 [hci0] 4.405984
        Status: Connection Rejected due to Limited Resources (0x0d)
        Handle: 12
        Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
```

Perhaps you have to disconnect your phone before you attempt to connect to it?

Electry, 2024-11-19:
I didn't have any other device connected to it before attempting to pair.

The device seems to show up in both LE and BR/EDR scans with the same address.
Once every X attemps, I get following:
```
[bluetooth]# pair 1C:86:9A:12:34:56
Attempting to pair with 1C:86:9A:12:34:56
[bluetooth]# hci0 device_flags_changed: 1C:86:9A:12:34:56 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 1C:86:9A:12:34:56 type LE Public connected eir_len 31
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 Connected: yes
[[AV] Samsung Soundbar Q990B]# [agent] Passkey: 335068
[[AV] Samsung Soundbar Q990B]# [NEW] Primary Service (Handle 0x0001)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[[AV] Samsung Soundbar Q990B]# [NEW] Characteristic (Handle 0x0002)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[[AV] Samsung Soundbar Q990B]# [NEW] Primary Service (Handle 0x0028)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0028
        ade3d529-c784-4f63-a987-eb69f70ee816
        Vendor specific
[[AV] Samsung Soundbar Q990B]# [NEW] Characteristic (Handle 0x0029)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0028/char0029
        e9241982-4580-42c4-8831-95048216b256
        Vendor specific
[[AV] Samsung Soundbar Q990B]# [NEW] Descriptor (Handle 0x0157)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0028/char0029/desc002b
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[[AV] Samsung Soundbar Q990B]# [NEW] Characteristic (Handle 0x002c)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0028/char002c
        ad7b334f-4637-4b86-90b6-9d787f03d218
        Vendor specific
[[AV] Samsung Soundbar Q990B]# [NEW] Descriptor (Handle 0x0157)
        /org/bluez/hci0/dev_1C_86_9A_12_34_56/service0028/char002c/desc002e
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 UUIDs: ade3d529-c784-4f63-a987-eb69f70ee816
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 ServicesResolved: yes
[[AV] Samsung Soundbar Q990B]# hci0 1C:86:9A:12:34:56 type LE Public disconnected with reason 2
[[AV] Samsung Soundbar Q990B]# [CHG] Device 1C:86:9A:12:34:56 ServicesResolved: no
[bluetooth]# [CHG] Device 1C:86:9A:12:34:56 Connected: no
[bluetooth]# Failed to pair: org.bluez.Error.AuthenticationCanceled
```

btmon:

<details><summary>Log</summary>

```
bluetoothctl[857]: @ MGMT Open: bluetoothctl (privileged) version 1.22                                                                                  {0x0002} 0.452442
bluetoothd[499]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                      {0x0001} 0.452445
bluetoothd[499]: @ MGMT Command: Add Device (0x0033) plen 8                                                                                      {0x0001} [hci0] 3.085924
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Added (0x001a) plen 8                                                                                                       {0x0002} [hci0] 3.085975
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Action: Allow incoming connection (0x01)
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                                                                              {0x0002} [hci0] 3.085999
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Supported Flags: 0x00000001
          Remote Wakeup
        Current Flags: 0x00000000
@ MGMT Event: Device Flags Changed (0x002a) plen 15                                                                                              {0x0001} [hci0] 3.085999
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
[... 656 lines cut ...]
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                       #99 [hci0] 34.568117
        Status: Success (0x00)
        Handle: 64 Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                               {0x0002} [hci0] 34.568167
        LE Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                               {0x0001} [hci0] 34.568167
        LE Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                 {0x0001} [hci0] 34.568214
      Pair Device (0x0019) plen 7
[... cut ...]
@ MGMT Event: Device Removed (0x001b) plen 7                                                                                                    {0x0002} [hci0] 34.703458
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                 {0x0001} [hci0] 34.703481
      Remove Device (0x0034) plen 7
        Status: Success (0x00)
        BR/EDR Address: 1C:86:9A:12:34:56 (Samsung Electronics Co.,Ltd)
```

</details>

Vudentz, 2024-11-19:
@Electry seems like it has some LE support, but I don't think it is actually LE Audio, btw are you putting it into what is typically referred as pairing mode? Some device may have a limit number of pairing slots so you may want to reset if that is the case.

Electry, 2024-12-07:
A bit more info on the issue

Here is what I tried:
- different Raspberry Pi Zero 2W using built-in BT BCM43436 controller - same issue
- different Raspberry Pi Zero 2W using Realtek RTL8761 USB BT controller - same issue
- Z790 Intel 14th gen desktop PC using the same Realtek RTL8761 USB BT controller - paired just fine

All tested with bluez 5.66 which is packaged in debian bookworm apt repo.

Here is dumpcap for the RPi:

```
    1   0.000000         host → controller   HCI_CMD 17 Sent Create Connection
    2   0.000605   controller → host         HCI_EVT 7 Rcvd Command Status (Create Connection)
    3   0.209064   controller → host         HCI_EVT 14 Rcvd Connect Complete
```
```
0000  01 05 04 0d 56 34 12 9a 86 1c 18 cc 02 00 00 00   ......o.........
0010  01                                                .

0000  04 0f 04 00 01 05 04                              .......

0000  04 03 0b 0d 0c 00 56 34 12 9a 86 1c 01 00         ........o.....
```

And one for the desktop pc:

```
    1   0.000000         host → controller   HCI_CMD 17 Sent Create Connection
    2   0.097477   controller → host         HCI_EVT 7 Rcvd Command Status (Create Connection)
    3   0.361493   controller → host         HCI_EVT 11 Rcvd Role Change
    4   0.369479   controller → host         HCI_EVT 14 Rcvd Connect Complete
    5   0.369702         host → controller   HCI_CMD 6 Sent Read Remote Supported Features
    6   0.375472   controller → host         HCI_EVT 7 Rcvd Command Status (Read Remote Supported Features)
    7   0.390476   controller → host         HCI_EVT 7 Rcvd Link Supervision Timeout Changed
    8   0.391468   controller → host         HCI_EVT 6 Rcvd Max Slots Change
    9   0.393468   controller → host         HCI_EVT 6 Rcvd Max Slots Change
   10   0.394474   controller → host         HCI_EVT 14 Rcvd Read Remote Supported Features
   11   0.394536         host → controller   HCI_CMD 7 Sent Read Remote Extended Features
   12   0.395468   controller → host         HCI_EVT 7 Rcvd Command Status (Read Remote Extended Features)
   13   0.405269 SamsungE_12:34:56 () → localhost () L2CAP 15 Rcvd Information Request (Extended Features Mask)
... and more
```
```
0000  01 05 04 0d 56 34 12 9a 86 1c 18 cc 02 00 00 00   ......o.........
0010  01                                                .

0000  04 0f 04 00 02 05 04                              .......

0000  04 12 08 00 56 34 12 9a 86 1c 01                  ......o....

0000  04 03 0b 00 05 00 56 34 12 9a 86 1c 01 00         ........o.....

0000  01 1b 04 02 05 00                                 ......

0000  04 0f 04 00 02 1b 04                              .......

0000  04 38 04 05 00 40 1f                              .8...@.

0000  04 1b 03 05 00 05                                 ......

0000  04 1b 03 05 00 05                                 ......

0000  04 0b 0b 00 05 00 bf 3e 8d fa db ff 7b 87         .......>....{.

0000  01 1c 04 03 05 00 01                              .......

0000  04 0f 04 00 02 1c 04                              .......

0000  02 05 20 0a 00 06 00 01 00 0a 02 02 00 02 00      .. ............
... and more
```

```
[bluetooth]# pair 1C:86:9A:12:34:56
Attempting to pair with 1C:86:9A:12:34:56
[CHG] Device 1C:86:9A:12:34:56 Connected: yes
Request confirmation
[agent] Confirm passkey 010347 (yes/no): yes
[CHG] Device 1C:86:9A:12:34:56 Bonded: yes
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:86:9A:12:34:56 UUIDs: a23d00bc-217c-123b-9c00-fc44577136ee
[CHG] Device 1C:86:9A:12:34:56 UUIDs: ade3d529-c784-4f63-a987-eb69f70ee816
[CHG] Device 1C:86:9A:12:34:56 ServicesResolved: yes
[CHG] Device 1C:86:9A:12:34:56 Paired: yes
Pairing successful
[CHG] Device 1C:86:9A:12:34:56 ServicesResolved: no
[CHG] Device 1C:86:9A:12:34:56 Connected: no
```

is there anything I can try next?

closed 2025-07-16 by github-actions[bot]