bluez/bluez #1027: Bluez 5.79 - Unable to receive files from my Xiaomi android phone

azioga, 2024-11-23:
I'm a Debian Testing user. 
After 5.79 update and new pairing with my phone I'm unable to receive files.
It works with 5.77.

```
# uname -a
Linux user 6.11.7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.11.7-1 (2024-11-09) x86_64 GNU/Linux
```
```
# btmgmt
[mgmt]# info
[mgmt]# Index list with 1 item
[mgmt]# hci0:   Primary controller
[mgmt]#         addr DC:41:A9:AC:0B:B5 version 11 manufacturer 2 class 0x6c0104
[mgmt]#         supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech
[mgmt]#         current settings: powered connectable bondable ssp br/edr le secure-conn wide-band-speech
[mgmt]#         name ilprof5
[mgmt]#         short name
[mgmt]# hci0:   Configuration options
[mgmt]#         supported options: public-address
[mgmt]#         missing options:
[mgmt]# hci0 AC:1E:9E:4C:48:30 type BR/EDR connected eir_len 5
[mgmt]# hci0 AC:1E:9E:4C:48:30 type BR/EDR disconnected with reason 2
```
btmon
```
> HCI Event: Connect Request (0x04) plen 10                                                                                                                   #89 [hci0] 183.077965
        Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Class: 0x5a020c
          Major class: Phone (cellular, cordless, payphone, modem)
          Minor class: Smart phone
          Networking (LAN, Ad hoc)
          Capturing (Scanner, Microphone)
          Object Transfer (v-Inbox, v-Folder)
          Telephony (Cordless telephony, Modem, Headset)
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                                                                                                 #90 [hci0] 183.078031
        Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                     #91 [hci0] 183.078961
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                                                                        #92 [hci0] 183.193971
        Status: Success (0x00)
        Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4                                                                                                                             #93 [hci0] 183.396964
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                  #94 [hci0] 183.399961
        Status: Success (0x00)
        Handle: 256
        Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                            #95 [hci0] 183.400140
        Handle: 256 Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
btmon[6686]: @ RAW Open: btmon (privileged) version 2.22                                                                                                        {0x0003} 183.400450
btmon[6686]: @ RAW Close: btmon                                                                                                                                 {0x0003} 183.400460
> HCI Event: Command Status (0x0f) plen 4                                                                                                                     #96 [hci0] 183.400971
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Page Scan Repetition Mode Change (0x20) plen 7                                                                                                   #97 [hci0] 183.405973
        Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Page scan repetition mode: R1 (0x01)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                                   #98 [hci0] 183.409955
        Handle: 256 Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Max slots: 5
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                                    #99 [hci0] 183.411932
[... 247 lines cut ...]
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                       #129 [hci0] 183.520956
        Num handles: 1
        Handle: 256 Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Count: 1
        #128: len 12 (24 Kb/s)
        Latency: 4 msec (3-28 msec ~4 msec)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                               #130 [hci0] 187.532695
        Handle: 256 Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                    #131 [hci0] 187.534907
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                               #132 [hci0] 187.620913
        Status: Success (0x00)
        Handle: 256 Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                        {0x0002} [hci0] 187.620983
        BR/EDR Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                        {0x0001} [hci0] 187.620983
        BR/EDR Address: AC:1E:9E:4C:48:30 (Xiaomi Communications Co Ltd)
        Reason: Connection terminated by local host (0x02)

```
Best, Luca.

azioga, 2024-11-24:
btmon alexa busy

```
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                                                                       #398 [hci0] 236.384949
        Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow peripheral (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                    #399 [hci0] 236.500111
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Vendor (0xff) plen 4                                                                                                                            #400 [hci0] 237.132110
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                 #401 [hci0] 237.135075
        Status: Success (0x00)
        Handle: 256
        Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                           #402 [hci0] 237.135309
        Handle: 256 Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
btmon[8371]: @ RAW Open: btmon (privileged) version 2.22                                                                                                        {0x0002} 237.135626
btmon[8371]: @ RAW Close: btmon                                                                                                                                 {0x0002} 237.135638
> HCI Event: Command Status (0x0f) plen 4                                                                                                                    #403 [hci0] 237.136076
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                                  #404 [hci0] 237.140059
        Handle: 256 Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Max slots: 5
> HCI Event: Read Remote Supported Features (0x0b) plen 11                                                                                                   #405 [hci0] 237.144103
        Status: Success (0x00)
        Handle: 256 Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Features: 0xbf 0x3e 0x8d 0xfe 0xdb 0xff 0x7b 0x87
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          SCO link
          HV2 packets
          HV3 packets
          CVSD synchronous data
          Power control
          Transparent synchronous data
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          Extended SCO link (EV3 packets)
          EV4 packets
          EV5 packets
          AFH capable peripheral
          AFH classification peripheral
[... 28 lines cut ...]
        Type: Extended features supported (0x0002)
@ MGMT Event: Device Connected (0x000b) plen 13                                                                                                          {0x0001} [hci0] 237.149449
        BR/EDR Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 0
> HCI Event: Read Remote Extended Features (0x23) plen 13                                                                                                    #409 [hci0] 237.152088
        Status: Success (0x00)
        Handle: 256 Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Page: 1/2
        Features: 0x0f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          Secure Simple Pairing (Host Support)
          LE Supported (Host)
          Simultaneous LE and BR/EDR (Host)
          Secure Connections (Host Support)
bluetoothd[2838]: = profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:41:6A:54:9B:53: Connection refused (111)                                                     276.649620
bluetoothd[2838]: = src/service.c:btd_service_connect() a2dp-source profile connect failed for AC:41:6A:54:9B:53: Device or resource busy                                276.649735
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                               #410 [hci0] 297.311065
        Status: Success (0x00)
        Handle: 256 Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                        {0x0001} [hci0] 297.311108
        BR/EDR Address: AC:41:6A:54:9B:53 (Amazon Technologies Inc.)
        Reason: Connection terminated by remote host (0x03)
```

Vudentz, 2024-11-26:
@azioga weird, don't think we had done any changes to A2DP in 5.79, we do have some changes done lately for codec switch but that hasn't been released yet, are you sure this is a regression on userspace? Perhaps the kernel has been changed as well, there was some backported patches that caused some problems in the past few weeks.

azioga, 2024-11-26:
thanks for the reply.

my echo pop right now ;)
```
nov 26 17:03:48 ilprof5 bluetoothd[7077]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:41:6A:54:9B:53: Connection refused (111)
nov 26 17:03:48 ilprof5 bluetoothd[7077]: src/service.c:btd_service_connect() a2dp-source profile connect failed for AC:41:6A:54:9B:53: Device or resource busy
```
It's a strange situation.
With kernel 6.10.11 and bluez 5.77 everything works.
With kernel 6.11.4 nothing works.
Now with kernel 6.11.9 and bluez 5.79 everything is very unstable. Not always, but often, I cannot connect the devices.
systemctl restart bluetooth sometimes helps, other times I send a file to the phone and everything starts working again.....
If there have been no changes to A2DP, 5.79 it should work fine with kernel 6.10.11....I'll give it a try

azioga, 2024-11-26:
I tried with kernel 6.10.11/bluez5.79 and I can actually connect my echo pop without problems. So, for this problem, the kernel is the culprit.
But even with this configuration I can't send files from the cell phone to the PC

closed 2025-07-16 by github-actions[bot]