bluez/bluez #1030: How to test LE Audio in bluetootctl

michael-kong754, 2024-11-26:
I want to test LE Audio use BlueZ 5.77 and linux kernel 6.6.50 and i use the broadcast-source.bt and  broadcast-sink.bt in https://github.com/bluez/bluez/tree/master/client/scripts . However, the broadcast sink didn't sync to broadcast source, did i miss some steps? Here is my log:
[broadcast-source.txt](https://github.com/user-attachments/files/17913997/broadcast-source.txt)
[broadcast-sink.txt](https://github.com/user-attachments/files/17914002/broadcast-sink.txt)
In addition, if i want to test unicast-server and unicast-client in bluetoothctl, which commands should i input?
Thank you!

Vudentz, 2024-11-26:
@michael-kong754 for unicast in theory all you need is to initialize bluetoothctl with -e to enable local endpoints and then on connect it should setup transports/streams automatically, but it is not recommended to do that alongside Pipewire since that would also register its own endpoints, the same applies to broadcast if Pipewire is registering endpoints for broadcast it will likely interfere with the setup.

Usually what I recommend for people testing LE Audio is to run it under [test-runner](https://github.com/bluez/bluez/blob/master/doc/test-runner.rst), that way you can guarantee nothing will interfere since it spawn a vm which you can run the latest changes from bluetooth-next, etc.

iulia-tanasescu, 2024-11-26:
@michael-kong754 Linux kernel version 6.6.50 does not have the full support needed for broadcast sink, it should work with the latest stable version (6.12.1).

michael-kong754, 2024-11-27:
> @michael-kong754 Linux kernel version 6.6.50 does not have the full support needed for broadcast sink, it should work with the latest stable version (6.12.1).

Hi @iulia-tanasescu ,
  For kernel 6.6.50, dose it have full support for unicast server and unicast-client, or i need to update the kernel to the latest stable version (6.12.1) too?
Thank you!

michael-kong754, 2024-11-28:
> @michael-kong754 for unicast in theory all you need is to initialize bluetoothctl with -e to enable local endpoints and then on connect it should setup transports/streams automatically, but it is [...]

Many thanks for the detailed answer, it indeed solved my problem.

closed 2024-11-28 by michael-kong754