bluez/bluez #1036: Pause A2DP stream on Source side

maheshtalewad, 2024-12-03:
Hi,
We are doing A2DP streaming from DUT with pw-play command.
pw-play 1.wav 
Here DUT is A2DP source. Before streaming we are starting pipewire with following command 
systemctl --user start pipewire wireplumber.
Since there is no option for pause in pw-play, we want to pause[suspend] the A2DP stream on DUT [A2DP Source] side.

I tried with sending following dbus command on DUT side but no success.
dbus-send --system --print-reply --dest=org.bluez /org/bluez/hci0/dev_11_11_22_CF_94_D6 org.bluez.MediaControl1.Pause

Could you please help us to get dbus command to pause A2DP stream or is there any way to pause[suspend] A2DP stream on DUT [A2DP Source] side in this case.

Thanks and regards,
Mahesh Vithal Talewad

pv, 2025-01-11:
You'll need to have the program outputting the audio to pause it, bluez can't do that.

pv, 2025-01-13:
Or, if you don't care what audio is precisely played, you can kill the pw-play process to AVDTP suspend, and start it again to  AVDTP resume.