bluez/bluez #1037: Can only connect to one A2DP sink at a time on 5.78

eternaleye, 2024-12-04:
I have two devices capable of acting as an A2DP sink:

- A pair of speakers (BESTISAN SR04)
- A pair of headphones (V-Moda Crossfade 2)

Historically, I've been able to connect both at the same time, in A2DP mode. However, when I tried just now, it failed:

- If I connect the speakers first, they connect as A2DP, and the headphones are forced into HSP/HFP
- If I connect the headphones first, they connect as A2DP, but the speakers fail to connect afterwards

Logs from speakers-first:

```
12 04 00:14:37 Arkadios bluetoothd[156210]: /org/bluez/hci0/dev_27_CC_81_21_28_BD/sep1/fd3: fd(42) ready
12 04 00:14:42 Arkadios bluetoothd[156210]: /bluez-5.78/src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

Logs from headphones-first:

```
12 04 00:15:27 Arkadios bluetoothd[156210]: /bluez-5.78/profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
```

Vudentz, 2024-12-11:
> I have two devices capable of acting as an A2DP sink:
>
>     * A pair of speakers (BESTISAN SR04)
>
>     * A pair of headphones (V-Moda Crossfade 2)
>
>
> Historically, I've been able to connect both at the [...]

Is this with pipewire? From the looks of it there isn't an endpoint available to select so I wonder how it worked before?

pv, 2025-01-11:
It probably worked before if sound server exposes multiple endpoints, eg. pipewire earlier used to expose different SBC configurations as separate endpoints, nowadays it's just one ep as some devices fail if there are too many eps.

This seems to be the problem that the a2dp/avdtp code in BlueZ does not allow different devices to use same SEP, and would need some restructuring to make it work.

eternaleye, 2025-01-12:
Unfortunately, even with Pipewire compiled with support for aptX, ldac, and aac (the last of which I recompiled to add), nothing but SBC and SBC-XQ show up for the two, so I cannot test the hypothesis.

EasyNetDev, 2025-02-16:
Hi all,

Today I was trying to connect 2 blue speakers: Sony SRS-XB12 and Ultimate MEGABOOM 3.
Bluez is failing like this when I'm trying to connect in secondary step one of them:

```
bluetoothd[8027]: profiles/audio/a2dp.c:a2dp_reconfigure() avdtp_set_configuration: Device or resource busy
bluetoothd[8027]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
bluetoothd[8027]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
```

To explain a little bit better:
1. Ultimate MEGABOOM 3 has only A2DP profiles.
2. Sony SRS-XB12 has A2DP and also HSP/HFP profiles (very poor quality codecs).
3. My Bluetooth adapter is 5.2 integrated into WiFi card: MEDIATEK Corp. MT7921K (RZ608) Wi-Fi 6E 80MHz

If I'm connecting MEGABOOM 3 as first one in Bluetooth it selects A2DP. SRS-XB12 will get an error like the ones above and it will show in Sounds Settings in KDE only with HSP/HFP profiles.
If I'm connecting SRS-XB12 as first device in Bluetooth, it gets also A2DP profile. MEGABOOM 3 will give an error like the ones above and will not be shown in the Sound Settings, because it has only A2DP profile which is already taken by SRS-XB12.

After 1 or 2h or searching, I've noticed this and also I found this issue reported in BlueZ.
Somehow, if BlueZ doesn't support multiple A2DP devices, why I can't find anything in manual or described anywhere?

Vudentz, 2025-05-06:
This will probably require several changes as explained in https://github.com/bluez/bluez/issues/1064#issuecomment-2590840533.

EasyNetDev, 2025-05-09:
@Vudentz for sure I will give it a try.

EasyNetDev, 2025-05-11:
I need to work a little bit on the patch because the version of BueZ 5.82 on Debian Trixie has some patches which conflicts in a specific area with your patch.

EasyNetDev, 2025-05-12:
Hi @Vudentz ,

I've tested it. In the first instance shows me both devices with A2DP Sink (2 of them: codec SBC and SBC+XQ). But the second device connected is not able to play.
If I'm connecting / disconnecting them at a certain point the Sony SRS-X12 is shows only HSP/HFP codec CVSD.

The second device connected is still not able to play anything.

Vudentz, 2025-05-12:
> Hi [@Vudentz](https://github.com/Vudentz) ,
>
> I've tested it. In the first instance shows me both devices with A2DP Sink (2 of them: codec SBC and SBC+XQ). But the second device connected is not able [...]

Can you please add the btmon traces? @pv does pipewire supports multiple transports with the same endpoint? I though this shouldn't care if there is e.g. SetConfiguration while another one already exists but perhaps there is still something that prevents us to do something like that.

pv, 2025-05-12:
>  does pipewire supports multiple transports with the same endpoint?

BAP is doing that and it works, should work for A2DP too.

closed 2025-05-16 by BluezTestBot

zevrochon-hash, 2026-10-12:
i get the same isue i think 2 jbl charge's :
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
updated to the latest BlueZ version and still hapens and also on my old 5.82 any help  would be greatly apreaciated!