bluez/bluez #1039: main.conf does not restore kernel defaults after commenting a previously-uncommented setting

abqjln, 2024-12-09:
Symptom: commenting out a previously-uncommented value in main.conf does not return it to the original default state, so the main.conf comments are misleading.

*****
Raspberry Pi 4
Operating System: Debian GNU/Linux 12 (bookworm)
Kernel: Linux 6.6.51+rpt-rpi-v8
bluetoothctl: 5.79

If you change a uncomment/change value in main.conf, and systemctl daemon-reload && systemctl restart bluetooth, it updates the variables in /sys/kernel/debug/bluetooth/hcix. However, if you then comment out the change, and  restart it does not restore the original default values, it just leaves them as they were.

This has led to many a frustrating phase error with this developer where I trusted the text comments about what is default.

On install, perhaps bluez saves a copy of the default kernel settings (main.conf.kernel?) and then starts with those as other changes are made? This would keep the comments in phase with the actual state of the machine. Thanks. 

*****
1) main.conf 
Set ConnectionSupervisionTimeout = 200
systemctl daemon-reload && systemctl restart bluetooth
/sys/kernel/debug/bluetooth/hci0/supervision_timeout contains 200

2) main.conf 
Comment #ConnectionSupervisionTimeout = 200
systemctl daemon-reload && systemctl restart bluetooth
/sys/kernel/debug/bluetooth/hci0/supervision_timeout still contains 200

3) main.conf 
Set ConnectionSupervisionTimeout = 42
systemctl daemon-reload && systemctl restart bluetooth
/sys/kernel/debug/bluetooth/hci0/supervision_timeout contains 42

Vudentz, 2024-12-11:
@abqjln that indeed sounds like a bug, we probably need to always reprogram or have some way to read the current values.

abqjln, 2025-04-28:
Is this in the queue to be fixed?  An interim workaround?

closed 2025-09-20 by github-actions[bot]