bluez/bluez #1049: Ubuntu 24.04 build error

Fanhang32, 2024-12-21:
fanhang@vbox:~/bluez$ uname -a
Linux vbox 6.8.0-50-generic #51-Ubuntu SMP PREEMPT_DYNAMIC Sat Nov  9 17:58:29 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

fanhang@vbox:~/bluez$ make
make --no-print-directory all-am
  CC       emulator/serial.o
In file included from /usr/include/features.h:502,
                 from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:28,
                 from emulator/serial.c:17:
In function ‘read’,
    inlined from ‘serial_read_callback’ at emulator/serial.c:90:8,
    inlined from ‘serial_read_callback’ at emulator/serial.c:75:13:
**/usr/include/x86_64-linux-gnu/bits/unistd.h:28:10: error: ‘__read_alias’ specified size between 18446744073709490177 and 18446744073709551615 exceeds maximum object size 9223372036854775807 [-Werror=stringop-overflow=]
   28 |   return __glibc_fortify (read, __nbytes, sizeof (char),**
      |          ^~~~~~~~~~~~~~~
emulator/serial.c: In function ‘serial_read_callback’:
emulator/serial.c:78:24: note: destination object allocated here
   78 |         static uint8_t buf[4096];
      |                        ^~~
/usr/include/x86_64-linux-gnu/bits/unistd-decl.h:29:16: note: in a call to function ‘__read_alias’ declared with attribute ‘access (write_only, 2, 3)’
   29 | extern ssize_t __REDIRECT_FORTIFY (__read_alias, (int __fd, void *__buf,
      |                ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:7883：emulator/serial.o] 错误 1
make: *** [Makefile:4698：all] 错误 2

Fanhang32, 2025-01-05:
fanhang@vbox:~/bluez$ x86_64-linux-gnu-gcc --version
x86_64-linux-gnu-gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pmontesShsConsultores, 2025-02-03:
Found the same issue on Bluez 5.79. On emulator/serial.c

```

again:
	len = read(serial->fd, buf + serial->pkt_offset,
			sizeof(buf) - serial->pkt_offset);
```
serial->pkt_offset is uint16_t, so the compiler warns that `nbytes = sizeof(buf) - serial->pkt_offset` could underflow, and so the large size.

1. From user perspective: I finished build ignoring hiding the error: ```make CFLAGS="-Wno-error=stringop-overflow"```. Note this does not address the issue (as I don't intently use the emulator I can't tell you the emulator works OK by ignoring this).

2. Trying to understand what's going on there: as the pkt_offset comes from mainloop I didn't find where is it initialized when it enters the callback the first time. The offset is 0 whenever the HCI packet is complete; if the packet byte count is lesser than  HCI_COMMAND_HDR_SIZE, the callback increases the packet offset and returns, waiting for the next packet. My patch for this is to print an error if the offset is too large, return (perhaps needs discarding the whole packet)

--- a/emulator/serial.c
+++ b/emulator/serial.c
@@ -79,6 +79,7 @@ static void serial_read_callback(int fd, uint32_t events, void *user_data)
        uint8_t *ptr = buf;
        ssize_t len;
        uint16_t count;
+       ssize_t available;
        if (events & (EPOLLERR | EPOLLHUP)) {
                mainloop_remove_fd(serial->fd);
@@ -87,8 +88,15 @@ static void serial_read_callback(int fd, uint32_t events, void *user_data)
        }
 
 again:
+
+       if(serial->pkt_offset > sizeof(buf)) {
+               printf("packet offset overflow\n");
+               serial->pkt_offset = 0;
+               // Discard packet?
+               return;
+       }
+
+       available = sizeof(buf) - serial->pkt_offset;
        len = read(serial->fd, buf + serial->pkt_offset,
-                       sizeof(buf) - serial->pkt_offset);
+                       available);
        if (len < 0) {

fdanis-oss, 2025-02-04:
Another way to fix this is to increase the buffer to max uint16, so `sizeof(buf) - serial->pkt_offset` couldn't underflow:
```
--- a/emulator/serial.c
+++ b/emulator/serial.c
@@ -75,7 +75,7 @@ static void serial_write_callback(const struct iovec *iov, int iovlen,
 static void serial_read_callback(int fd, uint32_t events, void *user_data)
 {
        struct serial *serial = user_data;
-       static uint8_t buf[4096];
+       static uint8_t buf[65536];
        uint8_t *ptr = buf;
        ssize_t len;
        uint16_t count;
```

fdanis-oss, 2025-02-04:
I also got the following error while building in 24.04.1:
```
In file included from /usr/include/string.h:548,
                 from emulator/btdev.c:20:
In function ‘memcpy’,
    inlined from ‘le_per_adv_new’ at emulator/btdev.c:5694:2,
    inlined from ‘cmd_pa_create_sync’ at emulator/btdev.c:5713:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:29:10: error: ‘__builtin_memcpy’ offset [0, 5] is out of the bounds [0, 0] [-Werror=array-bounds=]
   29 |   return __builtin___memcpy_chk (__dest, __src, __len,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   30 |                                  __glibc_objsize0 (__dest));
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/string.h:548,
                 from emulator/btdev.c:20:
In function ‘memcpy’,
    inlined from ‘le_per_adv_new’ at emulator/btdev.c:5694:2,
    inlined from ‘cmd_pa_create_sync’ at emulator/btdev.c:5713:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:29:10: error: ‘__builtin_memcpy’ offset [0, 5] is out of the bounds [0, 0] [-Werror=array-bounds=]
   29 |   return __builtin___memcpy_chk (__dest, __src, __len,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   30 |                                  __glibc_objsize0 (__dest));
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/string.h:548,
                 from emulator/btdev.c:20:
In function ‘memcpy’,
    inlined from ‘le_per_adv_new’ at emulator/btdev.c:5694:2,
    inlined from ‘cmd_pa_create_sync’ at emulator/btdev.c:5713:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:29:10: error: ‘__builtin_memcpy’ offset [0, 5] is out of the bounds [0, 0] [-Werror=array-bounds=]
   29 |   return __builtin___memcpy_chk (__dest, __src, __len,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   30 |                                  __glibc_objsize0 (__dest));
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:8784: emulator/android_android_tester-btdev.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:7878: emulator/btdev.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:9414: emulator/android_ipc_tester-btdev.o] Error 1
make: *** [Makefile:4696: all] Error 2
```
with:
```
$ uname -a
Linux fdanis 6.8.0-52-generic #53-Ubuntu SMP PREEMPT_DYNAMIC Sat Jan 11 00:06:25 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
$ gcc --version
gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

This seems to be due to the `new0` macro, replacing it by direct call to `calloc` fixed the build:
```
diff --git a/emulator/btdev.c b/emulator/btdev.c
index 70229d9ee..9be8eb3c9 100644
--- a/emulator/btdev.c
+++ b/emulator/btdev.c
@@ -5687,7 +5687,7 @@ static struct le_per_adv *le_per_adv_new(struct btdev *btdev,
 {
        struct le_per_adv *per_adv;
 
-       per_adv = new0(struct le_per_adv, 1);
+       per_adv = calloc(sizeof(struct le_per_adv), 1);
 
        per_adv->dev = btdev;
        per_adv->addr_type = addr_type;
```

Vudentz, 2025-02-04:
> I also got the following error while building in 24.04.1:
>
> ```
> In file included from /usr/include/string.h:548,
>                  from emulator/btdev.c:20:
> In function ‘memcpy’,
>     inlined from [...]

This is a different issue though, anyway Im adding some of these new warnings to BlueZ so we can capture it with the likes of bootstrap-configure.

closed 2025-02-18 by BluezTestBot