bluez/bluez #1055: [BlueZ v5.66] Python DBus method CreateNetwork failed

cyborgkid0110, 2025-01-01:
Try to create a mesh network with an initial node, similar to test-mesh example in Bluez v5.54:
```
uuid_bytes = uuid.uuid4().bytes
uuid_str = array_to_string(uuid_bytes)

print(set_yellow('Creating with UUID ') + set_green(uuid_str))
mesh_net.CreateNetwork(app.get_path(), uuid_bytes,
			reply_handler=create_cb,
			error_handler=create_error_cb)

... rest of code is similar to test-mesh example in bluez v5.54
```
However, it keep failed:
```
!!! Install termcolor module for better experience !!!
Register OnOff Server model on element 0
OnOff Server 
State is OFF
Register Vendor model on element 0
Register OnOff Client model on element 1
OnOff Client
Register OnOff Client model on element 1
OnOff Client
*** MAIN MENU ***
token  - set node ID (token)
create  - create mesh network
appkey-index  - set AppKey index
appkey-new  - create appkey
scan-start  - scan for near unprovisioned devs
scan-cancel  - cancel scanning
add  - add device to mesh network
join  - join mesh network
attach  - attach mesh node
remove  - delete node
uuid  - set remote uuid
vendor-send  - send raw vendor message
client-menu  - On/Off client menu
help  - show list command
quit  - exit the test
create
Creating with UUID d1eab95d83224e06ab0bc4ee27b464a5
Create procedure failed:  org.bluez.mesh.Error.Failed: Operation failed
```

Also, I notice that when test-mesh example remove 'create' feature in newer Bluez version, the CreateNetwork() method in Python DBus doesn't work anymore (with C DBus it works fine). Is it depreciated?

bgix, 2025-01-01:
Is the bluetooth-meshd daemon running with root privledges?

$ ps uax l grep bluetooth-meshd

?

cyborgkid0110, 2025-01-01:
> Is the bluetooth-meshd daemon running with root privledges? $ ps uax l grep bluetooth-meshd ?
> […](#)

yes, i run it with sudo:
```
pi@raspberrypi:~ $ ps aux | grep bluetooth-meshd
root      6191  0.0  0.2  13012  4224 pts/5    S+   11:21   0:00 sudo /home/pi/bluez-5.66/mesh/bluetooth-meshd -nd
root      6192  0.0  0.0   2420  1536 pts/5    S+   11:21   0:00 /home/pi/bluez-5.66/mesh/bluetooth-meshd -nd
pi        6609  0.0  0.1   7444  1920 pts/1    S+   11:51   0:00 grep --color=auto bluetooth-meshd
```

closed 2025-01-07 by cyborgkid0110