bluez/bluez #1056: Bluetooth OOB using QR code and Bluez

Altair-Bueno, 2025-01-02:
In order to comply with the latest EU regulations, we are experimenting with different Bluetooth pairing methods. In particular, Out of Band technology stood out as being the most convenient for our clients and use case. However, there is little documentation on how to set up this pairing method with QR codes and BlueZ. We are doubtful that it is even possible, although it is allegedly a valid pairing flow:

For instance, ST states the following:

> One method is called Out Of Band (OOB) pairing. QR codes and NFC are two examples of OOB pairing mechanisms [source](https://www.st.com/content/dam/AME/2019/technology-tour-2019/vancouver/presentations/T1S2_Vancouver_SecureBluetoothPairingNFC_J.Barlow.pdf)

[But fails to provide an example of OOB pairing with QR codes.](https://wiki.st.com/stm32mcu/wiki/Connectivity:STM32WB-WBA_BLE_security_-_out_of_band_pairing)

We also found Brabham, Martin and Watson, Myles, "Pairing Bluetooth Devices via QR Code", Technical Disclosure
Commons, (September 13, 2021)
https://www.tdcommons.org/dpubs_series/4589

Additionally, developers alike who have previously failed on this task: https://stackoverflow.com/questions/31027469/bluetooth-secure-simple-pairing-ssp-using-qr-code-as-out-of-band-oob-channel

### What do we know?

At one point there was a D-Bus OOB API. Apparently it was removed from BlueZ a long time ago due to [security concerns](https://github.com/bluez/bluez/issues/984#issuecomment-2422589110) (at least since 2935c04e3d17e9ec1d8f34471af24a0645feb1f5 where tests were removed). However, as stated by https://github.com/bluez/bluez/issues/984 and https://github.com/orgs/bluez/discussions/746, the OOB flow is still available for plugins (neard nfc daemon) and btmgmt.

### What do we want?

To perform OOB pairing/bonding using immutable pre-printed QR codes. Peripheral devices would be running Linux with BlueZ 5 and centrals would be iOS/Android phones.

### How do you plan on archiving this goal?

By leveraging btmgmt or contributing a custom plugin to BlueZ

### Where does BlueZ fit in all of this?

As stated earlier, we'll like to contribute a custom BlueZ plugin for this purpose. Or, at least, a document stating step by step on how to implement this solution using btmgmt. However, before doing any of these, we would like to gather feedback from the community. Is this flow actually discouraged? Is there any technical limitation on iOS/Android that prevents this flow? Is the BlueZ community interested in such contribution?

closed 2025-07-15 by github-actions[bot]

JohnAZoidberg, 2025-09-21:
@Altair-Bueno did you find out anything new?
I'm interested in this as well

Altair-Bueno, 2025-09-21:
We found zero documentation and zero community support. We went with implementing authentication and authorization at the application layer... I suggest you do the same.