bluez/bluez #1060: When Bluetooth started by default starting discovery even Discovering value is no

gadekula1, 2025-01-06:
When BT started by default scanning for devices 
 getting continuous events in **btmon logs**
 
 @ MGMT Event: Device Found (0x0012) plen 45                                                                                                          {0x0002} [hci0] 7652141.947641
        LE Address: 58:4B:6D:AE:70:A4 (Resolvable)
        RSSI: -70 dBm (0xba)
        Flags: 0x00000004
          Not Connectable
        Data length: 31
        Company: Apple, Inc. (76)
          Type: Unknown (7)
          Data: 0124200b998f110005225636c595b2d05c5837a0f166c9a440
 
 
 **bluetoothctl show**
Controller D8:3A:DD:1A:7E:23 (public)
        Manufacturer: 0x0131 (305)
        Version: 0x09 (9)
        Name: BlueZ 5.72
        Alias: BlueZ 5.72
        Class: 0x003c0000 (3932160)
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: yes
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0548
        **Discovering: no**
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name

tried to stop using blutoothctl
but not stopping and getting error
 bluetoothctl scan off
Failed to stop discovery: org.bluez.Error.Failed

**dbus logs**

method call time=1736136478.502259 sender=:1.24 -> destination=org.bluez serial=30 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StopDiscovery
error time=1736136478.502972 sender=:1.7 -> destination=:1.24 error_name=org.bluez.Error.InProgress reply_serial=30

gadekula1, 2025-01-17:
static void ctl_up(uint8_t status, uint16_t length,
					const void *param, void *user_data)
{
	int index = L_PTR_TO_UINT(user_data);
	uint16_t len;
	struct mgmt_cp_set_mesh *mesh;
	uint8_t mesh_ad_types[] = { MESH_AD_TYPE_NETWORK,
				MESH_AD_TYPE_BEACON, MESH_AD_TYPE_PROVISION };

	l_debug("HCI%d is up status: %d", index, status);
	if (status)
		return;

	len = sizeof(struct mgmt_cp_set_mesh) + sizeof(mesh_ad_types);
	mesh = l_malloc(len);

	mesh->enable = 1;
	mesh->window = L_CPU_TO_LE16(0x1000);
	mesh->period = L_CPU_TO_LE16(0x1000);
	mesh->num_ad_types = sizeof(mesh_ad_types);
	memcpy(mesh->ad_types, mesh_ad_types, sizeof(mesh_ad_types));

	pvt->rx_id = mesh_mgmt_register(MGMT_EV_MESH_DEVICE_FOUND,
				MGMT_INDEX_NONE, event_device_found, pvt,
				NULL);
	pvt->tx_id = mesh_mgmt_register(MGMT_EV_MESH_PACKET_CMPLT,
					index, send_cmplt, pvt, NULL);

	**mesh_mgmt_send(MGMT_OP_SET_MESH_RECEIVER, index, len, mesh,
			mesh_up, L_UINT_TO_PTR(index), NULL);**
	l_debug("done %d mesh startup", index);

	l_free(mesh);

	if (pvt->send_idx == MGMT_INDEX_NONE) {
		pvt->send_idx = index;
		if (pvt && pvt->io && pvt->io->ready) {
			pvt->io->ready(pvt->io->user_data, true);
			pvt->io->ready = NULL;
		}
	}
}

**Set Mesh Receiver Command** =========================
	Command Code:		0x0057
	Controller Index:	<controller id>
	Command Parameters:	Enable (1 Octets) Window (2 Octets)
				Period (2 Octets)
				Num AD Types (1 Octets)
				AD Types { }
	This command Enables or Disables Mesh Receiving. When enabled passive scanning remains enabled for this controller.
	The Window/Period values are used to set the Scan Parameters when no other scanning is being done.
	Num AD Types and AD Types parameter, filter Advertising and Scan responses by AD type. Reponses that do not contain at least one of the requested AD types will be ignored. Otherwise they will be delivered with the Mesh Device Found event.
	Possible errors:	Failed No Resources
				Invalid Parameters

above code triggering continuous scan

Note: we are starting bluetooth-mesh.service  along with bluetooth-service, during mesh initialization it's triggering continuous scan, due to that unable to start BT/BLR discovery

gadekula1, 2025-01-22:
> static void ctl_up(uint8_t status, uint16_t length, const void *param, void *user_data) { int index = L_PTR_TO_UINT(user_data); uint16_t len; struct mgmt_cp_set_mesh *mesh; uint8_t mesh_ad_types[] = { MESH_AD_TYPE_NETWORK, MESH_AD_TYPE_BEACON, MESH_AD_TYPE_PROVISION };
> 
> ```
> l_debug("HCI%d is up status: %d", index, status);
> if (status)
> 	return;
> 
> len = sizeof(struct mgmt_cp_set_mesh) + sizeof(mesh_ad_types);
> mesh = l_malloc(len);
> 
> mesh->enable = 1;
> mesh->window = L_CPU_TO_LE16(0x1000);
> mesh->period = L_CPU_TO_LE16(0x1000);
> mesh->num_ad_types = sizeof(mesh_ad_types);
> memcpy(mesh->ad_types, mesh_ad_types, sizeof(mesh_ad_types));
> 
> pvt->rx_id = mesh_mgmt_register(MGMT_EV_MESH_DEVICE_FOUND,
> 			MGMT_INDEX_NONE, event_device_found, pvt,
> 			NULL);
> pvt->tx_id = mesh_mgmt_register(MGMT_EV_MESH_PACKET_CMPLT,
> 				index, send_cmplt, pvt, NULL);
> 
> **mesh_mgmt_send(MGMT_OP_SET_MESH_RECEIVER, index, len, mesh,
> 		mesh_up, L_UINT_TO_PTR(index), NULL);**
> l_debug("done %d mesh startup", index);
> 
> l_free(mesh);
> 
> if (pvt->send_idx == MGMT_INDEX_NONE) {
> 	pvt->send_idx = index;
> 	if (pvt && pvt->io && pvt->io->ready) {
> 		pvt->io->ready(pvt->io->user_data, true);
> 		pvt->io->ready = NULL;
> 	}
> }
> ```
> 
> }
> 
> **Set Mesh Receiver Command** ========================= Command Code: 0x0057 Controller Index: Command Parameters: Enable (1 Octets) Window (2 Octets) Period (2 Octets) Num AD Types (1 Octets) AD Types { } This command Enables or Disables Mesh Receiving. When enabled passive scanning remains enabled for this controller. The Window/Period values are used to set the Scan Parameters when no other scanning is being done. Num AD Types and AD Types parameter, filter Advertising and Scan responses by AD type. Reponses that do not contain at least one of the requested AD types will be ignored. Otherwise they will be delivered with the Mesh Device Found event. Possible errors: Failed No Resources Invalid Parameters
> 
> above code triggering continuous scan
> 
> Note: we are starting bluetooth-mesh.service along with bluetooth-service, during mesh initialization it's triggering continuous scan, due to that unable to start BT/BLR discovery

code allows to share controller between mesh and Bluetooth service
location mesh/mesh-io-mgmt.c -> read_info_cb()

		l_info("Controller hci %u already in use (%x)",
						index, current_settings);

		/* Share this controller with bluetoothd */
		mesh_mgmt_send(MGMT_OP_SET_LE, index,
				sizeof(le), &le,
				ctl_up, L_UINT_TO_PTR(index), NULL);

bgix, 2025-01-22:
The intent is certainly that both classic bluetooth and bluetooth mesh are
usable at the same time.  The scanning logic for both are located in the
kernel and controlled via the mgmt interface.

Basic theory of functionality is that when bluetooth-meshd is part of a
mesh network (and has been enabled), the controller is *always* scanning.
This is usually passive scanning.  When the classic bluetoothd daemon needs
to scan (during discovery) it is allowed to perform more the more robust
scanning (active scanning, and ACL connection for SDP service discovery and
GATT service discovery). As long as some scanning is active, the mesh
daemon should continue to receive advertised mesh packets.

When classic daemon no longer needs to scan/service discover the kernel
based scanning engine should revert to passive scanning (rather than
disabling scanning altogether).

But that was TL;DR -- The existence of mesh passive scanning should *not*
prevent bluetoothd daemon from starting BT Discovery. If it does, there is
something wrong with the MGMT command flow.

gadekula1, 2025-01-23:
to run bluetooth-meshd and bluetoothd in parallel 
need to use bt_hci_new_raw_device

bgix, 2025-01-23:
The intent is for both daemons to share the same controller. Neither daemon is supposed to be using HCI_RAW mode.  Are you using the deprecated tool for mesh?

To run mesh, you should ONLY be using:

bluetooth-meshd  (run as root, with the correct dbus service definitions)
mesh-cfgclient  (baseline "mesh controller" app that may be used as a starting point for C based clients of the daemon)
test-mesh  (python app that may be used as a starting point for Python based clients of the daemon)

There is an old deprecated GATT based config client that used a controller in RAW mode, but it is unsupported, and should not be used because it does require exclusive access to a controller.

gadekula1, 2025-01-24:
> The intent is certainly that both classic bluetooth and bluetooth mesh are
> usable at the same time.  The scanning logic for both are located in the
> kernel and controlled via the mgmt interface. [...]

Thanks @bgix for your reply
we are using systemd to start bluetooth and bluetooth-mesh service and using dbus for communication
During mesh initialization as you mentioned  it's starting passive scanning.
As we started bluetooth daemon also we trigger discovery (using dbus) which is failing
Some analysis I already attached
please let me know if any further information required

closed 2025-07-15 by github-actions[bot]