bluez/bluez #1061: Fails to compile with --disable-avrcp

Dreamsorcerer, 2025-01-06:
With: `./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --disable-a2dp --disable-avrcp --disable-network --disable-manpages`

When compiling I get:
```
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `mcp_accept':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:331: undefined reference to `media_player_controller_create'
/usr/bin/ld: /home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:337: undefined reference to `media_player_set_callbacks'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `cb_media_state':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:189: undefined reference to `media_player_set_status'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `cb_track_position':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:182: undefined reference to `media_player_set_position'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `cb_track_duration':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:173: undefined reference to `media_player_set_metadata'
/usr/bin/ld: /home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:174: undefined reference to `media_player_metadata_changed'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `cb_track_title':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:160: undefined reference to `media_player_set_metadata'
/usr/bin/ld: /home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:161: undefined reference to `media_player_metadata_changed'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `cb_player_name':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:135: undefined reference to `media_player_set_name'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `mcp_disconnect':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:367: undefined reference to `media_player_destroy'
/usr/bin/ld: profiles/audio/bluetoothd-mcp.o: in function `mcp_data_free':
/home/pi/bthidhub/install/on_rpi/bluez/profiles/audio/mcp.c:286: undefined reference to `media_player_destroy'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:6109: src/bluetoothd] Error 1
make: *** [Makefile:4487: all] Error 2
```

I'm using 5.66, but from what I can see of the code, it looks the same on master. It appears that the 'profiles/audio/player.h' file is only included when avrcp is enabled, but then mcp requires it. I'm guessing that mcp should include the file as well, or that mcp should be disabled when avrcp is disabled.

Dreamsorcerer, 2025-01-06:
configure.ac seems to reference a --disable-mcp, but if I add that to the command, then I get:
```
/usr/bin/ld: src/bluetoothd-plugin.o:(.rodata+0x44): undefined reference to `__bluetooth_builtin_mcp'
```
So, seems like I can't disable that either.

Dreamsorcerer, 2025-01-06:
Adding `profiles/audio/player.h profiles/audio/player.c` to the MPC section of Makefile.plugins was enough to compile it.

Vudentz, 2025-01-08:
> Adding `profiles/audio/player.h profiles/audio/player.c` to the MPC section of Makefile.plugins was enough to compile it.

Do you care to send a patch? Or I should go ahead and do it myself?

Dreamsorcerer, 2025-01-08:
Feel free. I wasn't sure that was the correct solution, as I'm unclear why mcp is enabled. I figured that was just a workaround for me to get it working while waiting for the approved solution.

closed 2025-01-10 by BluezTestBot