bluez/bluez #1063: [Bluez v5.66] Cannot add node

cyborgkid0110, 2025-01-07:
I tried to add device with mesh-cfgclient and it works well.
However, when I tried to add device based on `test-mesh` example of Bluez v5.66 with additional 'create' and 'scan' features in `test-mesh` example in v5.54, 
it keep failed and terminal show this:
```
pi@raspberrypi:~/Desktop/test $ python mesh-gateway.py 487d697b7539b8db
!!! Install termcolor module for better experience !!!
Register OnOff Server model on element 0
OnOff Server
State is OFF
Register OnOff Client model on element 1
OnOff Client
Register OnOff Client model on element 1
OnOff Client
*** MAIN MENU ***
token  - set node ID (token)
create  - create mesh network
appkey-index  - set AppKey index
appkey-new  - create appkey
scan-start  - scan for near unprovisioned devs
scan-cancel  - cancel scanning
element-index  - set Element index
add  - add device to mesh network
join  - join mesh network
attach  - attach mesh node
remove  - delete node
uuid  - set remote uuid
vendor-send  - send raw vendor message
config-menu  - Node Configuration menu
client-menu  - On/Off client menu
help  - show list command
quit  - exit the test
attach
Attach mesh node to bluetooth-meshd daemon
Mesh app registered: /org/bluez/mesh/node696a1284418c421ea68a3ace19d0dfcb

scan-start
Scanning...
Scan procedure started
ScanResult RSSI -58 dddd08a6f7a08e4280000000000000dd000000000000

uuid
Enter 32-digit hex remote UUID:
dddd08a6f7a08e4280000000000000dd

add
Adding dev UUID dddd08a6f7a08e4280000000000000dd
AddNode procedure failed  org.bluez.mesh.Error.Failed: Failed to start provisioning initiator
``` 

Log of bluetooth-meshd when add device:
```
mesh/manager.c:add_node_call() AddNode request
mesh/manager.c:scan_cancel()
mesh/agent.c:mesh_agent_refresh() agent owner: :1.107
mesh/agent.c:mesh_agent_refresh() agent path: /mesh/test/agent
Agent failed (org.freedesktop.DBus.Error.UnknownMethod), Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 658, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 248, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: GetAll is not a valid method of interface org.freedesktop.DBus.Properties

mesh/manager.c:add_start() Start callback
```

I notice this: `GetAll is not a valid method of interface org.freedesktop.DBus.Properties`, so I check methods in interface `org.freedesktop.DBus.Properties` in `org.bluez.mesh` using d-feet. I checked when using mesh-cfgclient and test-mesh, and both cases have GetAll method.
So I think the problem may be Agent class in `agent.py` is not implemented correctly, but I have no idea how to fix.

bgix, 2025-01-07:
You can only add devices to a mesh network from a Config Client, and the python based test-mesh app does not contain a Config Client... only a Config Server.

To add one or more nodes to a mesh network, start with a network created from mesh-cfgclient, or any other complient mesh Config Client supporting master node.

To use test-mesh as a node:
1. use the "join" command from test-mesh
2.  scan for the node from a Config Client
3.  when found, add from the Config Client, the UUID that the joining device is advertising.
4. follow the prompts in each app for authentication (by default a u128 random number that is copy-pastable)

To simplify development, you may create any number of member "test-mesh" nodes that have individually been added to a network controlled by mesh-cfgclient.  The nodes will all be able to talk to each other wether they are local (sharing the same bluetooth controller) or remote (talking to the mesh network over bluetooth LE).

We have no python based config client apps, but you could build one if you are willing to also create a config database that can store the required network data, and be read/writable from the python app.

cyborgkid0110, 2025-01-08:
Thank you so much for very helpful information.

So if I build python based config client app, I have to build message handler for config client model, which is similar to OnOff Model in test-mesh right?

bgix, 2025-01-08:
The messaging would be similar... the more complex part would be handling all of the caching and maintenance of the config database. Network and App keys, subscriptions and publications, key refreshes, etc etc etc

These have all been done already in mesh-cfgclient in C, but it could be rewritten in python I suppose. But that wouldn't be included in the BlueZ release.  Both test-mesh and mesh-cfgclient are included as samples and starting points... not consumer products.

closed 2025-07-15 by github-actions[bot]