bluez/bluez #1066: After Bluetooth/PC restart, HSP/HFP mode is available only

keekkenen, 2025-01-08:
linux kernel: 6.6.67-gentoo-gentoo-dist
bluetoothd: 1.22
wireplumber: 0.5.6-r1 (enabled flags: elogind)
pipewire: 1.2.6 (enabled flags: X bluetooth dbus elogind flatpak gstreamer ieee1394 liblc3 lv2 modemmanager readline sound-server ssl v4l zeroconf)

btmon log on pairing: [pairing.log](https://github.com/user-attachments/files/18346486/pairing.log)
btmon log after restarts Bluetooth: [after_restart.log](https://github.com/user-attachments/files/18346502/after_restart.log)
 
I tried to use three headsets but result the same.

Vudentz, 2025-01-08:
> linux kernel: 6.6.67-gentoo-gentoo-dist bluetoothd: 1.22 

Wait, 1.22, it can't be right our version start from 4.0.

bluez: 5.79 wireplumber: 0.5.6-r1 (enabled flags: elogind) pipewire: 1.2.6 (enabled flags: X bluetooth dbus elogind flatpak gstreamer ieee1394 liblc3 lv2 modemmanager readline sound-server ssl v4l zeroconf)
> 
> btmon log on pairing: [pairing.log](https://github.com/user-attachments/files/18346486/pairing.log) btmon log after restarts Bluetooth: [after_restart.log](https://github.com/user-attachments/files/18346502/after_restart.log)
> 
> I tried to use three headsets but result the same.

5.79 seems more likely, anyway I don't see any problem with A2DP:

```
> ACL Data RX: Handle 51 flags 0x02 dlen 6                 #178 [hci0] 2.729774
      Channel: 66 len 2 [PSM 25 mode Basic (0x00)] {chan 2}
      AVDTP: Open (0x06) Response Accept (0x02) type 0x00 label 2 nosp 0
```

Is there anything on bluetoothd logs? Or better yet run btmon with -p debug and bluetoothd with -d so the logs from bluetoothd can be captured by btmon as well.

keekkenen, 2025-01-09:
I had started bluetoothd manually, then made pairing and after all work fine. I made a few times stop and start bluetoothd and it works fine. After I started Bluetooth service throws open-rc again it doesn't work.

pv, 2025-01-11:
Try to get logs that are not truncated to 80 columns.

Usual cause for A2DP connection looking ok in btmon but not appering as sound output, is that you are running multiple sound server instances at the same time, which are trying to control bluetooth devices. In this case the connection can be grabbed by a wrong one that can be different from where HFP goes.

ValeZAA, 2025-03-28:
I do not think it is a bug, if you select HFP device it will after reconnect default to HFP again. This is different from windows, though on Windows I doubt you can force HFP device only unless you change the A2DP driver to alternative one.

closed 2025-09-09 by github-actions[bot]