bluez/bluez #1067: Galaxy Buds Pro 3 lose authentication sync and stop connecting after a variable amount of time when used on multiple devices.

abl030, 2025-01-10:
Hi,

Pipewire Version: 1.2.7
Bluez: 5.79
Kernel: 6.12.8
OS: NixOS

Summary: Galaxy Buds Pro 3 seem to lose key sync over time leading to authentication failures.

Device: Galaxy Buds Pro 3

Steps to Reproduce:

Pair buds in bluetoothctl. Disconnect buds, pair buds with other computers and/or phones. Go back to original computer between each switch and use bluetoothctl to connect, after a seemingly random numbers of tries authentication will fail. This is anywhere from 1 to 8 tries in my testing. Trusting the device in bluetoothctl has no effect. Neither does restarting the daemon.

Actual Results:
`> HCI Event: Command Status (0x0f) plen 4                                        #29 [hci0] 11.316464
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Encryption Change (0x08) plen 4                                     #30 [hci0] 11.654995
        Status: PIN or Key Missing (0x06)
        Handle: 512 Address: 42:0A:5B:08:E9:6F (Resolvable)
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                   #31 [hci0] 11.655068
        Handle: 512 Address: 42:0A:5B:08:E9:6F (Resolvable)
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                        #32 [hci0] 11.655440
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                   #33 [hci0] 11.704118
        Status: Success (0x00)
        Handle: 512 Address: 42:0A:5B:08:E9:6F (Resolvable)
        Reason: Authentication Failure (0x05)
@ MGMT Event: Device Disconnected (0x000c) plen 8                           {0x0001} [hci0] 11.704184
        LE Address: 24:24:B7:58:C6:49 (Samsung Electronics Co.,Ltd)
        Reason: Connection terminated due to authentication failure (0x04)`

Expected Result:
Headphones should connect every time. Windows has no issue connecting on the same hardware. (Note this not a dual boot scenario)

Btmon log file from one connection event:
[blue.log](https://github.com/user-attachments/files/18377603/blue.log)

Thanks for all you do, it must be so annoying with all these bug reports for random devices.

cheers
Andy

Vudentz, 2025-01-10:
> Hi,
>
> Pipewire Version: 1.2.7 Bluez: 5.79 Kernel: 6.12.8 OS: NixOS
>
> Summary: Galaxy Buds Pro 3 seem to lose key sync over time leading to authentication failures.
>
> Device: Galaxy Buds Pro 3
>
> Steps to [...]

Wait, the device is responding to Status: PIN or Key Missing (0x06) and you are saying BlueZ is the one at fault here? Have you consider that Windows is throwing away the keys and repairing without requiring user input and that is actually a security problem?

abl030, 2025-01-10:
I took that one board and did some testing today, resetting the buds to factory settings and confirmed they had lost all pairing data. Then I replicated the above behavior on just two devices, my phone and one laptop. After repairing the buds lost authentication after only 1 more connection to my phone. Surely the buds are not deleting pairing data for two devices. 
I can confirm that my work (windows) computer will happily swap the buds between phone and laptop all day everyday. My gut feel here is there's something non-standard about the Buds 3 Pro in their pairing data. (Typical samsung.....)

abl030, 2025-01-20:
Apologies for wasting your time. After repairing everything and resetting the buds to factory I cannot replicate the issue. You were indeed correct that the headphones must have reached their limit and were throwing away keys.

closed 2025-01-20 by abl030