bluez/bluez #1069: With NVIDIA GPU driver, Bluetooth HSP/HFP profile is missing after suspend/resume

medicalwei, 2025-01-13:
This is an issue migrated from pipewire: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4461

Downstream bug: https://bugs.launchpad.net/ubuntu/+source/pipewire/+bug/2092158

---

Original report:

We found that with NVIDIA GPU, it can affect HSP/HFP profile after suspend/resume. The behavior of suspend/resume in NVIDIA driver (nvidia-sleep.sh) triggers the issue. This is reproducible on Pipewire 2.30.0 and Wireplumber 0.5.6 (Ubuntu 24.10)

For the driver to save the content of VRAM, NVIDIA driver uses the command "chvt 63" to blank the screen and save VRAM, before the system enters suspend, and resume it by "chvt" to previous active tty.

However, when pipewire detected the active session switch, it will also switch the audio context to the active user, and Bluetooth headphones will be disconnected if switching terminal to a text console without active user session.

We have determined the minimal reproducible code (without NVIDIA device/driver) is:

`sudo chvt 4; sleep 5; sudo systemctl suspend; sudo chvt 2`

[Reproducible Steps]

- Connect Bluetooth headphones
- Check that Bluetooth Handsfree is available in Sound settings
- Suspend the system and wake it up

[Expected Behavior]

Bluetooth Handsfree profile should be available after automatic reconnection

[Actual Behavior]

Bluetooth Handsfree profile is missing

---

Forwarding from pipewire developer @pv who traced the logs [pipewire-bug4461-logs.zip](https://github.com/user-attachments/files/18392286/pipewire-bug4461-logs.zip):

> From the bluetoothd debug logs, the cause here appears to be:
>
> 1. chvt
> 1. pipewire: shutdown DBus connection to bluetoothd to unregister all profiles
> 1. bluetoothd/profile.c: cleanup "external" HFP service
> 1. bluetoothd/policy.c: HFP service is removed from policy.c reconnect list
> 1. enter suspend
>
> HFP service is then not reconnected after suspend.
>
> A2DP profile is builtin in bluetoothd, so it is not removed from the reconnect list, which causes > the different behavior.
>
> This is not so easy to fix:
>
> There's no way I can see to allow different users on different seats connect BT devices to their own Pipewire instances, other than have the instances co-operatively close connection to bluetoothd and the SCO socket, when the seat is not active. Then the other user's Pipewire instance can manage BT devices. This is what we do now --- disconnect Pipewire from BT on seat user change.
>
> Some options:
>
> 1. Add logic to bluetoothd/policy.c that always connects HFP is A2DP is getting connected
> 1. Change how bluetoothd/policy.c remembers service reconnect stage (store by UUID, not service instance)
> 1. Add logic to Pipewire that always connects HFP is A2DP is getting connected
>
> I think position of bluetoothd maintainers was that connection policy is job of bluetoothd.
>
> I'd maybe think the option 2. would sound best, but needs fix on bluetoothd side.

Vudentz, 2025-01-13:
Hmm, sounds like we need to cache the service while the profile is offline, that said why is pipewire disconnecting from bluetoothd? Or that is because pipewire instances are run per-session?

We can modify the policy plugin to do some detection if HFP has been connected and it disappear and gets reconnected though, but since I don't have such setup I would expect someone that can reproduce it to go ahead and send a patch rather than just creating another bug to fix a corner case created by nvidia driver or some odd interaction with chvt, which sort of reminds me of people looking for multi-seat support but I don't think it would help if start running bluetoothd per-session, quite the contrary, as it can be experience here having this sort of setup is asking for problems.

pv, 2025-01-14:
> why is pipewire disconnecting from bluetoothd? Or that is because pipewire instances are run per-session?

Yes, there's one pipewire process per user that talks to bluetoothd. They have to coordinate on vt switch who is going to control Bluetooth audio. This is the default setup eg on Fedora.

Vudentz, 2025-01-14:
> > why is pipewire disconnecting from bluetoothd? Or that is because pipewire instances are run per-session?
> 
> Yes, there's one pipewire process per user that talks to bluetoothd. They have to coordinate on vt switch who is going to control Bluetooth audio. This is the default setup eg on Fedora.
> 
> I think the present issue is probably solvable in policy.c, just needs someone who has the time to work on it.

Well then we probably need another rule that detects when HFP becomes available again and then reconnect if A2DP is connected for example.

closed 2025-03-13 by BluezTestBot