bluez/bluez #1070: Segfault problem with blutoothd and mouse canyonMW44

HoTschir, 2025-01-13:
After connection was lost (e.g. by sleep of computer or power off of mouse) shortly (about 3 seconds) after reconnect, blutoothd crashes with segfault (see kernel.log).

I did also a valgrind execution. Command:
  `valgrind -s --leak-check=full --show-leak-kinds=all --track-origins=yes bluetoothd`
Unfortunately I'm unable to analyze the log.

Attached kernel log entry and valgrind log.

bluetoothd version is 5.66
linux kernel is 6.1.0-21-amd64

[kernel.log](https://github.com/user-attachments/files/18397434/kernel.log)
[canyonMW44_blutooth.valgrind.log](https://github.com/user-attachments/files/18397433/canyonMW44_blutooth.valgrind.log)

pv, 2025-01-13:
Try to get debug symbols for /usr/libexec/bluetooth/bluetoothd --- e.g. for Debian https://wiki.debian.org/DebugPackage

Vudentz, 2025-01-13:
@HoTschir 5.66 is sort of old though, could you perhaps try with something more recent?

HoTschir, 2025-01-14:
> @HoTschir 5.66 is sort of old though, could you perhaps try with something more recent?

I'm doing this. 

First attempt: to use debian sid, but it will break my Debian system (can not install a single package without touching any other). 
Second attempt: to compile from source code (done so far) and install it without breaking the systems package handling (ongoing).

Will report about results later.

HoTschir, 2025-01-15:
I installed following files of newest bluez-version (5.79):
```
/usr/bin/bluemoon                 
/usr/bin/bluetoothctl             
/usr/bin/btattach                 
/usr/bin/btmon                    
/usr/bin/hex2hcd                  
/usr/bin/l2ping                   
/usr/bin/l2test                   
/usr/bin/mpris-proxy              
/usr/bin/rctest                   
/usr/libexec/bluetooth/bluetoothd 
/usr/libexec/bluetooth/obexd      
/usr/bin/isotest
```

and restarted process:
`/etc/init.d/bluetooth restart`
After that, I could switch OFF mouse (disconnect seen) and ON (reconnect seen) and there was no segfault anymore. Connection was stable afer that.

But when PC go to sleep an were restored to work, there was no auto reconnect anymore. I had to give:
```
bluetoothctl
 >> scan on
```
and press the reconnect button on mouse.

closed 2025-07-14 by github-actions[bot]