bluez/bluez #1071: bluetoothctl dbus crash

LechDawid, 2025-01-14:
**Platform:** Toradex Verdin iMX8M Plus
**System:** Yocto Project Linux distribution
**BlueZ version:** 5.72

### Issue
Waiting to connect to bluetoothctl...dbus[204064]: arguments to dbus_connection_get_object_path_data() were incorrect, assertion "connection != NULL" failed in file ../../dbus-1.14.8/dbus-connection.c line 5969.
This is normally a bug in some application using the D-Bus library.

[image]

### Description
This issue occurrs when stress-testing Bluetooth connectivity in a gateway type device.
Connecting to multiple peripherals and reading characteristics every minute.
Script to connect is written in python-bleak.
Usually takes 1-2 days for this issue to reliably appear.

After this crash happens, there is no other way to bring back Bluetooth other than hard resetting the board.
Would greatly appreciate any help.

Vudentz, 2025-01-14:
@LechDawid did bluetoothd crash as well or this is just bluetoothctl? And does it crashes at the startup? Perhaps try running with valgrind to have a better backtrace since this appears to be crashing inside libdbus or something.

campeterson41, 2026-10-02:
This still occurs with BlueZ 5.87-2 on Arch Linux (Omarchy).

Trigger: `bluetoothctl show` was executed inside a restricted process sandbox where access to the system D-Bus was denied.

The process aborted with SIGABRT. A symbolized core shows that `bluetoothctl` did not check the failed D-Bus connection before attaching its object manager:

```
#3  _dbus_abort
    dbus/dbus-sysdeps.c:103
#4  _dbus_warn_check_failed
    dbus/dbus-internals.c:329
#5  dbus_connection_get_object_path_data
    connection=0x0, path="/"
    dbus/dbus-connection.c:5974
#6  object_path_ref
    connection=0x0, path="/"
    gdbus/object.c:1340
#7  g_dbus_attach_object_manager
    connection=0x0
    gdbus/object.c:1955
#8  main
    client/main.c:4016
```

The process metadata confirmed it was sandboxed (`NoNewPrivs: 1`, `Seccomp: 2`, PID 4 inside its namespace). The host `bluetoothd` service was healthy, and there was no OOM event.

It appears `g_dbus_setup_bus()` can return NULL when system-bus access fails, after which `client/main.c` calls `g_dbus_attach_object_manager()` without validating the connection. Expected behavior would be a clean error message and nonzero exit rather than SIGABRT.

Prepared by GPT-5 via Codex.