bluez/bluez #1073: Expose "Initiator" as Device Property to capture a device's central or peripheral role at moment of connection?

abqjln, 2025-01-16:
Using simultaneous BLE central and peripheral on the same adapter requires keeping track of whether a device is a central or a peripheral. The Core Spec says this role is defined at the moment of connection, but I cannot find where this is made accessible using an API.

The challenge is in determining whether a newly-appeared or connected/changed device is initiating the connection.  Understanding that Simultaneous Central and Peripheral is still experimental, it turns out that this is the missing piece in my application. Currently I can hard-code known centrals and all works. Heuristic methods of determining centrals, e.g., having no uuids, do not work if you are connecting to device that is also simultaneously a central and peripheral, such as an phone or fitness device.

A possible solution is contained in Bit 3 "Initiated Connection" of the flags variable in the Device Connected Event in mgmt-api.txt.

Is Bit 3 the right one? Would it be feasible to have a new Device Property of "Initiator" in the dbus API device.rst, similar to "Paired" or "Bonded"?

Issues #1060 , #562 , and #914 would all seem to benefit from the addition if I am reading them correctly.

I note that this flag's behavior was recently fixed [https://patches.linaro.org/project/linux-bluetooth/patch/20230911223341.2701182-1-luiz.dentz@gmail.com/](url)

Thanks.

Vudentz, 2025-01-16:
@abqjln well we don't have a object representing a connection and yes with things like dual-mode you may be central and peripheral at same time in different bearers.

Also central doesn't always means GATT client or peripheral GATT server, profiles, for the most part, don't care who was the central or peripheral and sometimes they will invert the roles or even have move devices acting as client and server, so Im not really sure what you are after but chances are there is no silver bullet, oh and lets not forget that in case of BR/EDR one can change roles.

abqjln, 2025-01-18:
@Vudentz ..I should have searched more before asking. Closing this to join #975--a more clearly stated request.

closed 2025-07-14 by github-actions[bot]