bluez/bluez #1074: While A2DP/SNK/SET/BV-05-C in PTS is running , the bluetoothd crashed by SEGV.

takeshi-dodo, 2025-01-17:
This happened during running the A2DP/SNK/SET/BV-05-C in PTS.
After the request timeout of AVDTP: Get Capabilities, the bluetoothd crashed by SEGV. 

■ version
bluez 5.72

■ procedure
[linux] Autostart bluetooth.service and bluealsa.service (refer systemctl_cat_service.txt)
[linux] Input bluealsa-aplay in console (speakers can play music)
[linux] Input bluetoothctl in console
[pts] Start A2DP/SNK/SET/BV-05-C
[linux] Input yes for auth in bluetoothctl
[linux] Listen the music that the pts plays through the speakers for a few seconds
[linux] bluetoothd crashed.

■ logs
[gdb.txt](https://github.com/user-attachments/files/18447304/gdb.txt)
[btmon.txt](https://github.com/user-attachments/files/18447306/btmon.txt)
[bluetootctl.txt](https://github.com/user-attachments/files/18447303/bluetootctl.txt)
[varlog.txt](https://github.com/user-attachments/files/18447307/varlog.txt)
[strace.txt](https://github.com/user-attachments/files/18447302/strace.txt)
[pts.txt](https://github.com/user-attachments/files/18447305/pts.txt)
[systemctl_cat_service.txt](https://github.com/user-attachments/files/18447393/systemctl_cat_service.txt)

Vudentz, 2025-01-17:
@takeshi-dodo can you try reproducing with 5.79? Also it is probably a good idea to run with valgrind to get a better backtrace.

Btw, if I understand this properly you are not using pulseaudio or pipewire, but instead something called bluealsa? This is sort of new to me, while it is not a problem to have other audio daemon this is quite a unique setup that is not tested as much as the other ones which perhaps contributes to the crash.

takeshi-dodo, 2025-01-20:
@Vudentz 

can you try reproducing with 5.79?
> There is not change in 5.79.

Btw, if I understand this properly you are not using pulseaudio or pipewire, but instead something called bluealsa? 
> Yes, I used bluealsa. But pipewire is also running.
    How to make a2dp work using only pipewire?

takeshi-dodo, 2025-01-20:
I could enable a2dp by following 2 files.
But the bluetoothd crashed after the request timeout of AVDTP: Get Capabilities.

```
root@XXX:~# cat /usr/share/wireplumber/wireplumber.conf.d/80-disable-logind.conf
wireplumber.profiles = {
  main = {
    monitor.bluez.seat-monitoring = disabled
  }
}
```

```
root@XXX:~# cat /usr/share/wireplumber/bluetooth.conf
monitor.bluez.properties = {
  bluez5.roles = [ a2dp_sink a2dp_source ]
  bluez5.codecs = [ sbc ]
}
```

Vudentz, 2025-01-22:
> [@Vudentz](https://github.com/Vudentz)
> 
> can you try reproducing with 5.79?
> 
> > There is not change in 5.79.
> 
> Btw, if I understand this properly you are not using pulseaudio or pipewire, but instead something called bluealsa?
> 
> > Yes, I used bluealsa. But pipewire is also running.
> > How to make a2dp work using only pipewire and wireplumber, without bluealsa?

Weird, why would you be using both? I mean pipewire already include A2DP endpoints so you shouldn't really need bluealsa.

closed 2025-07-14 by github-actions[bot]