bluez/bluez #1078: regression: mouses are stuttering

bam80, 2025-01-24:
The mouse movements are stuttered and happen with noticeable delay.

The issue is reproducible ~100% with different machines (so different BT adapters) on two mouses I have:
- Asus ROG Pugio II
- Lenovo Legion M600
both are BLE.

_Workaround for **Asus**_ - disconnect Bluetooth link with the mouse (by clicking the mouse in Gnome BT setting at upper right corner). Any mouse movement forces the reconnect then, and there is no issue after, until the next PC sleep.
! _The workaround doesn't work for the Lenovo_ !

This started to happening after Fedora upgrade to 40 (Bluez 5.73) or 41 (Bluez 5.79 currenlty).
I tried to narrow down the version, here is the results:
- on [Ubuntu 24.04 LTS](https://pkgs.org/search/?q=bluez#) Live ISO (Bluez 5.72), there is _no issue_
- on Ubuntu 24.10 (Bluez 5.77), there _is an issue_

So **Bluez 5.73 might be the guilty version**.
I'll try to make sure by running 5.72 vs 5.73 when I have a chance.

Logs do not seem to contain anything interesting, but let me know if you need them.

@Vudentz

Vudentz, 2025-01-24:
Please add the btmon logs, without them it would be hard to get where this could be failing.

bam80, 2025-01-27:
The symptoms are probably these, but I didn't have a chance to take `btmon` logs when it stutters really badly:
```
янв 27 20:42:41 old gnome-shell[1381]: libinput error: event17 - Legion M600 Mouse: client bug: event processing lagging behind by 89ms, your system is too slow

Vudentz, 2025-01-27:
> The symptoms are probably these, but I didn't have a chance to take `btmon` logs when it stutters really badly:
> 
> ```
> янв 27 20:42:41 old gnome-shell[1381]: libinput error: event17 - Legion M600 Mouse: client bug: event processing lagging behind by 89ms, your system is too slow
> ```

Well that would be up in the stack then, not really bluetooth related.

bam80, 2025-01-27:
That still might be two independent problems, as my system had bulk background load (with low priority though) at the time.
I'll continue investigate this.

On other system (Ubuntu Live ISO), the problem manifested as soon as I opened Gnome Bluetooth settings there, so BT scanning started. There was no any load, but I didn't look for `libinput error`s there.

bam80, 2025-04-26:
Please don't close

bam80, 2025-11-20:
I checked again and indeed, as stated above it's seems Bluez regression since 5.72.
I think I should be able to bisect it further.
The issue is actual.