bluez/bluez #1079: mouse doesn't reconnect if PC was awakened by BT keyboard

bam80, 2025-01-24:
The issue might be related with #1078, see Bluez versions there.

To reproduce:
- BLE devices connected: keyboard and mouse(s) (I have 2 of them here, it doesn't matter which one is used)
- The PC must be configured to be awake-able with Bluetooth (I have a custom udev rule inspired by Arch Wiki)
- Put PC to suspend to RAM
- Wake PC by pressing any key on the keyboard
- Wait a bit. Then try to move a mouse.
- Notice the keyboard reconnects and works, but mouse(s) don't reconnect and can't be connected forcibly.

Workaround:
Try to move the mouse (it won't work meanwhile), at the same time, run `btmgmt find` command.
The discovering drops with the mouse found, and the mouse immediately reconnects and works then normally:
```
$ sudo btmgmt find
[sudo] password for bam: 
hci0 type 7 discovering on
Discovery started
hci0 dev_found: 0C:EC:80:E6:4A:C3 type LE Public rssi -62 flags 0x0000 
AD flags 0x05 
eir_len 13
hci0 type 7 discovering off
bam@old:~
```

btmon trace in btsnoop format reproducing the problem (with the workaround applied).
To read:
`$ btmon -r` [btmon_no-mouse.log](https://github.com/user-attachments/files/18542545/btmon_no-mouse.log)

bam80, 2025-01-27:
Any chance to look at this?
It really makes BT mouses unusable for any unprepared user..

Vudentz, 2025-01-27:
> > Any chance to look at this? It really makes BT mouses unusable for any unprepared user.. [@Vudentz](https://github.com/Vudentz)
> 
> Looks like everything is proper but I couldn't confirm that device has been added to white/accept list:
> 
> ```
> < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                               #306 [hci0] 16.278524
>         Type: Passive (0x00)
>         Interval: 1280.000 msec (0x0800)
>         Window: 11.250 msec (0x0012)
>         Own address type: Public (0x00)
>         Filter policy: Ignore not in accept list (0x01)
> ```
> 
> That said it is disabled after resume:
> 
> ```
> < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                                                                                                                  #327 [hci0] 34.522252
>         Scanning: Disabled (0x00)
>         Filter duplicates: Disabled (0x00)
> = bluetoothd: Controller resume with wake event 0x1                                                                                                                                                                                                                                 34.526866
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                             #328 [hci0] 34.534848
>       LE Set Scan Enable (0x08|0x000c) ncmd 1
>         Status: Success (0x00)
> ```
> 
> So perhaps there is a bug if there are more devices in the accept list that shall not disable the scan.

Perhaps this is actually the culprit:

```
commit 68253f3cd715e819bc4bff2b0e6b21234e259d56
Author: Zhengping Jiang <jiangzp@google.com>
Date:   Mon Jul 11 17:05:30 2022 -0700

    Bluetooth: hci_sync: Fix resuming scan after suspend resume
    
    After resuming, remove setting scanning_paused to false, because it is
    checked and set to false in hci_resume_scan_sync. Also move setting
    the value to false before updating passive scan, because the value is
    used when resuming passive scan.
    
    Fixes: 3b42055388c30 (Bluetooth: hci_sync: Fix attempting to suspend with
    unfiltered passive scan)
    
    Signed-off-by: Zhengping Jiang <jiangzp@google.com>
    Reviewed-by: Abhishek Pandit-Subedi <abhishekpandit@chromium.org>
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

```

So without this the hci_resume_scan_sync would do nothing thus causing the scan to be disabled.

bam80, 2025-01-27:
Thanks for looking Luiz! You are very helpful as always :)

So I guess from my side reverting that patch and retest is needed, I'll do as I get a chance.
Thank you!

Vudentz, 2025-01-27:
> Thanks for looking Luiz! You are very helpful as always :)
> 
> So I guess from my side reverting that patch and retest is needed, I'll do as I get a chance. Thank you!

I wouldn't revert 3b42055388c30 but instead apply the above patch if that is missing in your kernel.

bam80, 2025-01-27:
Hmm, I see this patch in kernel's [ChangeLog-5.18.18](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.18.18), does it mean I already have it in my Linux 6.12.9-200.fc41.x86_64?
I added my system info in the OP.

I also have to add an important detail:
_the issue doesn't manifest with BT Classic keyboard, only with BLE one._

I can add the logs if needed.

Vudentz, 2025-01-27:
> Hmm, I see this patch in kernel's [ChangeLog-5.18.18](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.18.18), does it mean I already have it in my Linux 6.12.9-200.fc41.x86_64? I added my [...]

I think the first stable release was v6.0, so yeah if you are using 6.12 that should contain it already, so perhaps something different is going on, try enabling debugfs for net/bluetooth/hci_sync.c:

https://www.kernel.org/doc/html/v4.11/admin-guide/dynamic-debug-howto.html

Vudentz, 2025-01-27:
> Looks like everything is proper but I couldn't confirm that device has been added to white/accept list:
>
> ```
> < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen [...]

Actually I missed the fact the it is enabled again on frame 336:

```
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                                                                                                              #336 [hci0] 37.434938
        Type: Passive (0x00)
        Interval: 60.000 msec (0x0060)
        Window: 30.000 msec (0x0030)
        Own address type: Public (0x00)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                             #337 [hci0] 37.436773
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
```

And it is not even filtering reports since the policy is set to 0, so this really sounds like a controller bug or something, later it changes to active on frame 398:

```
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                                                                                                              #398 [hci0] 75.770374
        Type: Active (0x01)
        Interval: 22.500 msec (0x0024)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                             #399 [hci0] 75.772179
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
```

The only different is we switch the own address type to random since it is active, but afaik with filter policy being 0x00 we shall see all advertisements including those not for our own address thus I would assume it to be a bug if that is what happening here.

bam80, 2025-01-27:
I can try debugfs things tomorrow,
meantime, I added another important details, please see the logs in the OP:

> UPDATE:
>    the order in which devices reconnect seems important:

Vudentz, 2025-01-27:
Ok, checking what the remote device is doing it doesn't seem it is using directed advertisements:
```

> HCI Event: LE Meta Event (0x3e) plen 25                                                                                                                                                                                                                               #404 [hci0] 75.831255
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 0C:EC:80:E6:4A:C3 (Texas Instruments)
        Data length: 13
        Flags: 0x05
          LE Limited Discoverable Mode
          BR/EDR Not Supported
        Appearance: Mouse (0x03c2)
        16-bit Service UUIDs (partial): 2 entries
          Human Interface Device (0x1812)
          Battery Service (0x180f)
        RSSI: -62 dBm (0xc2)
```

Which means this shall have been reported with passive scanning as well, which further reinforces my assumption that the controller is not sending it to the host thus why it doesn't reconnect.

Vudentz, 2025-01-27:
> I can try debugfs things tomorrow, meantime, I added another important details, please see the logs in the OP:
> 
> > UPDATE:
> > the order in which devices reconnect seems important:

What controller are you using?

bam80, 2025-01-27:
BT controller where I got the problem:
> Bus 001 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

I have another PC with Intel BT controller, can test tomorrow there.

Vudentz, 2025-01-27:
> BT controller where I got the problem:
> 
> > Bus 001 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

Hmm, these CSR ones sometimes are clones that don't behave very well.

> I have another PC with Intel BT controller, can test tomorrow there.

Yeah, try to see if you can reproduce with that as well.

bam80, 2025-01-28:
So I tried the same mouse/keyboard on another PC: Ubuntu 24.10 Live ISO (Bluez 5.77 vs 5.79 here), Intel BT adapter.
The results are controversial.

While I couldn't reproduce the issue with the exact steps above, it happened in other circumstances.
Overall, user experience was even worse:
The devices lose connection unexpectedly, once disconnected, often they can't reconnect any more (keyboard included), mouse is stuttering periodically (this one is reproducible instantly).
I would say it was unusable mess..

From the good news,
Ubuntu 24.04 LTS Live ISO (Bluez 5.72) on the same PC worked flawlessly with the same devices - no disconnections, no stuttering, no reconnect problems.

So we clearly have severe regression somewhere.
Next time when I have a chance, I'll try to narrow the differences between good and bad states, to find the guilty component exactly.

Vudentz, 2025-01-28:
> So I tried the same mouse/keyboard on another PC: Ubuntu 24.10 Live ISO (Bluez 5.77 vs 5.79 here), Intel BT adapter. The results are controversial.
>
> While I couldn't reproduce the issue with the [...]

For disconnection I doubt bluetoothd/userspace has anything to do with that, so you really need to check what are the kernel versions.

> So we clearly have severe regression somewhere. Next time when I have a chance, I'll try to narrow the differences between good and bad states, to find the guilty component exactly.

Yeah, but didn't you find out there is something not quite right with libinput in the other bug?

bam80, 2025-01-28:
> Yeah, but didn't you find out there is something not quite right with libinput in the other bug?

Sorry, I was so blown away I forgot to look at the logs, maybe next time.

I intended to install Linux on that PC anyway (so no ISO needed), then I will be able to swap components one by one.
Also, on my current PC (Fedora), I think the easiest would be to downgrade Bluez package and retest.

I'll keep you informed here. Also, if you have particular request, please let me know.
Thank you for the cooperation Luiz!

bam80, 2025-01-28:
> Yeah, but didn't you find out there is something not quite right with libinput in the other bug?

Why I think `libinput` has nothing to do here:
the stuttering seem somehow depends on the other BT devices connected or visible, and also BT scanning dialog can sometimes provoke it (almost always on the other PC). I can't explain such behavior if libinput would be the reason here..

Vudentz, 2025-01-28:
> Why I think `libinput` has nothing to do here: the stuttering seem somehow depends on the other BT devices connected or visible, and also BT scanning dialog can sometimes provoke it (almost always [...]

Yeah, active scanning while connected is known to cause issues and many, if not all, controllers, not sure if there is a recommended way to deal with that since we might need to tune down the scanning window, etc, which will likely find devices a lot more slowly, also if you have multiple connection + scanning they all will contribute to drive latency up.

bam80, 2025-01-28:
Luiz, could you analyze that:

> UPDATE:
So it's happened to be Bluez regression indeed, v5.72 doesn't have the issue:
[btmon_keyboard-then-mouse_bluez-5.72_OK.log](https://github.com/user-attachments/files/18578472/btmon_keyboard-then-mouse_bluez-5.72_OK.log)

Vudentz, 2025-01-28:
These are the changes to input plugin for 5.72..5.73, these should only make a difference for classic though:

```
commit ee880bee85864b0e691370197c9de8d2f7ba0af2
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 13:50:00 2024 -0500

    input/device: Don't destroy UHID device on disconnect
    
    This makes classic HID behave like HoG which keeps the UHID device
    around while disconnected so it doesn't have to be recreated on every
    reconnection.

commit 9698870015b0972d3d93f30a6d9de1ed029ba87c
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 13:47:22 2024 -0500

    input.conf: Make UserspaceHID defaults to true
    
    This makes UserspaceHID defaults to true so the plugin has more control
    over the input device lifetime.

commit 247ae8524888268bb075ff90a3d77a9b4449d08f
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 09:55:24 2024 -0500

    hog-lib: Don't destroy UHID device on detach
    
    This makes bt_hog_detach not to destroy UHID device which means the
    
    This makes UserspaceHID defaults to true so the plugin has more control
    over the input device lifetime.

commit 247ae8524888268bb075ff90a3d77a9b4449d08f
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 09:55:24 2024 -0500

    hog-lib: Don't destroy UHID device on detach
    
    This makes bt_hog_detach not to destroy UHID device which means the
    device node don't need to be recreated in case of reconnections which
    speeds up the process.
    
    Fixes: https://github.com/bluez/bluez/issues/737

```

So it really beats what is going on here that you have such a change in behavior by just changing the daemon version.

Vudentz, 2025-01-28:
> Luiz, could you analyze that:
> 
> > UPDATE:
> > So it's happened to be Bluez regression indeed, v5.72 doesn't have the issue:
> > [btmon_keyboard-then-mouse_bluez-5.72_OK.log](https://github.com/user-attachments/files/18578472/btmon_keyboard-then-mouse_bluez-5.72_OK.log)

There is mix of active and passive scanning in this logs, do you always attempt to reconnect with bluetooth settings open or something? Anyway I don't think this is the same behavior we have seem with 5.79 where active scanning is only used after to confirm the device can be found using it but not with passive scanning.

bam80, 2025-01-28:
I'll test again but I don't remember I reconnected with scanning dialog opened. I might opened it after, though.

While for 5.79 I run `btmgr find` as workaround to have the reconstruction.

bam80, 2025-01-28:
> There is mix of active and passive scanning in this logs, do you always attempt to reconnect with bluetooth settings open or something?

You are right, the Bluetooth Settings were opened in the background.
I retested without, and on a second test couldn't reconnect again.

The 5.72 apparently completely fixed the stuttering issue though (which confirms my earlier observations on the other PC).

Vudentz, 2025-01-28:
> You are right, the Bluetooth Settings were opened in the background. I retested without, and on a second test couldn't reconnect again.
>
> The 5.72 apparently completely fixed the stuttering issue [...]

Both connection parameters seem about the same though, so Im not sure what could be possibly affecting the events.

Vudentz, 2025-01-28:
Perhaps this has something to do with https://github.com/bluez/bluez/commit/1782bfd791f76317a3f3ded2884606c1b5b164eb, maybe the replay logic is being used or something which causes the device to be behave badly, the default is not to persist devices anymore on 5.79 since we had way too many bugs with that, but perhaps you have UserspaceHID=persist?

bam80, 2025-01-28:
That's right, I have `UserspaceHID=persist` with 5.79 and on 5.77 it's probably implicit, so might be a reason.
I'll retest it on PC with Intel BT, where the stuttering effect is more exaggerated and easier to achieve.

bam80, 2025-01-29:
So the story continues.
`UserspaceHID=persist` has nothing to do with the issues (they occurs even without).
Next, I tried to run 5.72 on third PC (by stopping systemd service and run the built daemon manually),
and _the issues continued even then!_

So it either might be somehow not always legitimate to swap the `bluetoothd` version on the go, or the problem is with something else, for example in kernel or it's bt part.

That said, I still couldn't reproduce the any single problem on Ubuntu 24.04 LTS Live ISO (Bluez 5.72).

Vudentz, 2025-01-29:
> So the story continues. `UserspaceHID=persist` has nothing to do with the issues (they occur even without). Next, I tried to run 5.72 on third PC (by stopping systemd service and run the built daemon manually), and _the issues continued even then!_
> 
> So it either might be somehow not always legitimate to swap the `bluetoothd` version on the go, or the problem is with something else, for example in kernel or it's bt part.
> 
> That said, I still couldn't reproduce the any single problem on Ubuntu 24.04 LTS Live ISO (Bluez 5.72).

This is what Ive been telling you since the beginning, the daemon doesn't control the HCI flow, it is the kernel, so I bet you have been testing different versions and it so happen that Ubuntu Live image works, but as soon as you try to update anything the kernel might be updated as well.

bam80, 2025-01-29:
So Bluez 5.72 or whatever contains some kernel BT part as well as userspace one?

Vudentz, 2025-01-29:
> So Bluez 5.72 or whatever contains some kernel BT part as well as userspace one?

No it doesn't, but perhaps the packages contain dependencies on newer kernel or something, at least that is my impression since you said that if you build it from source it doesn't work either.

bam80, 2025-04-26:
Let's keep it open

bam80, 2025-09-08:
Please reopen

bam80, 2025-10-02:
I will, hopefully im a few days. Thanks.

berenddeschouwer, 2025-10-06:
Updated my installation to 5.84 + this patch and the next commit.

I've also added a debug statement to device_get_privacy() to see if it does return different values, and indeed it does on restart.

bam80, 2025-11-15:
@Vudentz Unfortunately, my mouse can't reconnect today.
I tried several times, but if keyboard awakes the PC first, mouse can't reconnect anymore. I need to do `btmgmt find` to make it work again.
But now PC can be awaken by mouse button or movement. Then it works.

Vudentz, 2025-11-17:
> The problem I faced with the patch above is that if my BT remote connected, the host now wakes up immediately after suspend entering. BT mouse doesn't produce such behavior (at least, not the stable one).
> 
> PS: now I actually see both that behavior and mouse wake up even without the patch, so I'm not sure what role the patch actually plays for me (if any).

What kernel version is that? Also it may help to add some more logs to check what has changed, btw what controller do you have in your system?

bam80, 2025-11-17:
@Vudentz 
```console
$ uname -a
Linux old 6.17.7-300.fc43.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Nov  2 15:30:09 UTC 2025 x86_64 GNU/Linux
```
It probably make sense to track the problem with BT remote immediately wakeup separately, with the new bunch of logs collected, etc.

Let's please concentrate here for the original issue, which I provided logs already for, IIRC.
If you need some more logs for the original problem, please let me know. It's very annoying.

First of all, I would appreciate if we reopen the bug.

bam80, 2025-12-11:
> Just include a new log, this time from the start

@Vudentz Here it is.

CSR dongle, kernel 6.4.0, bluez 8.5:
[btmon_keyboard-then-mouse_FAIL2.log](https://github.com/user-attachments/files/24106807/btmon_keyboard-then-mouse_FAIL2.log)
[btmon_btmgmt_find-mouser-recover.log](https://github.com/user-attachments/files/24106808/btmon_btmgmt_find-mouser-recover.log)

What I did:
1. Stopped bluez, started btmon, started bluez.
2. Pressed BT keyboard buttons, moved the mouse.
3. Suspended the system. It awaken immediately (other bug? or just something wakened it up), so I suspended again.
4. Awaken the system with keyboard.
5. Issued the password, got my GUI session
6. Tried to move the mouse - no evidence it's alive, no even packets in btmon seen.
7. Issued `sudo btmgmt find` to recovery the mouse (see the second log above).
8. Waited it finished, then moved the mouse and it reconnected this time.

@Vudentz Please let me know if that's enough.

bravhek, 2026-09-07:
Adding another data point for current BlueZ reconnect behavior.

On Fedora 44 with BlueZ 5.87-6.fc44, I am seeing delayed/unreliable automatic reconnect of a Logitech M720 BLE mouse after boot.

The device remains paired/bonded/trusted.

The interesting part is that the BLE connection itself is not slow:

    LE Create Connection -> LE Connection Complete: successful within milliseconds

and:

    bluetoothctl connect <M720-MAC>

completes in approximately 34 ms.

The failure appears later in the BlueZ HOG/UHID path, where I repeatedly get:

    bt_uhid_input: Invalid argument (22)
    No matching connection for device

and repeated recreation of the M720 HID/UHID device.

So this appears to be another case where BLE link establishment works but automatic HID reconnect/state handling does not complete cleanly.