bluez/bluez #1080: Invalid input MTU size for some of A2DP transports

pchelkin91, 2025-01-26:
Hi,

Regarding commit https://github.com/bluez/bluez/commit/ae5be371a9f53fed33d2b34748a95a5498fd4b77 and its kernel part 4b6e228e297b ("Bluetooth: Auto tune if input MTU is set to 0") - this unfortunately doesn't work on both sides:

1. Bluez L2CAP socket options setting code doesn't currently actually push the special "0" value down to the kernel because it treats a zero imtu as an absence of this parameter and the needed code paths are not triggered.
2. Kernel side currently denies setting zero imtu via setsockopt() anyway because it doesn't pass the check inside l2cap_valid_mtu().

I've got a setup to test this working with one of specific A2DP transports and prepared a couple of patches for Bluez and the kernel. They fix the problem for me but TBH I'm not totally convenient about the Bluez ones, so would like to discuss the approach in the issue at first.

I'll attach the patches below in plain. My question concerns the A2DP-server Bluez patch №2. There is no good way to check for BT_PHY at that point - as done in https://github.com/bluez/bluez/commit/ae5be371a9f53fed33d2b34748a95a5498fd4b77. Also, Bluez oopses in some another place in SDP code if the current kernel patch is not applied yet. So, the kernel patch is some sort of a prerequisite for the Bluez ones.

I can't imagine a scenario where auto-tuning the imtu for the server A2DP socket would lead to some unexpected results but still would be glad to know what you think on all this.

Bluez patch №1
```
From 0aec65f5f9c54a8ed2b36565b66ba937d86d3ae5 Mon Sep 17 00:00:00 2001
From: Fedor Pchelkin <pchelkin@ispras.ru>
Date: Sun, 26 Jan 2025 18:47:24 +0300
Subject: [PATCH 1/2] audio: actually try to enable MTU auto-tuning

A "0" for the input MTU passed to the underlying socket is supposed to
indicate that its value should be determined by the L2CAP layer. However,
the current code treats a zero imtu just as if there is nothing to change.

Introduce an additional flag to indicate that the zero imtu is
explicitly requested by the caller for the purpose of auto-tuning.
Otherwise, the similar behavior remains.

Fixes: ae5be371a9f5 ("avdtp: Enable MTU auto tunning")
---
 btio/btio.c | 23 +++++++++++++----------
 1 file changed, 13 insertions(+), 10 deletions(-)

diff --git a/btio/btio.c b/btio/btio.c
index 2d277e409..74a4003b6 100644
--- a/btio/btio.c
+++ b/btio/btio.c
@@ -66,6 +66,7 @@ struct set_opts {
 	uint16_t imtu;
 	uint16_t omtu;
 	int central;
+	uint8_t auto_mtu;
 	uint8_t mode;
 	int flushable;
 	uint32_t priority;
@@ -610,7 +611,7 @@ static uint8_t mode_l2mode(uint8_t mode)
 }
 
 static gboolean set_l2opts(int sock, uint16_t imtu, uint16_t omtu,
-						uint8_t mode, GError **err)
+				uint8_t auto_mtu, uint8_t mode, GError **err)
 {
 	struct l2cap_options l2o;
 	socklen_t len;
@@ -622,7 +623,7 @@ static gboolean set_l2opts(int sock, uint16_t imtu, uint16_t omtu,
 		return FALSE;
 	}
 
-	if (imtu)
+	if (imtu || auto_mtu)
 		l2o.imtu = imtu;
 	if (omtu)
 		l2o.omtu = omtu;
@@ -666,17 +667,17 @@ static gboolean set_le_mode(int sock, uint8_t mode, GError **err)
 }
 
 static gboolean l2cap_set(int sock, uint8_t src_type, int sec_level,
-				uint16_t imtu, uint16_t omtu, uint8_t mode,
-				int central, int flushable, uint32_t priority,
-				GError **err)
+				uint16_t imtu, uint16_t omtu, uint8_t auto_mtu,
+				uint8_t mode, int central, int flushable,
+				uint32_t priority, GError **err)
 {
-	if (imtu || omtu || mode) {
+	if (imtu || omtu || auto_mtu || mode) {
 		gboolean ret = FALSE;
 
 		if (src_type == BDADDR_BREDR)
-			ret = set_l2opts(sock, imtu, omtu, mode, err);
+			ret = set_l2opts(sock, imtu, omtu, auto_mtu, mode, err);
 		else {
-			if (imtu)
+			if (imtu || auto_mtu)
 				ret = set_le_imtu(sock, imtu, err);
 
 			if (ret && mode)
@@ -986,6 +987,8 @@ static gboolean parse_set_opts(struct set_opts *opts, GError **err,
 			opts->imtu = va_arg(args, int);
 			if (!opts->mtu)
 				opts->mtu = opts->imtu;
+			if (!opts->imtu)
+				opts->auto_mtu = 1;
 			break;
 		case BT_IO_OPT_CENTRAL:
 			opts->central = va_arg(args, gboolean);
@@ -1890,7 +1893,7 @@ gboolean bt_io_set(GIOChannel *io, GError **err, BtIOOption opt1, ...)
 	switch (type) {
 	case BT_IO_L2CAP:
 		return l2cap_set(sock, opts.src_type, opts.sec_level, opts.imtu,
-					opts.omtu, opts.mode, opts.central,
+					opts.omtu, opts.auto_mtu, opts.mode, opts.central,
 					opts.flushable, opts.priority, err);
 	case BT_IO_RFCOMM:
 		return rfcomm_set(sock, opts.sec_level, opts.central, err);
@@ -1941,7 +1944,7 @@ static GIOChannel *create_io(gboolean server, struct set_opts *opts,
 				server ? opts->psm : 0, opts->cid, err) < 0)
 			goto failed;
 		if (!l2cap_set(sock, opts->src_type, opts->sec_level,
-				opts->imtu, opts->omtu, opts->mode,
+				opts->imtu, opts->omtu, opts->auto_mtu, opts->mode,
 				opts->central, opts->flushable, opts->priority,
 				err))
 			goto failed;
-- 
2.39.5
```

Bluez patch №2
```
From f121799224be79834522af824ff964400bc9cd39 Mon Sep 17 00:00:00 2001
From: Fedor Pchelkin <pchelkin@ispras.ru>
Date: Sun, 26 Jan 2025 18:48:02 +0300
Subject: [PATCH 2/2] a2dp: enable input MTU auto-tuning for the server

---
 profiles/audio/a2dp.c | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/profiles/audio/a2dp.c b/profiles/audio/a2dp.c
index c97bd6e89..05a2e2e80 100644
--- a/profiles/audio/a2dp.c
+++ b/profiles/audio/a2dp.c
@@ -2592,6 +2592,8 @@ static bool a2dp_server_listen(struct a2dp_server *server)
 				BT_IO_OPT_MODE, mode,
 				BT_IO_OPT_SEC_LEVEL, BT_IO_SEC_MEDIUM,
 				BT_IO_OPT_CENTRAL, true,
+				/* Set Input MTU to 0 to auto-tune */
+				BT_IO_OPT_IMTU, 0,
 				BT_IO_OPT_INVALID);
 	if (server->io)
 		return true;
-- 
2.39.5

```

Kernel patch
```
From c68b495937dacaf4cae64d474bf5b4c582a29d9b Mon Sep 17 00:00:00 2001
From: Fedor Pchelkin <pchelkin@ispras.ru>
Date: Mon, 16 Dec 2024 23:09:00 +0300
Subject: [PATCH] Bluetooth: L2CAP: accept zero as a special value for MTU
 auto-selection

One of the possible ways to enable the input MTU auto-selection for L2CAP
connections is supposed to be through passing a special "0" value for it
as a socket option. Commit [1] added one of those into avdtp. However, it
simply wouldn't work because the kernel still treats the specified value
as invalid and denies the setting attempt. Recorded BlueZ logs include the
following: 

  bluetoothd[496]: profiles/audio/avdtp.c:l2cap_connect() setsockopt(L2CAP_OPTIONS): Invalid argument (22)

[1]: https://github.com/bluez/bluez/commit/ae5be371a9f53fed33d2b34748a95a5498fd4b77

Found by Linux Verification Center (linuxtesting.org).

Fixes: 4b6e228e297b ("Bluetooth: Auto tune if input MTU is set to 0")

---
 net/bluetooth/l2cap_sock.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/net/bluetooth/l2cap_sock.c b/net/bluetooth/l2cap_sock.c
index 49f97d4138ea..46ea0bee2259 100644
--- a/net/bluetooth/l2cap_sock.c
+++ b/net/bluetooth/l2cap_sock.c
@@ -710,12 +710,12 @@ static bool l2cap_valid_mtu(struct l2cap_chan *chan, u16 mtu)
 {
 	switch (chan->scid) {
 	case L2CAP_CID_ATT:
-		if (mtu < L2CAP_LE_MIN_MTU)
+		if (mtu && mtu < L2CAP_LE_MIN_MTU)
 			return false;
 		break;
 
 	default:
-		if (mtu < L2CAP_DEFAULT_MIN_MTU)
+		if (mtu && mtu < L2CAP_DEFAULT_MIN_MTU)
 			return false;
 	}
 
-- 
2.39.5
```

Thanks!

Vudentz, 2025-01-27:
> Hi,
> 
> Regarding commit [ae5be37](https://github.com/bluez/bluez/commit/ae5be371a9f53fed33d2b34748a95a5498fd4b77) and its kernel part 4b6e228e297b ("Bluetooth: Auto tune if input MTU is set to 0") - this unfortunately doesn't work on both sides resulting in a loss of L2CAP frames with some exotic A2DP transports usually coming up with weird custom MTU sizes:
> 
>     1. Bluez L2CAP socket options setting code doesn't currently actually push the special "0" value down to the kernel because it treats a zero imtu as an absence of this parameter and the needed code paths are not triggered.
> 
>     2. Kernel side currently denies setting zero imtu via setsockopt() anyway because it doesn't pass the check inside l2cap_valid_mtu().
> 
> 
> I've got a setup to test this working with one of specific A2DP transports and prepared a couple of patches for Bluez and the kernel. They fix the problem for me but TBH I'm not totally convinced about the Bluez ones, so would like to discuss the approach in the issue at first.
> 
> I'll attach the patches below in plain. My question concerns the A2DP-server Bluez patch №2. There is no good way to check for BT_PHY at that point - as done in [ae5be37](https://github.com/bluez/bluez/commit/ae5be371a9f53fed33d2b34748a95a5498fd4b77). Also, Bluez oopses in some another place in SDP code if the current kernel patch is not applied yet. So, the kernel patch is some sort of a prerequisite for the Bluez ones.
> 
> I can't imagine a scenario where auto-tuning the imtu for the server A2DP socket would lead to some unexpected results but still would be glad to know what you think on all this.
> 
> Bluez patch №1
> 
> ```
> From 0aec65f5f9c54a8ed2b36565b66ba937d86d3ae5 Mon Sep 17 00:00:00 2001
> From: Fedor Pchelkin <pchelkin@ispras.ru>
> Date: Sun, 26 Jan 2025 18:47:24 +0300
> Subject: [PATCH 1/2] audio: actually try to enable MTU auto-tuning
> 
> A "0" for the input MTU passed to the underlying socket is supposed to
> indicate that its value should be determined by the L2CAP layer. However,
> the current code treats a zero imtu just as if there is nothing to change.
> 
> Introduce an additional flag to indicate that the zero imtu is
> explicitly requested by the caller for the purpose of auto-tuning.
> Otherwise, the similar behavior remains.
> 
> Fixes: ae5be371a9f5 ("avdtp: Enable MTU auto tunning")
> ---
>  btio/btio.c | 23 +++++++++++++----------
>  1 file changed, 13 insertions(+), 10 deletions(-)
> 
> diff --git a/btio/btio.c b/btio/btio.c
> index 2d277e409..74a4003b6 100644
> --- a/btio/btio.c
> +++ b/btio/btio.c
> @@ -66,6 +66,7 @@ struct set_opts {
>  	uint16_t imtu;
>  	uint16_t omtu;
>  	int central;
> +	uint8_t auto_mtu;
>  	uint8_t mode;
>  	int flushable;
>  	uint32_t priority;
> @@ -610,7 +611,7 @@ static uint8_t mode_l2mode(uint8_t mode)
>  }
>  
>  static gboolean set_l2opts(int sock, uint16_t imtu, uint16_t omtu,
> -						uint8_t mode, GError **err)
> +				uint8_t auto_mtu, uint8_t mode, GError **err)
>  {
>  	struct l2cap_options l2o;
>  	socklen_t len;
> @@ -622,7 +623,7 @@ static gboolean set_l2opts(int sock, uint16_t imtu, uint16_t omtu,
>  		return FALSE;
>  	}
>  
> -	if (imtu)
> +	if (imtu || auto_mtu)
>  		l2o.imtu = imtu;
>  	if (omtu)
>  		l2o.omtu = omtu;
> @@ -666,17 +667,17 @@ static gboolean set_le_mode(int sock, uint8_t mode, GError **err)
>  }
>  
>  static gboolean l2cap_set(int sock, uint8_t src_type, int sec_level,
[... 90 lines cut ...]
> 
> diff --git a/net/bluetooth/l2cap_sock.c b/net/bluetooth/l2cap_sock.c
> index 49f97d4138ea..46ea0bee2259 100644
> --- a/net/bluetooth/l2cap_sock.c
> +++ b/net/bluetooth/l2cap_sock.c
> @@ -710,12 +710,12 @@ static bool l2cap_valid_mtu(struct l2cap_chan *chan, u16 mtu)
>  {
>  	switch (chan->scid) {
>  	case L2CAP_CID_ATT:
> -		if (mtu < L2CAP_LE_MIN_MTU)
> +		if (mtu && mtu < L2CAP_LE_MIN_MTU)
>  			return false;
>  		break;
>  
>  	default:
> -		if (mtu < L2CAP_DEFAULT_MIN_MTU)
> +		if (mtu && mtu < L2CAP_DEFAULT_MIN_MTU)
>  			return false;
>  	}
>  
> -- 
> 2.39.5
> ```

Ack, that said I would probably recommend we do add tests to l2cap-tester this time to ensure CI test the auto-tune logic, both with L2CAP_OPTIONS and BT_RCVMTU.

Vudentz, 2025-01-28:
@pchelkin91 do you have plan to upstream these changes? Id like to have it fixed asap.

pchelkin91, 2025-01-28:
> [@pchelkin91](https://github.com/pchelkin91) do you have plan to upstream these changes? Id like to have it fixed asap.

Yes, I'm currently addressing BT_RCVMTU vs BREDR improvement in the Bluez patch and will send it out to mailing lists shortly.

pchelkin91, 2025-01-28:
> Hmm, Looks like we didn't test it properly and were expecting it work with BT_RCVMTU but BREDR doesn't use that, that said Ive change this a little bit to always use BT_RCVMTU if auto is selected, that way it should works with existing kernels.

I've been trying to address this issue but would it work for existing kernels?

If we do the following, i.e. do setsockopt(BT_RCVMTU) for BREDR src_type:
```
 static gboolean l2cap_set(int sock, uint8_t src_type, int sec_level,
-				uint16_t imtu, uint16_t omtu, uint8_t mode,
-				int central, int flushable, uint32_t priority,
-				GError **err)
+				uint16_t imtu, uint16_t omtu, uint8_t auto_mtu,
+				uint8_t mode, int central, int flushable,
+				uint32_t priority, GError **err)
 {
-	if (imtu || omtu || mode) {
+	if (imtu || omtu || auto_mtu || mode) {
 		gboolean ret = FALSE;
 
-		if (src_type == BDADDR_BREDR)
+		if (src_type == BDADDR_BREDR) {
+			/* Input MTU auto-tuning is expected to work with BT_RCVMTU.
+			 * BREDR doesn't use that.
+			 */
+			if (auto_mtu) {
+				ret = set_le_imtu(sock, imtu, err);
+				if (!ret)
+					return ret;
+			}
 			ret = set_l2opts(sock, imtu, omtu, mode, err);
-		else {
-			if (imtu)
+		} else {
+			if (imtu || auto_mtu)
 				ret = set_le_imtu(sock, imtu, err);

```

 the kernel rejects it due to the incorrect type:

```
static int l2cap_sock_setsockopt(struct socket *sock, int level, int optname,
				 sockptr_t optval, unsigned int optlen)
{
...
	case BT_RCVMTU:
		if (!bdaddr_type_is_le(chan->src_type)) {
			err = -EINVAL;
			break;
		}
...
}
```

But maybe I've misunderstood your suggestion... So should I do any modifications for the patches originally published here at https://github.com/bluez/bluez/issues/1080#issue-2811769482 or better send them as is for further discussion at mailing lists?

Vudentz, 2025-01-28:
> > Hmm, Looks like we didn't test it properly and were expecting it work with BT_RCVMTU but BREDR doesn't use that, that said Ive change this a little bit to always use BT_RCVMTU if auto is selected, that way it should works with existing kernels.
> 
> I've been trying to address this issue but would it work for existing kernels?
> 
> If we do the following, i.e. do setsockopt(BT_RCVMTU) for BREDR src_type:
> 
> ```
>  static gboolean l2cap_set(int sock, uint8_t src_type, int sec_level,
> -				uint16_t imtu, uint16_t omtu, uint8_t mode,
> -				int central, int flushable, uint32_t priority,
> -				GError **err)
> +				uint16_t imtu, uint16_t omtu, uint8_t auto_mtu,
> +				uint8_t mode, int central, int flushable,
> +				uint32_t priority, GError **err)
>  {
> -	if (imtu || omtu || mode) {
> +	if (imtu || omtu || auto_mtu || mode) {
>  		gboolean ret = FALSE;
>  
> -		if (src_type == BDADDR_BREDR)
> +		if (src_type == BDADDR_BREDR) {
> +			/* Input MTU auto-tuning is expected to work with BT_RCVMTU.
> +			 * BREDR doesn't use that.
> +			 */
> +			if (auto_mtu) {
> +				ret = set_le_imtu(sock, imtu, err);
> +				if (!ret)
> +					return ret;
> +			}
>  			ret = set_l2opts(sock, imtu, omtu, mode, err);
> -		else {
> -			if (imtu)
> +		} else {
> +			if (imtu || auto_mtu)
>  				ret = set_le_imtu(sock, imtu, err);
> ```
> 
> the kernel rejects it due to the incorrect type:
> 
> ```
> static int l2cap_sock_setsockopt(struct socket *sock, int level, int optname,
> 				 sockptr_t optval, unsigned int optlen)
> {
> ...
> 	case BT_RCVMTU:
> 		if (!bdaddr_type_is_le(chan->src_type)) {
> 			err = -EINVAL;
> 			break;
> 		}
> ...
> }
> ```
> 
> But maybe I've misunderstood your suggestion... So should I do any modifications for the patches originally published here at [#1080 (comment)](https://github.com/bluez/bluez/issues/1080#issue-2811769482) or better send them as is for further discussion at mailing lists?

Nevermind then, it looks like I missed the fact that we check for LE address in order to use BT_RCVMTU, so we will need to patch the kernel no matter what, thus it is important to have the Fixes tag so it is automatically pickup for backport.