bluez/bluez #1081: GATT Confirmation for indication is not received in Bluez 5.78

maheshtalewad, 2025-01-27:
Hi, 

While testing gatt in bluez 5.78, we come to know that confirmation for indication is not received in bluez[user space]. But received in btmon log.
We tried with bluez 5.72, confirmation for indication received in both bluez and btmon.

Bluez 5.78 logs:
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/shared/gatt-server.c:write_complete_cb() Write Complete: err 0
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/gatt-database.c:acquire_notify_reply() AcquireNotify success: fd 26 MTU 517
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/gatt-database.c:**send_notification_to_device() GATT server sending indication**
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/shared/att.c:can_read_data() (chan 0xaaaae4e8c470) ATT PDU received: 0x12
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/shared/gatt-server.c:write_cb() Write Req - handle: 0x006a
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/shared/gatt-server.c:write_complete_cb() Write Complete: err 0
bluetoothd[685]: /usr/src/debug/bluez5/5.78/src/shared/att.c:can_read_data() (chan 0xaaaae4e8c470) ATT PDU received: 0x12

Bluez 5.72 logs:
bluetoothd[608]: /usr/src/debug/bluez5/5.72/src/shared/gatt-server.c:write_complete_cb() Write Complete: err 0
bluetoothd[608]: /usr/src/debug/bluez5/5.72/src/gatt-database.c:acquire_notify_reply() AcquireNotify success: fd 26 MTU 517
bluetoothd[608]: /usr/src/debug/bluez5/5.72/src/gatt-database.c:**send_notification_to_device() GATT server sending indication**
bluetoothd[608]: /usr/src/debug/bluez5/5.72/src/gatt-database.c:**conf_cb() GATT server received confirmation**

On comparing bluez 5.78 and 5.72, we come to know that in func sock_io_read() [src\gatt-database.c], in 5.78,  , gatt_notify_cb() func is called and in 5.72, send_notification_to_devices() is called.  In send_notification_to_devices(), conf_cb is registered to receive confirmation. But in  gatt_notify_cb() there is no sucn function registered for receiving confirmation.
In  conf_cb(), g_dbus_proxy_method_call is called for "Confirm"

[image]

Is this expected behavior? Kindly let us know.

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2025-01-28:
That actually seem intentional, sock_io_read means the server is using AcquireNotify to pass notifications and in that case D-Bus Confirm shall not be called:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.GattCharacteristic.rst#fd-uint16-acquirenotifydict-options-optional

If we want to send back confirmation then we need to write back to the fd not make a D-Bus method call since the whole point of fd passing is to avoid D-Bus traffic.

Vudentz, 2025-01-28:
@maheshtalewad please test with the above changes, it will likely require changes to application side implementing the GATT Characteristic so it listen to incoming confirmation via fd.

maheshtalewad, 2025-02-14:
Hi Luiz,
Thank you for sharing the patch. Could you please help to share any example template/reference that uses socket fd for testing GATT confirmation events. It seems most of the examples is based on DBUS transport.
 
If we have an example template, we can refer and can modify the application.

Thanks and regards,
Mahesh Vithal Talewad