bluez/bluez #1082: Mesh provisioning cannot see unprovisioned devices

pmontesShsConsultores, 2025-01-28:
Hello all, I am using Bluez 5.79 on two Raspberry Pis, one as ConfigServer and other as ConfigClient. 

## Brief

Reboot both devices. Both bluetoothd and bluetooth-meshd services are disabled.

Server: two terminals
```
sudo /usr/libexec/bluetooth/bluetooth-meshd -ndb
python3 test-mesh # join
```

Client: two terminals
```
sudo /usr/libexec/bluetooth/bluetooth-meshd -ndb
mesh-cfgclient
[mesh-cfgclient]# discover-unprovisioned on 15
Unprovisioned scan started
[mesh-cfgclient]# list-unprovisioned
Unprovisioned devices:
```
The unprovisioned scan callback success/failure does not arrive. I expected something similar to the final log shown in #625.

## Environment

- Two Config Clients, a Bluez Rpi and a [Nordic Semi nRF Mesh](https://www.nordicsemi.com/Products/Development-tools/nRF-Mesh) on a mobile device
- Two Config Servers, a second Rpi and a Zephyr OS - Based device

The mobile device does not see unprovisioned RPi running test-mesh.py join.
A RPi running mesh-cfgclient does not find the Zephyr device nor the other mesh server (test-mesh.py join).

Probably I am mixing PB-ADV (BlueZ) and PB-GATT (nRF Mesh) here, but I expected `mesh-cfgclient discover-unprovisioned on` to find unprovisioned Zephyr, or at least the Raspberry ConfigServer. I also tried setting a random token `0123456789abcdef` on test-mesh before join with no difference.

## Logs

### CfgServer daemon
```
# Rpi Server
$ sudo /usr/libexec/bluetooth/bluetooth-meshd -ndb
[DBUS] ...
Loading node configuration from /var/lib/bluetooth/mesh
../mesh/mesh-mgmt.c:mesh_mgmt_list() send read index_list
../mesh/mesh.c:mesh_init() io 0x55556fa76720
../mesh/mesh-mgmt.c:read_index_list_cb() Number of controllers: 1
../mesh/mesh-mgmt.c:read_info_cb() hci 0 status 0x00
../mesh/mesh-mgmt.c:read_info_cb() settings: supp 0001feff curr 00000ac1
../mesh/mesh-mgmt.c:features_cb() Status: 0, Length: 4
up:1 pwr: 1 mesh: 1
../mesh/mesh-io-mgmt.c:read_info_cb() hci 0 status 0x00
Controller hci 0 not in use
../mesh/mesh-io-mgmt.c:ctl_up() HCI0 is up status: 0
../mesh/mesh-io-mgmt.c:ctl_up() done 0 mesh startup
mesh_ready_callback
Added Network Interface on /org/bluez/mesh
[DBUS] ...
../mesh/mesh-io-mgmt.c:mesh_up() HCI0 Mesh up status: 0
# Upon calling python3 test-mesh
[DBUS]
# join
../mesh/mesh.c:join_network_call() Join network request
../mesh/node.c:node_join()
[DBUS]
../mesh/node.c:get_app_properties() path /example
../mesh/node.c:get_element_properties() path /example/ele00
../mesh/node.c:get_element_properties() path /example/ele01
../mesh/mesh-config-json.c:mesh_config_create() New node config /var/lib/bluetooth/mesh/.../node.json
[DBUS NameOwnerChanged]
```

I assume that all `status 0` here mean "OK". If the process was started earlier (even if killed with Ctrl+C), hci shows `Controller hci 0 already in use (ac1)` or `(281)`. From the code I see it is supposed to be OK if bluetoothd was also in use, but it is not my case. Calling join from server
[... 208 lines cut ...]
and Client
` $ uname -a
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux `

I checked the relevant kernel modules were available. On the RPi Client I loaded some kernel modules explicitly; on RPi Server I checked they were available and didn't care to load them myself. I assume the Bluez building/executing process would have failed if there was something wrong there, and `af_alg` is loaded (however, mesh-cfgclient fails silently on this device!?). 

I started using Bluez 5.79, downgraded to 5.66 and returned to 5.79.

## More info

Perhaps this is not related to device discovery but to how the daemon starts passive scanning: this happens only on Raspberry Pi Client. Whenever the bluetooth-meshd service is started, some logs appear on `dmesg`

```
[   13.477084] Bluetooth: hci0: Opcode 0x2005 failed: -16
[   23.467550] Bluetooth: hci0: Opcode 0x2005 failed: -16
[   33.473306] Bluetooth: hci0: Opcode 0x2005 failed: -16
```

which I interpret as if there is more than one instance trying to get the controller for scanning.

I'd be glad to know 

- if the passive scan process that is yielding the hci controller and it is not shown in `ps au` is expected. This implies "controller in use" when the daemon has started once previously
- if I missed any configuration other than enabling D-Bus for bluetooth-mesh
- if calls cancel TX from provisioner are expected when unprovisioned devices are present

pmontesShsConsultores, 2025-02-03:
```
mesh-cfgclient fails silently on this device!?
```

Found that mesh-cfgclient failed on a fresh RPi with OS Lite because there was no .config folder, so mkdir $HOME/.config/mesh-cfg failed on setup_cfg_storage. Running `mkdir $HOME/.config` solves that.

tulsivanol, 2025-04-17:
@pmontesShsConsultores So after creating directory, you was able to detected unprovisioned device in pi? I am stuck with that  and I can't detect. Can you pls help me?

pmontesShsConsultores, 2025-04-21:
@tulsivanol Not yet. Creating the directory helped me run mesh-cfgclient on a fresh Pi, but unprovisioned ADV packets still were not discovered. I have not solved this issue so far.

tulsivanol, 2025-04-21:
@pmontesShsConsultores I am able to run provisioner on 1st pi and unprovisioned node on 2nd pi and I am also able to discover it. But I am facing issues while trying to provision it.

pmontesShsConsultores, 2025-04-21:
@tulsivanol In my case the unprovisioned node scan always returned an empty list: at most I discovered packets at BLE level. I will pull any recent changes and take another look at it again.

pmontesShsConsultores, 2025-04-24:
@tulsivanol Using 5.82 now my CfgClient sees unprovisioned device packets, even though it is not provisioning them. Pi CfgServer does not broadcast advertising packets for me, however.

I use two CfgSrv (Pi + Microcontroller), two CfgCli (another Pi + smartphone).

- smartphone may detect and provision microcontroller
- now CfgCli Pi may detect microcontroller, but fails to provision
- CfgSrv Pi is supposed to send ADV packets but they are not detected by CfgCli Pi nor smartphone

I have tried to provision from mesh-cfgclient using UUID both upper and lower case. The smartphone may provision without OOB, I assume Bluez provisioning defaults to No OOB and yet fails. The microcontroller shows nothing when trying to provision.

How are you running the Configuration Server on Pi? Using

```shell
(T1) sudo /usr/libexec/bluetooth/bluetooth-meshd -ndb

(T2) python3 $BLUEZ_DIR/test/test-mesh
join
```
says `Joining with UUID <uuid>, join procedure started` but no TX packet is seen in daemon. The daemon shows that the Join network request signal is sent through DBus and it does create a node variable, etc. What do you do in addition to `join`?

tulsivanol, 2025-04-24:
@pmontesShsConsultores yes I am doing the same thing. My 1st PI acts as provisioner using `mesh-cfgclient` and I am using `test-mesh` in my 2nd PI. It gives me `uuid` with which it starts request to get provisioned and my 1st PI discovers it. 

But whenever I am trying to provision, it is throwing me `timeout error`. I tried several versions of bluez starting from `5.49` but no luck so far.

pmontesShsConsultores, 2025-04-24:
@tulsivanol In my case the 2nd Pi does not even seem to send packets, but I would suggest you to look at the 1st Pi daemon debug logs to see if the provision request does something apart from receiving the DBus message, and the 2nd Pi logs to see if it receives any packet from the first.

tulsivanol, 2025-04-24:
@pmontesShsConsultores Sure, I'll try to analyze logs and I'll also post it over here in case it will help you.

pmontesShsConsultores, 2025-05-02:
> [@pmontesShsConsultores](https://github.com/pmontesShsConsultores) Just to let you know. I am able to provision and add keys to the node running on my PI. But I am not able to bind keys to the model(specific vendor)

Are you using Bluez also as provisioner?

tulsivanol, 2025-05-02:
@pmontesShsConsultores Yes using bluez. Everything works smoothly except sending messages from one node to another.

closed 2025-09-08 by github-actions[bot]