bluez/bluez #1086: Google TV Remote Controls built-in mic is not detected

bam80, 2025-02-04:
There are a lot of Remotes on the market following [Google TV Remote Controls Reference Design](https://www.twelectronics.co.uk/_files/ugd/6c263b_1f300f5de2614c54b1123f05e3f7fe5b.pdf).
All (most) of them have built-in mic.

Unfortunately, I didn't find a way to make use of the mic in linux.
Any advise?

My [Ugoos BT Remote Control UR02](https://ugoos.com/ugoos-bt-remote-control-ur02) clone has following info:
```
[UR02]# info A4:C1:38:6A:9E:37 
Device A4:C1:38:6A:9E:37 (public)
	Name: UR02
	Alias: UR02
	Appearance: 0x0180 (384)
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: no
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (00001920-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
	UUID: Vendor specific           (ab5e0001-5a21-4f05-bc7d-af01f617b664)
	Modalias: bluetooth:v0508p1980d0000
	ManufacturerData.Key: 0x005d (93)
	ManufacturerData.Value:
  55 66                                            Uf              
	AdvertisingFlags:
  05                                               .               
	Battery Percentage: 0x51 (81)
```
Please let me know if I can provide more info.
Thank you.

@Vudentz

bam80, 2025-02-08:
The remote's mic follows Google Voice over BLE spec (seems not available publicly).

Some parts of the spec can still be found in different sources:
- https://download.csdn.net/download/swlyy1986/61771336 (needs registration to get a full copy)
- [Ohsung C314 Remote Controller User Manual](https://manuals.plus/ohsung/c314-remote-controller-manual)
  - [(the same manual in PDF, better formatting)](https://manuals.plus/m/cd5120266e8d287c2f91ba3ac68a2ded8e4baaa5c5d8b451d05c3d7682b68f9f_optim.pdf)
-  [IR/BLE Remote Contorl of Humax D’live, technical specification](https://fcc.report/FCC-ID/TX4RC89/3947468.pdf)

From the spec:
> <h3>GATT profile UUIDS for ATV Voice Service</h3> 
> <table>
>   <thead>
>     <tr>
>       <th >Type</th>
>       <th >Short-form</th>
>       <th >UUID</th>
>       <th >Properties</th>
>     </tr>
>   </thead>
>   <tbody>
>     <tr>
>       <td>ATV Voice Service</th>
>       <td>ATVV_SERVICE_UUID</td>
>       <td align="center">AB5E0001-5A21-4F05-BC7D-AF01F617B664</td>
> <td></td>
>     </tr>
>     <tr>
>       <td>Write Characteristic</th>
>       <td >ATVV_CHAR_TX</td>
>       <td align="center">AB5E0002-5A21-4F05-BC7D-AF01F617B664</td>
>       <td align="center">Write without Response</td>
>     </tr>
>     <tr>
>       <td>Read Characteristic</td>
>       <td >ATVV_CHAR_AUDIO</td>
>       <td align="center">AB5E0003-5A21-4F05-BC7D-AF01F617B664</td>
>       <td align="center">Notify</td>
>     </tr>
>     <tr>
>       <td>Control Characteristic</td>
>       <td >ATVV_CHAR_CTL</td>
>       <td align="center">AB5E0004-5A21-4F05-BC7D-AF01F617B664</td>
>       <td align="center">Notify</td>
>     </tr>    
>   </tbody>
> </table>

As we see, the ATVV_SERVICE_UUID doesn't get recognized:
> 	$ btmgmt info <RCU's MAC>
...
UUID: Vendor specific           (ab5e0001-5a21-4f05-bc7d-af01f617b664)

That may be part of the reason why we don't see the service's Characteristics advertised in the output (Write/Read/Control).

bam80, 2025-02-10:
Even before it all was standardized, there were Android TV devices on the market bundled with voice remote controls.
Among them are [ADT-1 development kit](https://en.wikipedia.org/wiki/Android_TV#History), [Nexus Player](https://en.wikipedia.org/wiki/Nexus_Player), [Nvidia Shield TV](https://en.wikipedia.org/wiki/Nvidia_Shield_TV).

[Here how they were implemented](https://stackoverflow.com/a/38608283/1063363):
> So for audio streaming, either you:
> - use a custom L2CAP channel or a custom GATT service, this requires a custom android service app and/or modifications to Bluedroid to handle those, it will need a way to inject audio stream as ALSA, most probably with a "loop" audio device driver,
> - declare audio as custom HID reports, this way, Bluedroid injects them back to the kernel, then add a custom HID driver that processes these reports and exposes an audio device.

> Audio over BLE is not standard, so all implementations do not do the actual same thing.
_In Nexus Player case, implementation uses HID: It streams an ADPCM audio stream, chunked in HID reports. There is a special HID driver ["hid-atv-remote.c"](https://android.googlesource.com/kernel/x86_64/+/f2aa02f9e6f019f7cee08b6b06d4636b210ffe36/drivers/hid/hid-atv-remote.c) in Android linux kernel that exposes an ALSA device in addition to input device._ Bluedroid has no information about audio, all it does is forwarding HID reports from BLE to UHID.

For Nvidia Shield TV, remote's Android kernel mic audio implementation is here:
https://github.com/LineageOS/android_kernel_nvidia_nvidia/blob/lineage-22.1/drivers/hid/hid-atv-jarvis.c
It was later partially ported upstream to Linux kernel, for Game Controller only, which doesn't have mic functionality:
https://github.com/torvalds/linux/blob/master/drivers/hid/hid-nvidia-shield.c

Before-mentioned initial ADT-1/Nexus Player mic driver apparently already does all the heavy-lifting for us:
https://github.com/LineageOS/android_kernel_nvidia_shield/blob/lineage-18.1/drivers/hid/hid-atv-remote.c

All what seems needed is to port it to main Linux and adjust for the Google Voice over BLE spec.
That hopefully shouldn't require too monumental amount of work as current remotes basically remained the same, they just became more standardized by following the Google spec. That said, audio packets format might changed a bit, needs to be clarified.

Other's more competent participants analysis of the situation is welcome.
@Vudentz

bam80, 2025-04-25:
I wouldn't close it, at least it contains important data I gathered..

grgergo1, 2025-08-26:
I found a copy of the spec here on GitHub: [Google Voice over BLE spec 1.0](https://wangefan.github.io/linux_kernel_driver/resources/Google_Voice_over_BLE_spec_v1.0.pdf)