bluez/bluez #1088: CIS desychronization with Intel AX210

pv, 2025-02-06:
(Not BlueZ issue, but recording here so that it's written down:)

Since Intel AX210 firmware updates in ~April-May 2024 the BAP desync issue in https://github.com/bluez/bluez/issues/515 for the most part was resolved. However, left/right audio desynchronization, by apparently 1 ISO packet, still sometimes occurs.

Setup:
- Sony Linkbuds S -- this is TWS device, so left/right earbuds appear as CSIP device set
- Intel AX210
- Firmware Version: 91-33.24, Firmware timestamp 2024.33 buildtype 1 build 81755, Firmware SHA1: 0xd028ffe4 (and earlier versions too)
- Latest BlueZ & Pipewire
- Kernel 6.12.11 (but also other versions)
- Set BAP source only mode in Pipewire (Linkbuds S doesn't do 48kHz duplex), `echo "monitor.bluez.properties = { bluez5.roles = [ bap_source ] }" > ~/.config/wireplumber/wireplumber.conf.d/bap-sink.conf`, `systemctl restart --user wireplumber`

Reproducer (random, occurs maybe ~1/20 of tries):
- Start playback (`pw-play sample.flac`, eg this one https://github.com/bluez/bluez/files/11423692/ax210-samsung-gbuds2pro-tws-desync-3-input.flac.log)
- Connect Linkbuds S
- Set default audio output to the Linkbuds S device (if not already so)
- If "lucky", the left/right audio channels are desynchronized by some ms, probably one ISO packet

Stopping playback for a few seconds so that the CIS are torn down and resuming results to normal audio.

btmon dump (persistent desync since playback start): [2025-02-06-sonylinkbudss-desync-2.btsnoop.log](https://github.com/user-attachments/files/18695499/2025-02-06-sonylinkbudss-desync-2.btsnoop.log)

Vudentz, 2025-02-19:
> (Not BlueZ issue, but recording here so that it's written down:)
> 
> Since Intel AX210 firmware updates in ~April-May 2024 the BAP desync issue in [#515](https://github.com/bluez/bluez/issues/515) for the most part was resolved. However, left/right audio desynchronization, by apparently 1 ISO packet, still sometimes occurs.
> 
> Setup:
> 
>     * Sony Linkbuds S -- this is TWS device, so left/right earbuds appear as CSIP device set
> 
>     * Intel AX210
> 
>     * Firmware Version: 91-33.24, Firmware timestamp 2024.33 buildtype 1 build 81755, Firmware SHA1: 0xd028ffe4 (and earlier versions too)
> 
>     * Latest BlueZ & Pipewire
> 
>     * Kernel 6.12.11 (but also other versions)
> 
>     * Set BAP source only mode in Pipewire (Linkbuds S doesn't do 48kHz duplex), `echo "monitor.bluez.properties = { bluez5.roles = [ bap_source ] }" > ~/.config/wireplumber/wireplumber.conf.d/bap-sink.conf`, `systemctl restart --user wireplumber`
> 
> 
> Reproducer (random, occurs maybe ~1/20 of tries):
> 
>     * Start playback (`pw-play sample.flac`, eg this one https://github.com/bluez/bluez/files/11423692/ax210-samsung-gbuds2pro-tws-desync-3-input.flac.log)
> 
>     * Connect Linkbuds S
> 
>     * Set default audio output to the Linkbuds S device (if not already so)
> 
>     * If "lucky", the left/right audio channels are desynchronized by some ms, probably one ISO packet
> 
> 
> Stopping playback for a few seconds so that the CIS are torn down and resuming results to normal audio.
> 
> btmon dump: [2025-02-06-sonylinkbudss-desync-2.btsnoop.log](https://github.com/user-attachments/files/18695499/2025-02-06-sonylinkbudss-desync-2.btsnoop.log)
> 
> What was heard in the earbuds was persistent desync from playback start spanning whole btmon dump. This means clearly audible time difference between left/right earbud streams so no stereo image (and the blip sound separates into two). 7.5ms time difference between streams is plausible by ear, but didn't verify via recording.
> 
> Decoding the LC3 payload, with `btsnoop-lc3dec ../2025-02-06-sonylinkbudss-desync-2.btsnoop 1380` with [this script](https://gitlab.freedesktop.org/pvir/btutil/-/blob/934cda83b1090fcf834abe70e361fef4e694071a/btsnoop-lc3dec) looks like the LC3 data sent is well in sync:
> 
> [image]
> 
> The packet sequence numbers assigned by kernel also seem to stay in sync between the streams.
> 
> The Number of Completed Packets patterns in the dump look a bit suspicious, as the latencies vary:
> 
> ```
> < ISO Data TX: Handle 2305 flags 0x02 dlen 121                                                                 #9228 [hci1] 28.975216
> < ISO Data TX: Handle 2304 flags 0x02 dlen 121                                                                 #9229 [hci1] 28.975221
> > HCI Event: Number of Completed Packets (0x13) plen 9                                                         #9230 [hci1] 28.979353
>         Num handles: 2
>         Handle: 2304 Address: F8:4E:17:CB:C2:83 (Sony Corporation)
>         Count: 2
>         #9225: len 121 (88 Kb/s)
>         Latency: 11 msec (2-11 msec ~7 msec)
>         #9229: len 121 (242 Kb/s)
>         Latency: 4 msec (2-11 msec ~5 msec)
>         Handle: 2305 Address: C9:72:16:D5:2C:FC (Static)
>         Count: 1
>         #9228: len 121 (242 Kb/s)
>         Latency: 4 msec (2-10 msec ~3 msec)
> < ISO Data TX: Handle 2304 flags 0x02 dlen 121                                                                 #9231 [hci1] 28.982713
> < ISO Data TX: Handle 2305 flags 0x02 dlen 121                                                                 #9232 [hci1] 28.982725
> > HCI Event: Number of Completed Packets (0x13) plen 9                                                         #9233 [hci1] 28.986397
>         Num handles: 2
>         Handle: 2304 Address: F8:4E:17:CB:C2:83 (Sony Corporation)
>         Count: 1
>         #9231: len 121 (322 Kb/s)
>         Latency: 3 msec (2-11 msec ~4 msec)
>         Handle: 2305 Address: C9:72:16:D5:2C:FC (Static)
>         Count: 1
>         #9232: len 121 (322 Kb/s)
>         Latency: 3 msec (2-10 msec ~3 msec)
> ```
> 
> The kernel packet scheduler also appears to have done some packet reordering here `2304 2305 2304 2305 2305 2304 ...`, but I'm not sure this is the cause for the problem.

Yeah, I think we should actually be using Unsegmented Framed packets once controllers implements it: 

https://www.bluetooth.com/core-specification-6-feature-overview/

The actually have a picture that explain well this problem where host and controller are unsynchronized https://www.bluetooth.com/core-specification-6-feature-overview/ (6.1.3.2 Issues addressed by ISOAL):

[image]

Vudentz, 2025-02-19:
```
2. ISOAL features

...
Unframed PDUs shall only be used when the ISO_Interval is equal to or an integer multiple of the SDU_Interval and a **constant** time offset alignment is maintained between the SDU generation and the timing in the isochronous transport. This requires the upper layer to synchronize generation of its data to the effective transport timing. When the Host requests the use of framed PDUs, the Controller shall use framed PDUs.
```

So I assume we shall never be using unframed, apart from testing, since it near impossible to maintain a constant time offset when the clock domains are different, also further down there is the following table:

[image]

This probably means that even if we don't use 0x02, if the controller is capable, it can use that over the air once it detects that packets are not really segmented.

pv, 2025-05-03:
The framed/unsegmented mode doesn't seem to solve the issue that in HCI Host->Controller communication, controller cannot know which ISO packets are supposed to go out at same time, except from packet arrival times.

No changes in the problem in recent AX210 firmware (Firmware timestamp 2024.48 buildtype 1 build 81864 & Firmware SHA1: 0xc115e35a), seems to manage to get stuck in desynchronized state.

In desynchronized state, btmon / TX timestamping latency values look as if they are synchronized. However, kernel varies between sending in 2304/2305 and 2305/2304 orders, don't know if the firmware is written in such way that it gets confused by that.

pv, 2025-05-04:
One way to reproduce seems to be to run the stack in qemu and forward the AX210 USB device there. This setup appears susceptible to delays in the USB subsystem, so eg. when changing window focus outside the VM, `btusb_tx_complete` can be delayed by ~30+ ms. Sometimes when this occurs, CIS synchronization is lost and controller does not recover even though subsequent packet pairs appear to be delivered a the same time.

<details>

```
11:52:13.693964: btusb_send_frame_intel:2578: hci0
11:52:13.694020: btusb_send_frame_intel:2578: hci0
11:52:13.694052: btusb_tx_complete:1823: hci0 urb 0000000006f88a13 status 0 count 125
11:52:13.694836: btusb_tx_complete:1823: hci0 urb 00000000463fbeb1 status 0 count 125
11:52:13.698845: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.701860: btusb_send_frame_intel:2578: hci0
11:52:13.701925: btusb_send_frame_intel:2578: hci0
11:52:13.704816: btusb_tx_complete:1823: hci0 urb 00000000463fbeb1 status 0 count 125
11:52:13.704851: btusb_tx_complete:1823: hci0 urb 0000000006f88a13 status 0 count 125
11:52:13.708928: btusb_send_frame_intel:2578: hci0
11:52:13.709018: btusb_send_frame_intel:2578: hci0
11:52:13.716903: btusb_send_frame_intel:2578: hci0
11:52:13.716951: btusb_send_frame_intel:2578: hci0
11:52:13.716977: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.724004: btusb_send_frame_intel:2578: hci0
11:52:13.724136: btusb_send_frame_intel:2578: hci0
11:52:13.731829: btusb_send_frame_intel:2578: hci0
11:52:13.731881: btusb_send_frame_intel:2578: hci0
11:52:13.746911: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.747261: btusb_tx_complete:1823: hci0 urb 0000000006f88a13 status 0 count 125
11:52:13.747432: btusb_tx_complete:1823: hci0 urb 00000000463fbeb1 status 0 count 125
11:52:13.747476: btusb_tx_complete:1823: hci0 urb 00000000fbdb8d34 status 0 count 125
11:52:13.747513: btusb_tx_complete:1823: hci0 urb 00000000ff5b967e status 0 count 125
11:52:13.747548: btusb_tx_complete:1823: hci0 urb 0000000001cafdec status 0 count 125
11:52:13.747631: btusb_tx_complete:1823: hci0 urb 000000008f067270 status 0 count 125
11:52:13.747679: btusb_send_frame_intel:2578: hci0
11:52:13.747713: btusb_tx_complete:1823: hci0 urb 000000000c2d80eb status 0 count 125
11:52:13.747747: btusb_send_frame_intel:2578: hci0
11:52:13.747781: btusb_tx_complete:1823: hci0 urb 000000007e9b8304 status 0 count 125
11:52:13.747862: btusb_send_frame_intel:2578: hci0
11:52:13.747897: btusb_send_frame_intel:2578: hci0
11:52:13.747929: btusb_tx_complete:1823: hci0 urb 000000008f067270 status 0 count 125
11:52:13.747963: btusb_tx_complete:1823: hci0 urb 000000000c2d80eb status 0 count 125
11:52:13.747996: btusb_tx_complete:1823: hci0 urb 000000007e9b8304 status 0 count 125
11:52:13.748862: btusb_tx_complete:1823: hci0 urb 0000000001cafdec status 0 count 125
11:52:13.752869: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.753908: btusb_send_frame_intel:2578: hci0
11:52:13.753972: btusb_send_frame_intel:2578: hci0
11:52:13.754007: btusb_tx_complete:1823: hci0 urb 0000000001cafdec status 0 count 125
11:52:13.754847: btusb_tx_complete:1823: hci0 urb 000000007e9b8304 status 0 count 125
11:52:13.756875: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.761851: btusb_send_frame_intel:2578: hci0
11:52:13.761972: btusb_send_frame_intel:2578: hci0
11:52:13.762008: btusb_tx_complete:1823: hci0 urb 000000007e9b8304 status 0 count 125
11:52:13.762053: btusb_tx_complete:1823: hci0 urb 0000000001cafdec status 0 count 125
11:52:13.763887: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
11:52:13.768830: btusb_send_frame_intel:2578: hci0
11:52:13.768894: btusb_send_frame_intel:2578: hci0
11:52:13.769833: btusb_tx_complete:1823: hci0 urb 0000000001cafdec status 0 count 125
11:52:13.769886: btusb_tx_complete:1823: hci0 urb 000000007e9b8304 status 0 count 125
11:52:13.772842: btusb_intr_complete:1357: hci0 urb 00000000b248ba91 status 0 count 11
```

</details>

closed 2025-09-20 by github-actions[bot]