bluez/bluez #1090: Retain devices in cache indefinitely when Temporary Timeout = 0 in main.conf

Vishal8523, 2025-02-07:
Version: BlueZ 5.79

It seems that the setting to keep temporary unpaired devices in BlueZ 5.79 does not work as expected. When set to a very high value like 99999, the unpaired devices are still missing from the cache. 
Ideally, when this value is set to 0, the timer should be disabled and the devices shall remain in the cache indefinitely.

Request you to look into this issue
Working on a PR on this as well

Vudentz, 2025-02-07:
> Version: BlueZ 5.79
>
> It seems that the setting to keep temporary unpaired devices in BlueZ 5.79 does not work as expected. When set to a very high value like 99999, the unpaired devices are still [...]

Well that doesn't do what you are saying though, it is actually make the device persistent which would mean something entirely different, perhaps the problem is that you are assuming that any device found would be stored in the cache which doesn't happens for temporary devices, that is on purpose otherwise one can end up with a lot of devices being stored that they never used.

Vishal8523, 2025-02-11:
@Vudentz : Ok. Understood about the point of setting the value to 0. But when set to a very high value like 99999, the unpaired devices are still missing from the cache. Ideally, it should be present for 99999 seconds.

Just check out the pull request. It contains the fix for this 99999 seconds  issue

Vishal8523, 2025-02-14:
@Vudentz : Actually, the code that I had written does the automatic removal of devices from cache which are not of any interest to me. So, I own that responsibility. How do I make it happen ?

Vishal8523, 2025-02-14:
@Vudentz : Something like https://github.com/bluez/bluez/issues/146.

closed 2025-02-17 by Vishal8523