bluez/bluez #1092: Sony UFT field 1 cannot be reconnected after reboot

Xdavius, 2025-02-10:
Hi dev,

I have bought this Sony Bluetooth Speaker (ULT field 1) to my wife.

After each computer reboot, I need to discard the speaker and pair it again.

When it tries to reconnect, the speaker freeze and go power off.

On the PC, I can see connected, and after 5/10s, disconnected with BR error. The speaker doesn't play his sound for successful connection and goes poweroff.

It's happening only after a reboot of the computer. Until I don't, I can connect and disconnect and power off/power on the speaker, the reconnexion works.

Tested with 2 computers, with 2 different BT chip (ax200 and AMD Wifi 5 chip), Archlinux and Debian Testing/Sid. The issue is reproductible.

Not tested with android smartphone (I haven't rebooted it, so the speaker still reconnect successfully)

I have a Sony XB-40 BT speaker who works perfectly without any issue...

Any help will be appreciated. 

Best regards 

Davius

Xdavius, 2025-02-11:
Hi dev,

I think I have solved but not sure what was working.

I have activated fastconnection and disabled LE mode.

Now it works.

Have a nice day

Xdavius, 2025-02-12:
Hi.

Some news :

The Sony ULT FIELD 1 works correctly if I switch to only BR. If I use both, nothing to do... But Xbox controller is working on LE

So impossible to have both working together on bluetooth...

I have no issues with Sony XB-40...

I have tried to block LE device without effect (only effect is the name of the device doesn't change anymore to le-ult-1 on reconnect.)

Still can't power of the speaker and power it on again.

Now, it doesn't reconnect even if the PC is not rebooted

I don't understand and I blocked. I need help please...

Regards

Vudentz, 2025-02-12:
@Xdavius can say what kernel version BlueZ version are you running, it would also help if you add the btmon logs when the problem happens.

Xdavius, 2025-02-12:
> @Xdavius can say what kernel version BlueZ version are you running, it would also help if you add the btmon logs when the problem happens.

Hi

Kernel:
6.12.12-amd64

bluetoothctl --version :
5.79

btmon log : 
https://pastebin.com/SQAsKq9f

After many tries again, I'm able to reconnect if I manualy do bluetoothctl connect, and power on the speaker imediatly after launching the command (or clicking on connect on bluedevil)

If I just power on the device, the connection fail and, I also can see the device name change to LE-ULT-FIELD-1 (which is not le_ult_1, because the LE mac adresse isn't paired...)

The normal name is "ULT FIELD 1"

Thank you for replying and hope the log will help.

Regards

Vudentz, 2025-03-06:
@Xdavius could you try with the above changes by setting the PreferredBearer to bredr?

ginkcode, 2025-08-14:
I have the same issue with "ULT FIELD 1" speaker on Ubuntu 24.04.3 LTS 
After the first pair, the device working well and I can see in device info with
```
Name=ULT FIELD 1
SupportedTechnologies=BR/EDR;
```
But after turn off the speaker and turn on again. It suddenly changes the name to
```
Name=LE-ULT FIELD 1
```
The speaker connects in a short time but can't be detected as sound device. Then it turns off.
I guess it start to reconnect again by LE but need to switch to BR/EDR after that, which is unsuccessful.

Xdavius, 2025-08-14:
> @Xdavius could you try with the above changes by setting the PreferredBearer to bredr?

Hi !
Sorry for very late reply.

I have tried to change lots of changes. Maybe with new version I will test again.

For now, my workaround is :

- start computer with speaker off
- on KDE click the "connect" button and just after, power on the speaker.

This is working and the connection is made in br. This speaker seems to need LE activated to work and connect. I have tried to force only BR but speaker was falling to connect. It seems to need both.

ASpoonfulOfSalt, 2026-01-01:
@Xdavius 
Thanks for the tip! Probably how I'll do it for now. 
But I wish I could get PreferredBearer to work since it sounds like it should. :/