bluez/bluez #1094: Received a GATT Write/Read Response That Does Not Match the GATT Write/Read Request Opcode After Enabling EATT

xihua13104, 2025-02-14:
Hi,

Unicast sink: Bluez 5.79 + Linux kernel 6.6.18 + Pipewire 1.2.0
Unicast source: Pixel 9

**Configuration:**
/etc/bluetooth/main.conf
[GATT]  
Channels = 6  

**Code Change in src/main.c:**
--- a/src/main.c
+++ b/src/main.c
@@ -1119,7 +1119,7 @@ static void parse_gatt(GKeyFile *config)
 	parse_config_u16(config, "GATT", "ExchangeMTU", &btd_opts.gatt_mtu,
 				BT_ATT_DEFAULT_LE_MTU, BT_ATT_MAX_LE_MTU);
 	parse_config_u8(config, "GATT", "Channels", &btd_opts.gatt_channels,
-				1, 5);
+				1, 6);
 	parse_config_bool(config, "GATT", "Client", &btd_opts.gatt_client);
 	parse_gatt_export(config);
 }

**Issue Description:**
To test EATT, I enabled it by modifying main.conf as shown above. When connecting a Pixel 9 to the DUT, the connection failed.
From bluetoothd.log, I observed that a GATT Read Request sent by the DUT to the Pixel timed out, causing the connection to drop:
```shell
bluetoothd[2225]: src/shared/att.c:timeout_cb() (chan 0x2460e60) Operation(opcode:0xa) timed out,op-id:0x1d, 0x18c, op->timeout_id:0xf7b84918
bluetoothd[2225]: src/shared/mcp.c:mcp_pending_complete() pending_id:0xf
bluetoothd[2225]: src/shared/mcp.c:read_content_control_id() Unable to read content control id: error 0x0e
bluetoothd[2225]: src/shared/att.c:disconnect_cb() (chan 0x2464fb0) Failed to obtain disconnect error: Bad file descriptor
bluetoothd[2225]: src/shared/att.c:disconnect_cb() Channel 0x2464fb0 disconnected: Success
bluetoothd[2225]: src/shared/att.c:disconnect_cb() (chan 0x2460e60) Failed to obtain disconnect error: Bad file descriptor
bluetoothd[2225]: src/shared/att.c:disconnect_cb() Channel 0x2460e60 disconnected: Success
bluetoothd[2225]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000c
bluetoothd[2225]: src/adapter.c:dev_disconnected() Device E8:D5:2B:59:57:A6 disconnected, reason 3
bluetoothd[2225]: src/adapter.c:adapter_remove_connection() 
bluetoothd[2225]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[2225]: src/shared/att.c:disconnect_cb() Channel 0x2455c58 disconnected: Connection reset by peer
``` 
However, according to the BTSnoop log, the Pixel 9 did send a response.
Upon further investigation, I found that a received GATT Response did not match the expected Opcode in the pending request for the channel. It seems that EATT mixed up data and channels during reception:
```shell
bluetoothd[2225]: src/shared/att.c:handle_rsp() (chan 0x2465598 fd:22) unmatched rsp-opcode:0x12, expect_opcode:0xa  
bluetoothd[2225]: src/shared/att.c:dump_eatt_channels() Dumping all active EATT channels:  
bluetoothd[2225]: src/shared/att.c:dump_eatt_channels() Channel(0x2465598): fd=22, io=0x245a4a0, mtu=256  
bluetoothd[2225]: src/shared/att.c:handle_rsp() (chan 0x2465598 fd:22) Failed to handle response PDU; opcode: 0x13, pending_req:0x245e088, pending_opcode:0xa  
``` 
**Additional Observations:**
I tested the same modification with another phone (Redmi K70), and it worked fine.
The key difference is that Pixel 9 creates all 5 EATT channels at once using L2CAP Credit-Based Connection, whereas K70 creates them one by one .

Any help would be greatly appreciated!

**DUT log connect with Pixel 9:**

[btsnoop.cfa.log](https://github.com/user-attachments/files/18793010/btsnoop.cfa.log)

[bluetoothd.log](https://github.com/user-attachments/files/18793012/bluetoothd.log)

[btmon.txt](https://github.com/user-attachments/files/18793016/btmon.txt)

Vudentz, 2025-02-14:
```
bluetoothd[2227]: < ACL Data TX: Hand.. flags 0x00 dlen 9  #131 [hci0] 6.582585
      Channel: 120 len 5 sdu 3 [PSM 39 mode Enhanced Credit (0x81)] {chan 4}
      ATT: Read Request (0x0a) len 2
        Handle: 0x0036 Type: Track Position (0x2b99)
> ACL Data RX: Handle 16 flags 0x02 dlen 11                #132 [hci0] 6.583067
      Channel: 64 len 7 sdu 5 [PSM 39 mode Enhanced Credit (0x81)] {chan 4}
      ATT: Read Response (0x0b) len 4
        Value[4]: ffffffff
        Handle: 0x0033 Type: Track Duration (0x2b98)
        Value[4]: ffffffff
          Track Duration: 4294967295
```

I think there is something wrong with the order we respond to the channels:

```
> ACL Data RX: Handle 16 flags 0x02 dlen 26                #108 [hci0] 6.544418
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
        PSM: 39 (0x0027)
        MTU: 256
        MPS: 251
        Credits: 65535
        Source CID: 116
        Source CID: 117
        Source CID: 118
        Source CID: 119
        Source CID: 120
> ACL Data RX: Handle 16 flags 0x02 dlen 11                #109 [hci0] 6.544954
      ATT: Read By Type Request (0x08) len 6
        Handle range: 0x0001-0xffff
        Attribute type: Client Supported Features (0x2b29)
< ACL Data TX: Handle 16 flags 0x00 dlen 26                #110 [hci0] 6.553555
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 2 len 18
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
        Destination CID: 68
        Destination CID: 67
        Destination CID: 66
        Destination CID: 65
        Destination CID: 64
```

The order is reverted it seems, then we disconnect here do:

```
< ACL Data TX: Handle 16 flags 0x00 dlen 12               #402 [hci0] 42.307256
      LE L2CAP: Disconnection Response (0x07) ident 3 len 4
        Destination CID: 68
        Source CID: 120
```

It should have been 64 to pair with 120, but it got 68, I don't recall if we fixed something like that in the kernel but it is probably worth checking before we attempt to fix it.

xihua13104, 2025-02-14:
Hi

```shell
bluetoothd[2227]: < ACL Data TX: Hand.. flags 0x00 dlen 9  #131 [hci0] 6.582585
      Channel: 120 len 5 sdu 3 [PSM 39 mode Enhanced Credit (0x81)] {chan 4}
      ATT: Read Request (0x0a) len 2
        Handle: 0x0036 Type: Track Position (0x2b99)

> ACL Data RX: Handle 16 flags 0x02 dlen 11                #139 [hci0] 6.596909
      Channel: 64 len 7 sdu 5 [PSM 39 mode Enhanced Credit (0x81)] {chan 4}
      ATT: Read Response (0x0b) len 4
        Value[4]: ffffffff
        Handle: 0x0036 Type: Track Position (0x2b99)
        Value[4]: ffffffff
          Track Position: 4294967295
``` 
In the initial interactions after the EATT channels are created, the Source CID and Destination CID match. However, when the issue occurs, it seems that the Destination CID is flipped when the DUT responds to the L2CAP disconnection request. Does this indicate that the Destination CID was unintentionally modified during EATT operation?

I have reviewed the Bluetooth-related commits in the kernel but did not find any commits addressing this issue.

Vudentz, 2025-02-14:
@xihua13104 could you please try with the changes above?

xihua13104, 2025-02-17:
Yes, with your changes, the connection with Pixel 9 is success. The order of Destination CIDs in Enhanced Credit Connection Response  
is correct.

```shell
> ACL Data RX: Handle 16 flags 0x02 dlen 26               #122 [hci0] 25.240845
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
        PSM: 39 (0x0027)
        MTU: 256
        MPS: 251
        Credits: 65535
        Source CID: 75
        Source CID: 76
        Source CID: 78
        Source CID: 79
        Source CID: 74

< ACL Data TX: Handle 16 flags 0x00 dlen 26               #125 [hci0] 25.244363
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 2 len 18
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
        Destination CID: 64
        Destination CID: 65
        Destination CID: 66
        Destination CID: 67
        Destination CID: 68
``` 

[btmon-pixel-pass.txt](https://github.com/user-attachments/files/18817670/btmon-pixel-pass.txt)

But the connection with K70 is failed. In the last Enhanced Credit Connection Response, we mistakenly responded with two Destination CIDs
```shell
> ACL Data RX: Handle 16 flags 0x02 dlen 18               #491 [hci0] 34.451908
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 6 len 10
        PSM: 39 (0x0027)
        MTU: 517
        MPS: 251
        Credits: 255
        Source CID: 70
< ACL Data TX: Handle 16 flags 0x00 dlen 20               #493 [hci0] 34.482123
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 6 len 12
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
        Destination CID: 64
        Destination CID: 68
``` 

[btmon-k70-fail.txt](https://github.com/user-attachments/files/18817694/btmon-k70-fail.txt)

Vudentz, 2025-02-17:
> **But the connection with K70 is failed.** In the last Enhanced Credit Connection Response, we mistakenly responded with two Destination CIDs
> 
> > ACL Data RX: Handle 16 flags 0x02 dlen 18               #491 [hci0] 34.451908
>       LE L2CAP: Enhanced Credit Connection Request (0x17) ident 6 len 10
>         PSM: 39 (0x0027)
>         MTU: 517
>         MPS: 251
>         Credits: 255
>         Source CID: 70
> < ACL Data TX: Handle 16 flags 0x00 dlen 20               #493 [hci0] 34.482123
>       LE L2CAP: Enhanced Credit Connection Response (0x18) ident 6 len 12
>         MTU: 517
>         MPS: 247
>         Credits: 3
>         Result: Connection successful (0x0000)
>         Destination CID: 64
>         Destination CID: 68
> 
> [btmon-k70-fail.txt](https://github.com/user-attachments/files/18817694/btmon-k70-fail.txt)

Weird, is this with the latest bluetooth-next or you did apply to an older kernel?

xihua13104, 2025-02-18:
I apply your changes to an older kernel version 6.12.

xihua13104, 2025-02-19:
Unfortunately, the situation is even worse with this patch, we don’t have the Destination CID in each Enhanced Credit Connection Response.
```shell
> ACL Data RX: Handle 16 flags 0x02 dlen 18               #381 [hci0] 15.075387
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 10
        PSM: 39 (0x0027)
        MTU: 517
        MPS: 251
        Credits: 255
        Source CID: 72
< ACL Data TX: Handle 16 flags 0x00 dlen 16               #382 [hci0] 15.076284
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 2 len 8
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
	
> ACL Data RX: Handle 16 flags 0x02 dlen 18               #386 [hci0] 35.085779
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 3 len 10
        PSM: 39 (0x0027)
        MTU: 517
        MPS: 251
        Credits: 255
        Source CID: 73
< ACL Data TX: Handle 16 flags 0x00 dlen 16               #387 [hci0] 35.086667
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 3 len 8
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
``` 

[btmon-k70-fail-no-DCID.txt](https://github.com/user-attachments/files/18857662/btmon-k70-fail-no-DCID.txt)

In the previous patch, K70 sent 5 Enhanced Credit Connection Requests. In the last Enhanced Credit Connection Response, we mistakenly replied with two destination CIDs. The first four times were correct.

What is the linux kernel version you test with tools/l2cap-tester？
>IVe tried with tools/l2cap-tester -s "EATT Server - Success" which sends just one channel request:

> ACL Data RX: Handle 42 flags 0x02 dlen 18         #152 [hci0] 11:37:00.172648
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 1 len 10
        PSM: 39 (0x0027)
        MTU: 64
        MPS: 64
        Credits: 5
        Source CID: 65
< ACL Data TX: Handle 42 flags 0x00 dlen 18         #155 [hci0] 11:37:00.176649
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 1 len 10
        MTU: 672
        MPS: 188
        Credits: 4
        Result: Connection successful (0x0000)
        Destination CID: 64

xihua13104, 2025-02-19:
I think we should set the FLAG_ECRED_CONN_REQ_SENT flag to prevent repeat ECRED response.
I made the changes below, and the issue is fixed.
```c
void __l2cap_ecred_conn_rsp_defer(struct l2cap_chan *chan)
{
	struct l2cap_conn *conn = chan->conn;
	struct l2cap_ecred_rsp_data data;
	u16 id = chan->ident;
	int result = 0;

        ......

	l2cap_send_cmd(conn, id, L2CAP_ECRED_CONN_RSP,
		       sizeof(data.pdu.rsp) + (data.count * sizeof(__le16)),
		       &data.pdu);
	
        +test_and_set_bit(FLAG_ECRED_CONN_REQ_SENT, &chan->flags);
}
``` 

But this solution may  be incomplete, we should also clear this flag somewhere appropriate.

xihua13104, 2025-02-20:
I tested https://github.com/BluezTestBot/bluetooth-next/pull/2720, the issue is fixed. Thanks for your help.

xihua13104, 2025-02-20:
```shell
diff --git a/src/main.c b/src/main.c
index d13ada2af..50346abbb 100644
--- a/src/main.c
+++ b/src/main.c
@@ -1119,7 +1119,7 @@ static void parse_gatt(GKeyFile *config)
 	parse_config_u16(config, "GATT", "ExchangeMTU", &btd_opts.gatt_mtu,
 				BT_ATT_DEFAULT_LE_MTU, BT_ATT_MAX_LE_MTU);
 	parse_config_u8(config, "GATT", "Channels", &btd_opts.gatt_channels,
-				1, 5);
+				1, 6);
 	parse_config_bool(config, "GATT", "Client", &btd_opts.gatt_client);
 	parse_gatt_export(config);
 }
diff --git a/src/main.conf b/src/main.conf
index 60ab60edc..749ed6d25 100644
--- a/src/main.conf
+++ b/src/main.conf
@@ -264,7 +264,8 @@
 #ExchangeMTU = 517
 
 # Number of ATT channels
-# Possible values: 1-5 (1 disables EATT)
+# Possible values: 1-6 (1 disables EATT)
+# The ATT channels contains ATT connection(1) and L2CAP CB connection(1-5)
 # Default to 1
 #Channels = 1
``` 

By the way, we made this changes to expand the maximum GATT channels to 6, because the ATT takes one, and the EATT takes the rest 5 channels. This can also be confirmed in the interaction log with pixel. Without this change, the 5th Enhanced Credit Connection Request will be refused.

```shell
> ACL Data RX: Handle 16 flags 0x02 dlen 26               #122 [hci0] 25.240845
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
        PSM: 39 (0x0027)
        MTU: 256
        MPS: 251
        Credits: 65535
        Source CID: 75
        Source CID: 76
        Source CID: 78
        Source CID: 79
        Source CID: 74

< ACL Data TX: Handle 16 flags 0x00 dlen 26               #125 [hci0] 25.244363
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 2 len 18
        MTU: 517
        MPS: 247
        Credits: 3
        Result: Connection successful (0x0000)
        Destination CID: 64
        Destination CID: 65
        Destination CID: 66
        Destination CID: 67
        Destination CID: 68

> ACL Data RX: Handle 16 flags 0x02 dlen 18               #497 [hci0] 14.485092
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 6 len 10
        PSM: 39 (0x0027)
        MTU: 517
        MPS: 251
        Credits: 255
        Source CID: 91

< ACL Data TX: Handle 16 flags 0x00 dlen 16               #499 [hci0] 14.515086
      LE L2CAP: Enhanced Credit Connection Response (0x18) ident 6 len 8
        MTU: 517
        MPS: 247
        Credits: 3
        Result: All Connections refused - insufficient authorization (0x0006)
``` 

So is there any plan to merge this PR https://github.com/BluezTestBot/bluez/commit/93d35a558e66f1256c24bdd8de23d7226e992fcf ?

Vudentz, 2025-02-20:
> diff --git a/src/main.c b/src/main.c
> index d13ada2af..50346abbb 100644
> --- a/src/main.c
> +++ b/src/main.c
> @@ -1119,7 +1119,7 @@ static void parse_gatt(GKeyFile *config)
>  	parse_config_u16(config, "GATT", "ExchangeMTU", &btd_opts.gatt_mtu,
>  				BT_ATT_DEFAULT_LE_MTU, BT_ATT_MAX_LE_MTU);
>  	parse_config_u8(config, "GATT", "Channels", &btd_opts.gatt_channels,
> -				1, 5);
> +				1, 6);
>  	parse_config_bool(config, "GATT", "Client", &btd_opts.gatt_client);
>  	parse_gatt_export(config);
>  }
> diff --git a/src/main.conf b/src/main.conf
> index 60ab60edc..749ed6d25 100644
> --- a/src/main.conf
> +++ b/src/main.conf
> @@ -264,7 +264,8 @@
>  #ExchangeMTU = 517
>  
>  # Number of ATT channels
> -# Possible values: 1-5 (1 disables EATT)
> +# Possible values: 1-6 (1 disables EATT)
> +# The ATT channels contains ATT connection(1) and L2CAP CB connection(1-5)
>  # Default to 1
>  #Channels = 1
> 
> By the way, we made this changes to expand the maximum GATT channels to 6, because the ATT takes one, and the EATT takes the rest 5 channels. This can also be confirmed in the interaction log with pixel. Without this change, the 5th Enhanced Credit Connection Request will be refused.
> 
> > ACL Data RX: Handle 16 flags 0x02 dlen 26               #122 [hci0] 25.240845
>       LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
>         PSM: 39 (0x0027)
>         MTU: 256
>         MPS: 251
>         Credits: 65535
>         Source CID: 75
>         Source CID: 76
>         Source CID: 78
>         Source CID: 79
>         Source CID: 74
> 
> < ACL Data TX: Handle 16 flags 0x00 dlen 26               #125 [hci0] 25.244363
>       LE L2CAP: Enhanced Credit Connection Response (0x18) ident 2 len 18
>         MTU: 517
>         MPS: 247
>         Credits: 3
>         Result: Connection successful (0x0000)
>         Destination CID: 64
>         Destination CID: 65
>         Destination CID: 66
>         Destination CID: 67
>         Destination CID: 68
> 
> 
> > ACL Data RX: Handle 16 flags 0x02 dlen 18               #497 [hci0] 14.485092
>       LE L2CAP: Enhanced Credit Connection Request (0x17) ident 6 len 10
>         PSM: 39 (0x0027)
>         MTU: 517
>         MPS: 251
>         Credits: 255
>         Source CID: 91
> 
> < ACL Data TX: Handle 16 flags 0x00 dlen 16               #499 [hci0] 14.515086
>       LE L2CAP: Enhanced Credit Connection Response (0x18) ident 6 len 8
>         MTU: 517
>         MPS: 247
>         Credits: 3
>         Result: All Connections refused - insufficient authorization (0x0006)
> 
> So is there any plan to merge this PR [BluezTestBot@93d35a5](https://github.com/BluezTestBot/bluez/commit/93d35a558e66f1256c24bdd8de23d7226e992fcf) ?

It has been merged already: https://github.com/bluez/bluez/commit/dd5316091c71c0cb3141a5f44580d973d3770eed

xihua13104, 2025-02-20:
OK. Thank you.

closed 2025-02-20 by xihua13104