bluez/bluez #1097: DJI Mic Mini nable to connect to bluetooth (latest Ubuntu / 6.12.13)

kolarski, 2025-02-15:
Using DJI Mic Mini (https://www.dji.com/bg/mic-mini)
Using Ubuntu 24 with latest kernel + updated to the latest stable kernel (6.12.13)

The mic connects with Bluetooth and disconnects after few seconds with this message: 
```
bluetoothd[12381]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
```
How can I collect more debug information ?

kolarski, 2025-02-15:
Here is a Wireshark (btmon packet capture) 

[dji_mic_mini.zip](https://github.com/user-attachments/files/18812372/dji_mic_mini.zip)

Basically this is the communication: 
```
AT+BRSF=255
+BRSF: 1632
OK
AT+BAC=1,2
OK
AT+CIND=?
+CIND:("SERVICE",(0-1)),("CALL",(0-1)),("CALLSETUP",(0-3)),("CALLHELD",(0-2)),("SIGNAL",(0-5)),("ROAM",(0-1)),("BATTCHG",(0-5))
OK
AT+CIND?
+CIND: 0,0,0,0,0,0,4
OK
AT+CMER=3,0,0,1
OK
+BCS: 2
AT+COPS=3,0
OK
AT+CMEE=1
OK
AT+BTRH?
+CME ERROR: 1
AT+CLIP=1
OK
AT+CCWA=1
+CME ERROR: 1
```

and these are the bluetoothd logs for the event:
```
bluetoothd[152654]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0013
bluetoothd[152654]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0 method 0
bluetoothd[152654]: src/adapter.c:trigger_start_discovery() 
bluetoothd[152654]: src/adapter.c:cancel_passive_scanning() 
bluetoothd[152654]: src/device.c:connect_profiles() /org/bluez/hci0/dev_58_B8_58_A9_1B_D9 (all), client :1.86
bluetoothd[152654]: src/service.c:btd_service_ref() 0x593dc9c719f0: ref=2
bluetoothd[152654]: src/service.c:change_state() 0x593dc9c719f0: device 58:B8:58:A9:1B:D9 profile Hands-Free Voice gateway state changed: disconnected -> connecting (0)
bluetoothd[152654]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
bluetoothd[152654]: src/adapter.c:connected_callback() hci0 device 58:B8:58:A9:1B:D9 connected eir_len 26
bluetoothd[152654]: src/profile.c:ext_connect() Hands-Free Voice gateway connected to 58:B8:58:A9:1B:D9
bluetoothd[152654]: src/service.c:change_state() 0x593dc9c719f0: device 58:B8:58:A9:1B:D9 profile Hands-Free Voice gateway state changed: connecting -> connected (0)
bluetoothd[152654]: src/device.c:device_profile_connected() Hands-Free Voice gateway Success (0)
bluetoothd[152654]: src/device.c:device_profile_connected() returning response to :1.86
bluetoothd[152654]: src/service.c:btd_service_ref() 0x593dc9c719f0: ref=3
bluetoothd[152654]: plugins/policy.c:service_cb() Added Hands-Free Voice gateway reconnect 0
bluetoothd[152654]: src/device.c:search_cb() 58:B8:58:A9:1B:D9: No service update
bluetoothd[152654]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_58_B8_58_A9_1B_D9 err 0
bluetoothd[152654]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
bluetoothd[152654]: src/service.c:change_state() 0x593dc9c719f0: device 58:B8:58:A9:1B:D9 profile Hands-Free Voice gateway state changed: connected -> disconnected (0)
bluetoothd[152654]: src/service.c:btd_service_unref() 0x593dc9c719f0: ref=2

```

kolarski, 2025-02-15:
Another note, The microphone works great on MacOS (Tested on MacOS Mojave 10.14.2) connects and works as expected

pongnguy, 2025-05-04:
Same issue here.  Connecting with android phone works.

migueloporto, 2025-05-12:
+1 on this issue

```
bluetoothd[2678]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
bluetoothd[2678]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
bluetoothd[2678]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
bluetoothd[2678]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
bluetoothd[2678]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
bluetoothd[2678]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)

julian-alarcon, 2025-08-19:
Is this still an issue? What about the Mic Mini or the Mic 2?

GaKoDe, 2025-09-03:
Tested on Ubuntu - 20.04.06 / kernel version 5.15.0-139-generic
Bluez version - 5.53
the bug is reproducing.

Eyolon, 2025-09-10:
Hello
I finaly make it works on my arch linux

[image]
my main approche was to be sure i've HFP A2DP profils ready to use.
but they are not
so this is what i did to make it finaly fucking work
first 
`pactl list modules | grep bluetooth` was empty so i had "load-module module-bluetooth-policy" at the end of /etc/pulse/default.pa
 and exec 
```
pactl load-module module-bluetooth-policy
pactl load-module module-bluetooth-discover

```
then add HFP with ofono
so install it with yay (arch requirement !) enable and start it 
`
sudo systemctl enable ofono
sudo systemctl start ofono
`
modified the conf file /etc/ofono/phonesim.conf
and added 
[General]
Enable=1

finaly to be sure pulseAudio work with ofono i edited again my default.pa
and added
load-module module-bluetooth-discover headset=ofono

finaly "full restart"
`
systemctl restart ofono
pulseaudio -k
pulseaudio --start
`

connect my mic and Tadaaa ~~~

Jeffrey-H, 2025-10-16:
+1 on this issue

closed 2025-11-22 by github-actions[bot]