bluez/bluez #1100: Retain temporary devices

Vishal8523, 2025-02-20:
Version: BlueZ 5.79

I had written production code in BlueZ 5.54. Recently, I migrated to BlueZ 5.79
In BlueZ 5.54, I have handled the removal of devices in cache
But the behaviour of keeping unpaired devices in cache is changed in BlueZ 5.79

How do I retain the devices in BlueZ 5.79 ?

@Vudentz / @vcgomes / @elmarco / @rgilton / @teg

Vudentz, 2025-02-20:
Well only permanent devices shall be save in the storage, or you really meant runtime memory of bluetoothd? Or are you asking for an API to mark devices as permanent without connecting to them?

Vishal8523, 2025-02-21:
**APIs to mark devices as permanent**. So that application can handle the device removal from cache whenever feasible and necessary. That would really help in backward compatibility of older application developed in BlueZ

Vishal8523, 2025-05-05:
Not stale

Vishal8523, 2025-05-06:
In version 5.54, there was no concept of TemporaryTimeout, so the applications using BlueZ would be responsible for device removal from cache. Such applications will definitely behave strange with the introduction of TemporaryTimeout, where the application does not have the control of device removal. The device objects received at the application over Dbus would be invalid as it;s corresponding object in cache is non existent. Atleast if there is a setting to disable TemporaryTimeout, then it would be very flexible and easy to integrate. Thanks

Vudentz, 2025-05-06:
> In version 5.54, there was no concept of TemporaryTimeout, so the applications using BlueZ would be responsible for device removal from cache. Such applications will definitely behave strange with [...]

Ever since the introduction of the patch bellow setting TemporaryTimeout=0 will keep the temporary devices forever:

```

commit 24d986e72d4bb7e412b8dd42a54b99ad24a1f2dd
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Thu Feb 10 14:11:42 2022 -0800

    device: Fix crash when removing device
    
    Calling btd_adapter_remove_device from device_remove_connection can
    cause a crash, so instead of removing it immediatelly this set a the
    temporary timeout to 0.
    
    Fixes: https://github.com/bluez/bluez/issues/290
```

I will fix the documentation since previously it was assumed that 0 means remove immediately.