bluez/bluez #1103: Bluez is showing a device that should not be visible

RyuzakiKK, 2025-02-20:
## Steps to reproduce

- Install bluez 5.79
- Set the Creative Zen Air Pro in pairing mode
- Notice how `bluetoothctl devices` shows an unexpected device with a MAC address of `00:00:00:00:00:00`
- Trying to pair it would stall bluez until it eventually goes into timeout

## Additional info

```
$ bluetoothctl devices
[...]
Device 00:00:00:00:00:00 00-00-00-00-00-00
Device 00:02:3C:B9:02:B5 Zen Air Pro
Device 00:02:3C:BD:97:48 Zen Air Pro  # This entry is grayed out

$ bluetoothctl info 00:02:3C:B9:02:B5
Device 00:02:3C:B9:02:B5 (public)
	Name: Zen Air Pro
	Alias: Zen Air Pro
	Class: 0x00244404 (2376708)
	Icon: audio-headset
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (5052494d-2dab-0741-6972-6f6861424c45)
	UUID: Vendor specific           (fa349b5f-8050-0030-0010-00001bbb231d)
	ManufacturerData.Key: 0xff00 (65280)
	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
  00 ff 0f 03 02 00                                ......          
	ServiceData.00001855-0000-1000-8000-00805f9b34fb:
  2a 00                                            *.              
	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
  00                                               .               
	RSSI: 0xffffffdf (-33)
	AdvertisingFlags:
  02                                               .               
	AdvertisingData.Key: 0x2e (46)
	AdvertisingData.Value:
  d1 51 60 30 82 40                                .Q`0.@          

$ bluetoothctl info 00:02:3C:BD:97:48
Device 00:02:3C:BD:97:48 (public)
	Name: Zen Air Pro
	Alias: Zen Air Pro
	Appearance: 0x0941 (2369)
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (5052494d-2dab-0741-6972-6f6861424c45)
	ManufacturerData.Key: 0xff00 (65280)
	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
  00 ff 0f 03 02 00                                ......          
	ServiceData.00001855-0000-1000-8000-00805f9b34fb:
  2a 00                                            *.              
	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
  00                                               .               
	RSSI: 0xffffffdb (-37)
	AdvertisingFlags:
  04                                               .               
	AdvertisingData.Key: 0x2e (46)
	AdvertisingData.Value:
  ef 34 5c 7e e0 4c                                .4\~.L          

$ bluetoothctl info 00:00:00:00:00:00
Device 00:00:00:00:00:00 (public)
	Alias: 00-00-00-00-00-00
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
```

By doing a git bisect I saw that this `00:00:00:00:00:00` device started to be shown since 9391332938293b37cd29013dde65fdeeec614b5d. I'm assuming that the flag `MGMT_DEV_FOUND_NOT_CONNECTABLE` is not set, and because bluez is not using `eir_data.rsi` anymore, it now unexpectedly started to show this device.

This is the btmon debug log when using Bluez 5.79 [5.79.btsnoop.log](https://github.com/user-attachments/files/18886806/5.79.btsnoop.log), and this is the log when using Bluez 5.77 (when the device is correctly hidden) [5.77.btsnoop.log](https://github.com/user-attachments/files/18886824/5.77.btsnoop.log)

Vudentz, 2025-02-20:
> ## Steps to reproduce
> 
>     * Install bluez 5.79
> 
>     * Set the Creative Zen Air Pro in pairing mode
> 
>     * Notice how `bluetoothctl devices` shows an unexpected device with a MAC address of `00:00:00:00:00:00`
> 
>     * Trying to pair it would stall bluez until it eventually goes into timeout
> 
> 
> ## Additional info
> 
> ```
> $ bluetoothctl devices
> [...]
> Device 00:00:00:00:00:00 00-00-00-00-00-00
> Device 00:02:3C:B9:02:B5 Zen Air Pro
> Device 00:02:3C:BD:97:48 Zen Air Pro  # This entry is grayed out
> 
> $ bluetoothctl info 00:02:3C:B9:02:B5
> Device 00:02:3C:B9:02:B5 (public)
> 	Name: Zen Air Pro
> 	Alias: Zen Air Pro
> 	Class: 0x00244404 (2376708)
> 	Icon: audio-headset
> 	Paired: no
> 	Bonded: no
> 	Trusted: no
> 	Blocked: no
> 	Connected: no
> 	LegacyPairing: no
> 	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
> 	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
> 	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
> 	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
> 	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
> 	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
> 	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
> 	UUID: Vendor specific           (5052494d-2dab-0741-6972-6f6861424c45)
> 	UUID: Vendor specific           (fa349b5f-8050-0030-0010-00001bbb231d)
> 	ManufacturerData.Key: 0xff00 (65280)
> 	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
>   00 ff 0f 03 02 00                                ......          
> 	ServiceData.00001855-0000-1000-8000-00805f9b34fb:
>   2a 00                                            *.              
> 	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
>   00                                               .               
> 	RSSI: 0xffffffdf (-33)
> 	AdvertisingFlags:
>   02                                               .               
> 	AdvertisingData.Key: 0x2e (46)
> 	AdvertisingData.Value:
>   d1 51 60 30 82 40                                .Q`0.@          
> 
> $ bluetoothctl info 00:02:3C:BD:97:48
> Device 00:02:3C:BD:97:48 (public)
> 	Name: Zen Air Pro
> 	Alias: Zen Air Pro
> 	Appearance: 0x0941 (2369)
> 	Paired: no
> 	Bonded: no
> 	Trusted: no
> 	Blocked: no
> 	Connected: no
> 	LegacyPairing: no
> 	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
> 	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
> 	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
> 	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
> 	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
> 	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
> 	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
> 	UUID: Vendor specific           (5052494d-2dab-0741-6972-6f6861424c45)
> 	ManufacturerData.Key: 0xff00 (65280)
> 	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
>   00 ff 0f 03 02 00                                ......          
> 	ServiceData.00001855-0000-1000-8000-00805f9b34fb:
>   2a 00                                            *.              
> 	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
>   00                                               .               
> 	RSSI: 0xffffffdb (-37)
> 	AdvertisingFlags:
>   04                                               .               
> 	AdvertisingData.Key: 0x2e (46)
> 	AdvertisingData.Value:
>   ef 34 5c 7e e0 4c                                .4\~.L          
> 
> $ bluetoothctl info 00:00:00:00:00:00
> Device 00:00:00:00:00:00 (public)
> 	Alias: 00-00-00-00-00-00
> 	Paired: no
> 	Bonded: no
> 	Trusted: no
> 	Blocked: no
> 	Connected: no
> 	LegacyPairing: no
> ```
> 
> By doing a git bisect I saw that this `00:00:00:00:00:00` device started to be shown since [9391332](https://github.com/bluez/bluez/commit/9391332938293b37cd29013dde65fdeeec614b5d). I'm assuming that the flag `MGMT_DEV_FOUND_NOT_CONNECTABLE` is not set, and because bluez is not using `eir_data.rsi` anymore, it now unexpectedly started to list this device.
> 
> This is the btmon debug log when using Bluez 5.79 [5.79.btsnoop.log](https://github.com/user-attachments/files/18886806/5.79.btsnoop.log), and this is the log when using Bluez 5.77 (when the device is correctly hidden) [5.77.btsnoop.log](https://github.com/user-attachments/files/18886824/5.77.btsnoop.log)

Just to confirm, it is not grayed out, or it is? That said I don't think all zero address is actually valid to begin with.

RyuzakiKK, 2025-02-20:
> Just to confirm, it is not grayed out, or it is? That said I don't think all zero address is actually valid to begin with.

The all zero address is not grayed out.

Vudentz, 2025-02-20:
> > Just to confirm, it is not grayed out, or it is? That said I don't think all zero address is actually valid to begin with.
> 
> The all zero address is not grayed out.

Then it is a bug, are you planning to work having it fixed?

RyuzakiKK, 2025-02-20:
I didn’t start looking too closely at a possible fix yet. So, if you want to take this on, it would definitely be appreciated.

ValeZAA, 2025-03-28:
Yes, I can see 

Device 00:00:00:00:00:00 00-00-00-00-00-00

too, why is that not supposed to be shown?

Sometimes it starts printing (in ) [bluetooth]# [bluetooth]# [NEW] Device 00:00:00:00:00:00 00-00-00-00-00-00
 every other line for pages

ValeZAA, 2025-03-29:
I am gonna revert the commit in here and check whether I can connect to my Samsung Buds3 Pro.

Cause clearly my buds show strange all 0 address and clearly it has RSI.

RyuzakiKK, 2025-07-12:
Still an issue. I have it in my TODO list to eventually working on a fix, but higher priorities continue to push this down the queue. Sooner or later I hope to have time to work on it.

RyuzakiKK, 2025-07-31:
This has been fixed in Bluez 5.83 with the commit https://github.com/bluez/bluez/commit/af7d5dc61154e2ad226502de8e5b0f32cfe34312

closed 2025-07-31 by RyuzakiKK