bluez/bluez #1104: btmon crashes claiming buffer overflow

blaueente, 2025-02-20:
When I try to execute btmon on Ubuntu 24.10, it aborts:

```
# btmon
Bluetooth monitor ver 5.77
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```

I attached the output of
`valgrind -s --leak-check=full --show-leak-kinds=all --track-origins=yes --log-file=btmon-valgrind.txt btmon`

[btmon-valgrind.txt](https://github.com/user-attachments/files/18893490/btmon-valgrind.txt)

Vudentz, 2025-02-20:
@blaueente it is probably worth checking if this hasn't been fixed upstream already.

FGK42, 2025-03-12:
I'd like to add that I am having this issue as well, but only when using btmon -t and btmon -T:

btmon:
```
usr/bin$ sudo btmon
Bluetooth monitor ver 5.72
btmon[76415]: = Note: Linux version 6.8.0-55-generic (x86_64)          0.744308
btmon[76415]: = Note: Bluetooth subsystem version 2.22                 0.744311
= New Index: 00:0E:8E:CA:C4:AE (Primary,USB,hci0)               [hci0] 0.744312
= Open Index: 00:0E:8E:CA:C4:AE                                 [hci0] 0.744313
= Index Info: 00:0E:8E:CA:C4:AE (Qualcomm)                      [hci0] 0.744314
bluetoothd[57016]: @ MGMT Open:... (privileged) version 1.22  {0x0001} 0.744316
```

btmon -T:
``` usr/bin$ sudo btmon -T
Bluetooth monitor ver 5.72
btmon[76297]: = Note: Linux version 6.8.0-55-gen..   2025-03-12 13:28:20.330247
btmon[76297]: = Note: Bluetooth subsystem versio..   2025-03-12 13:28:20.330250
= New Index: 00:0E:8E:C.. (Primary,USB,hci0)  [hci0] 2025-03-12 13:28:20.330253
= Open Index: 00:0E:8E:CA:C4:AE               [hci0] 2025-03-12 13:28:20.330253
= Index Info: 00:0E:8E:CA:C4:AE (Qualcomm)    [hci0] 2025-03-12 13:28:20.330254
*** buffer overflow detected ***: terminated
Aborted
```

It seems that something about the bluetoothd line is causing the buffer overflow detected error.

Vudentz, 2025-03-12:
Make sure you have the following fix:

```
commit fdbfe0018f6f216ff089c0e1db848ec8ae16585e
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Feb 18 11:45:15 2025 -0500

    monitor: Fix crash due to negative max_len
    
    While calculating the max_len for printing the packet details that can
    monitor: Fix crash due to negative max_len
    
    While calculating the max_len for printing the packet details that can
    come out negative, so this discards extra text if that happens otherwise
    it can cause the following error:
    
     Process terminating with default action of signal 6 (SIGABRT)
        at 0x4A800F4: __pthread_kill_implementation (in /usr/lib64/libc.so.6)
        by 0x4A26FDD: raise (in /usr/lib64/libc.so.6)
        by 0x4A0E941: abort (in /usr/lib64/libc.so.6)
        by 0x4A0F7A6: __libc_message_impl.cold (in /usr/lib64/libc.so.6)
        by 0x4B0E3B8: __fortify_fail (in /usr/lib64/libc.so.6)
        by 0x4B0DD53: __chk_fail (in /usr/lib64/libc.so.6)
        by 0x4B0F544: __snprintf_chk (in /usr/lib64/libc.so.6)
        by 0x11F3E6: snprintf (stdio2.h:68)
```

We are working on a new release.

FGK42, 2025-03-12:
Yup, now it works. Thank you for taking the time to reply!

closed 2025-03-17 by Vudentz