bluez/bluez #1105: Invalid attribute value length coming for ATT Write Req on characteristic  [UUID - 2B29]

maheshtalewad, 2025-02-21:
Hi Luiz,

We are connected IUT with nRF app. 
After connecting nRF app,  from nRF app we trying to write 'hi' to the 'client supported feature' characterstic [UUID - 2B29]
of service - 'Generic attribute' [UUID 0x1801]. 'client supported feature' characterstic [UUID - 2B29] has Read and Write
properties.
After writing 'hi' to this characteristic [UUID - 2B29] we are getting error 'Error: Invalid Attribute Value Length (0x0d)'.

In fuc - gatt_db_attribute_write()[src\shared\gatt-db.c], below condition mentioned in the snippet passed and returned err
BT_ATT_ERROR_INVALID_ATTRIBUTE_VALUE_LEN. Here offset is 0 len = 2 and attrib->value_len =1.
could you please let us know...why this condition is put here because of this we are getting error - 'Invalid Attribute Value Length (0x0d)'.
or is it expected behavior?

code snippet:
bool gatt_db_attribute_write(struct gatt_db_attribute *attrib, uint16_t offset,
					const uint8_t *value, size_t len,
					uint8_t opcode, struct bt_att *att,
					gatt_db_attribute_write_t func,
					void *user_data)
{

 ....
 ....
 			if (offset + len > attrib->value_len) {
				err = BT_ATT_ERROR_INVALID_ATTRIBUTE_VALUE_LEN;
				goto done;
			}
		}
.....
.....
}

btmon logs:
> ACL Data RX: Handle 128 flags 0x02 dlen 9                                                                 #319 471.455677
      ATT: Write Request (0x12) len 4
        Handle: 0x000d Type: Client Supported Features (0x2b29)
          Data[2]: 6869
bluetoothd[667]: < ACL Data TX: Handle 128 flags 0x00 dlen 9                                                #320 471.456224
      ATT: Error Response (0x01) len 4
        Write Request (0x12)
        Handle: 0x000d
        Error: Invalid Attribute Value Length (0x0d)
		
bluetoothd logs:
bluetoothd[667]: src/shared/att.c:can_read_data() (chan 0xaaab06b567d0) ATT PDU received: 0x12
bluetoothd[667]: src/shared/gatt-server.c:write_cb() Write Req - handle: 0x000d
bluetoothd[667]: src/shared/gatt-server.c:write_complete_cb() Write Complete: err 13

Please find attached logs: 

[logs.zip](https://github.com/user-attachments/files/18901098/logs.zip)

Thanks and regards,
Mahesh Vithal Talewad

maheshtalewad, 2025-02-24:
Hi Luiz,

Thank you for the reply.

Asper specs we can write ATT with more than one byte. Asper below table, Attribute value can be of variable length.  So how to achieve this in Bluez, since the macro CLI_FEAT_SIZE is blocking us to write value of variable length? 

[Core Spec 6.0 : Section: 7.2 Client Supported Features in "7 Defined Generic Attribute Profile service" in Part G: Generic Attribute Profile(GATT)]

[image]

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2025-02-24:
Yeah, but there is only bits defined for the first octet:

[image]

So what is the stack expected to do with octets not defined?

maheshtalewad, 2025-02-25:
Hi Luiz,
Thank you for reply.

In specs its not mandated that those bit fields are defined for first octet.
Below point is mentioned in spec
The Client Supported Features characteristic value is an array of octets, each of which is a bit field.
The allocation of these bits is specified in Table 7.6.
 
Hence this attribute value is an array of octets, and each octet is of bit field mentioned in below table.
Please correct me if I am wrong.

[image]

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2025-02-25:
Wait, the fields defined are all in octet, so it doesn't matter if it is defined as array if only octet 0 has any field, so a client writing on octet 1 is writing to an undefined/reserved bit which, afaik, will cause the same error, so Id only agree to change the error and discard the other values if, and only if, the spec explicitly says: the server shall discard any fields undefined but accept the write procedure.

maheshtalewad, 2025-02-27:
Hi Luiz,

I understand your point, spec does not explicitly mention if we receive more than one octet.
However we checked this, on android using nRF app, it accepts more than one octet.
Considering this behavior, do you think, do we need to stick to current implementation in bluez or needs any modification?

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2025-02-27:
> Hi Luiz,
> 
> I understand your point, spec does not explicitly mention if we receive more than one octet. However we checked this, on android using nRF app, it accepts more than one octet.

When you say it accepts you are saying it doesn't cause any errors, so silently discards undefined bits? I think this behavior is actually worse since if any test is added to test invalid behavior with respect to undefined bits it would break since the stack just accepts anything.

Note that the spec already indicates that should not have zeroed trailing bytes, although it is a should not shall, and it explain the a short array shall be treated as zeros, so Id assume that is a strong indication that long arrays, without trailing zeros, are to be considered invalid.

maheshtalewad, 2025-02-28:
Hi Luiz,
 
It accepts more than one octet and displays as unknown features [when we tried to write 'hi'].
when we tried to write 01010101, it displays as follows:
	Robust Caching supported
	Unknown features (octet 1): 01
	Unknown features (octet 2): 01
	Unknown features (octet 4): 01

[image]
[image]

[Log 2025-02-24 14_26_28.txt](https://github.com/user-attachments/files/19023409/Log.2025-02-24.14_26_28.txt)
[Log 2025-02-28 11_52_45.txt](https://github.com/user-attachments/files/19023410/Log.2025-02-28.11_52_45.txt)

Thanks and regards,
Mahesh Vithal Talewad

maheshtalewad, 2025-04-28:
could you pls update?

closed 2025-09-20 by github-actions[bot]