bluez/bluez #1106: Multiple devices connect to the gatt server

CXsCode, 2025-02-24:
Hi，
  I am using bluez version 5.66 on my device, and what I am trying to achieve is as follows:
 my device acts as a peripheral (gatt server) and allows multiple cell phones to connect to my device, and each cell phone enables notifications for a characteristic of my device, and then the cell phone sends data to my device via _write_, and my device sends data to the cell phone via _notifications_.
I refer to the bluetoothctl source code to create two broadcasts and a service, and under the service I create a characteristic with a notification attribute, and then I use two cell phones to connect to my device and enable notifications on each of them. But  I found that when the first cell phone enables notifications, it can receive the notification-enabled dbus, but when the other cell phone enables notifications, bluetoothd will not send out the notification-enabled dbus, what should I do in this case?

Vudentz, 2025-02-24:
If you want to be able to determine what devices did enable the notification and them send notification per-device then you will need to able AcquireNotify:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.GattCharacteristic.rst#fd-uint16-acquirenotifydict-options-optional

With that you will get one file descriptor per device then the server application is responsible to write to device individually, which is not possible via https://github.com/bluez/bluez/blob/master/doc/org.bluez.GattCharacteristic.rst#arraybyte-value-read-only-optional since it is a property its signal PropertiesChanged cannot carry the Device information.

CXsCode, 2025-02-25:
Hi:
  Thank you very much for your reply. I am a novice in the use of bluez and dbus, so I don't quite understand what you mean by "need to able AcquireNotify". Is "AcquireNotify" a dbus message that I need to listen for, or a method that I need to call?
My understanding is that I need to listen for this message. But currently when the first phone connects, bluetoothd prints the following:

[image]

Meanwhile btmon captured the following logs:

[image]

And the "dbus-monitor --system" command I used caught the following logs:

[image]

But when the second phone enables notifications, bluetoothd prints the following:

[image]

btmon captured the following logs:

[image]

Most importantly _dbus-monitor_ does not output any prints.How do I know that the second phone is enabled for notifications and establish communication with it in this case?Am I using the _dbus-monitor_ command incorrectly?Or is there another way to get notifications that the second phone is enabled?

Vudentz, 2025-02-25:
@CXsCode it seems you version is too old and it doesn't contain the following fix: 

```
commit 8eb1dee87e019f29b6c8233dfe0f9aef8ee44461
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Mon Jul 24 13:57:34 2023 -0700

    gatt: Fix not establishing a socket for each device
    
    AcquireWrite and AcquireNotify shall establish a socket pair for each
    device connected otherwise the application cannot distinct the
    operations of each client.
    
    Fixes: https://github.com/bluez/bluez/issues/460
```

CXsCode, 2025-03-03:
This seems to be useful.Thanks for your reply.

closed 2025-03-03 by CXsCode