bluez/bluez #1110: Broadcast sink can not sync with source at the 2nd time

xihua13104, 2025-02-26:
Hi,

Broadcast sink: Bluez 5.79 + Linux kernel 6.6.18 + Pipewire 1.2.0
Broadcast source: TV box with android 14 

**Issue Description:**
We developed an application called "lea manager" for an LE Audio broadcast sink. Upon startup, lea manager automatically scans and synchronizes with nearby broadcast source, then selects the transport and plays audio. When exiting, it unselects the transport and stops playback.

The first launch after powering on the broadcast sink, the lea manager works perfectly. However, subsequent launches consistently fail to synchronize with the broadcast source unless we manually remove the broadcast source device via bluetoothctl. Is there an intentional limitation in this workflow?

[bis-sink-btsnoop.cfa.log](https://github.com/user-attachments/files/18979279/bis-sink-btsnoop.cfa.log)

[bis-sink-bluetoothd.log](https://github.com/user-attachments/files/18979282/bis-sink-bluetoothd.log)

[bis-sink-btmon.txt](https://github.com/user-attachments/files/18979284/bis-sink-btmon.txt)

michael-kong754, 2025-02-28:
Hi @xihua13104 ,
For broadcast sink, what's your startup steps, and what's your pipewire & wireplumber confs?

xihua13104, 2025-02-28:
Hi @michael-kong754 

Broadcast sink only do BLE scan when startup. And we wrote an application to auto select transports when sync with the broadcast source.

wireplumber.conf
```shell
## Template WirePlumber Bluetooth configuration

wireplumber.settings = {
}

monitor.bluez.properties = {
  ## The properties used when constructing the 'api.bluez5.enum.dbus' plugin
  ## Enabled roles (default: [ a2dp_sink a2dp_source bap_sink bap_source hfp_hf hfp_ag ])
  ## Currently some headsets (Sony WH-1000XM3) are not working with
  ## both hsp_ag and hfp_ag enabled, so by default we enable only HFP.
  ## Supported roles: hsp_hs (HSP Headset),
  ##                  hsp_ag (HSP Audio Gateway),
  ##                  hfp_hf (HFP Hands-Free),
  ##                  hfp_ag (HFP Audio Gateway)
  ##                  a2dp_sink (A2DP Audio Sink)
  ##                  a2dp_source (A2DP Audio Source)
  ##                  bap_sink (LE Audio Basic Audio Profile Sink)
  ##                  bap_source (LE Audio Basic Audio Profile Source)
  bluez5.roles = [ a2dp_sink bap_sink bap_bcast_sink ]

}

monitor.bluez-midi.properties = {
}

monitor.bluez-midi.servers = [
]

monitor.bluez.rules = [
]

monitor.bluez-midi.rules = [
]

``` 

pipewire.conf
```shell
# Daemon config file for PipeWire version "1.0.4" #
#
# Copy and edit this file in /etc/pipewire for system-wide changes
# or in ~/.config/pipewire for local changes.
#
# It is also possible to place a file with an updated section in
# /etc/pipewire/pipewire.conf.d/ for system-wide changes or in
# ~/.config/pipewire/pipewire.conf.d/ for local changes.
#

context.properties = {
    ## Configure properties in the system.
    #library.name.system                   = support/libspa-support
    #context.data-loop.library.name.system = support/libspa-support
    #support.dbus                          = true
    #link.max-buffers                      = 64
    link.max-buffers                       = 16                       # version < 3 clients can't handle more
    #mem.warn-mlock                        = false
    #mem.allow-mlock                       = true
    #mem.mlock-all                         = false
    #clock.power-of-two-quantum            = true
    #log.level                             = 2
    #cpu.zero.denormals                    = false

    core.daemon = true              # listening for socket connections
    core.name   = pipewire-0        # core name and socket name
[... 159 lines cut ...]
    { factory = spa-node-factory
        args = {
            factory.name    = support.node.driver
            node.name       = Dummy-Driver
            node.group      = pipewire.dummy
            priority.driver = 20000
            #clock.id       = monotonic # realtime | tai | monotonic-raw | boottime
            #clock.name     = "clock.system.monotonic"
        }
    }
    { factory = spa-node-factory
        args = {
            factory.name    = support.node.driver
            node.name       = Freewheel-Driver
            priority.driver = 19000
            node.group      = pipewire.freewheel
            node.freewheel  = true
        }
    }
]

context.exec = [
]

```

xihua13104, 2025-05-28:
The root cause is transport is not release when exist. Remove the BIS source manually device can fix this issue.

closed 2025-05-28 by xihua13104