bluez/bluez #1113: Possible documentation error regarding DiscoveryFilter parameter default value (duplicate data)

sebastianhaas, 2025-02-27:
Hi!

I tend to think that there is an error in https://github.com/bluez/bluez/blob/4465c577778d812702d752dfd2812e25a2f69b31/doc/org.bluez.Adapter.rst?plain=1#L128 where it says that `DuplicateData` defaults to `true`. 

However, looking at https://github.com/bluez/bluez/blob/f4617c531abe2cd263ce3b9ba7ba77dc5859215c/src/adapter.c#L2717 I believe that is not the case?

I also collected evidence that, at least on my setup, de-duplication is **on** by default, and only by explicitly passing `DuplicateData`:`true` as filter allows me to continuously pick up advertised ServiceData and ManufacturerData updates.

For reference: I am running BlueZ version 5.79, Linux 6.13.2 (arch) with what I believe is a Qualcomm WCN685x.

Vudentz, 2025-02-27:
Nevermind, I just went ahead and fixed it myself.

sebastianhaas, 2025-02-27:
Thank you so much!

closed 2025-02-27 by BluezTestBot