bluez/bluez #1117: How to implement PBP, TAMP profile in bluez

michael-kong754, 2025-03-03:
In BlueZ, the connect ops needed to be called to implement broadcast-source. I briefly sorted out the process of this function in the kernel. In this function, extended advertising (EA) is bound to periodic advertising (PA), so I cannot edit EA data in user space.
[image]
In such a scenario, how can I implement some BAP-based profiles such as PBP or TAMP? Can I only add service data in EA by modifying the kernel?

Best regards

michael-kong754, 2025-03-05:
Hi @Vudentz ,

  Could you help to look at this issue when you have time?

Best regards

michael-kong754, 2025-03-27:
Hi @Vudentz 
  Can you help to look at this issue? In my understanding, compared to BAP, profiles such as PBP and TMAP are the LE Audio profiles for final application.
Best regards

michael-kong754, 2025-03-28:
> Afaik TMAP is basically specific stream setting, so Im not sure what you are after here? If there is a service, etc, we can probably register them as well.
> 
> For PBP we need to add as part of Extended Advertising data, it shall be possible to create your own advertising instance and then point to it by setting BIS ID to the advertising instance previously created:
> 
> ```
> 	if (instance) {
> 		adv = hci_find_adv_instance(hdev, instance);
> 		/* Create an instance if that could not be found */
> 		if (!adv) {
> 			adv = hci_add_per_instance(hdev, instance, flags,
> 						   data_len, data,
> 						   sync_interval,
> 						   sync_interval);
> 			if (IS_ERR(adv))
> 				return PTR_ERR(adv);
> 			adv->pending = false;
> 			added = true;
> 		}
> 	}
> ```
> 
> Perhaps this is not clear because we haven't done anything that indicates that EA setup can be done separately, using existing advertising API, or we could do a dedicated socket option, like BT_ISO_BASE, e.g BT_ISO_PBP so it can also configure the EA via socket.
> 
> [@iulia-tanasescu](https://github.com/iulia-tanasescu) any thoughts on the above?
Hi @Vudentz ,
  Thank you for your replying, you are so helpful!
  Maybe I didn't express my meaning clearly. For other LE Audio profiles, such as PBP, TMAP and GMAP, they all have their own service UUIDs and need to set different EA. However, the current EA was done in the kernel (6.12.16) and there is no specific API in BlueZ to do this.
  Thank you very much!

Vudentz, 2025-03-28:
> Hi [@Vudentz](https://github.com/Vudentz) , Thank you for your replying, you are so helpful! Maybe I didn't express my meaning clearly. For other LE Audio profiles, such as PBP, TMAP and GMAP, they [...]

Perhaps I should give you and example, you can do the following:

```
[bluetoothctl]> advertise.service 0x1856 0x00 0x00
[bluetoothctl]> advertise on
< HCI Command: LE Set Exten.. (0x08|0x0037) plen 13  #49 [hci0] 15:57:39.654569
        Handle: 0x01
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0x09
        Service Data: Public Broadcast Announcement (0x1856)
          Data[2]: 0000
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
> HCI Event: Command Complete (0x0e) plen 4          #50 [hci0] 15:57:39.654589
      LE Set Extended Advertising Data (0x08|0x0037) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extend.. (0x08|0x0039) plen 6  #51 [hci0] 15:57:39.654632
        Extended advertising: Enabled (0x01)
        Number of sets: 1 (0x01)
        Entry 0
          Handle: 0x01
          Duration: 0 ms (0x00)
          Max ext adv events: 0
> HCI Event: Command Complete (0x0e) plen 4          #52 [hci0] 15:57:39.654667
      LE Set Extended Advertising Enable (0x08|0x0039) ncmd 1
        Status: Success (0x00)
```

Then you can hook it up with BIS by passing its Handle as BIS ID, though it seems we are missing the option to set the BIS ID manually in the endpoint.config, but I will fix that.

Anyway the general idea is to reuse the existing advertising API, so user can customize the data via D-Bus, or we can directly add it via bluetoothd or kernel, though it seems there are some pieces missing in the kernel in order to fully work with an existing advertising instance.

@iulia-tanasescu or perhaps you have something on the works already?

Vudentz, 2025-03-31:
@iulia-tanasescu we might need to do something like:

```
diff --git a/net/bluetooth/hci_sync.c b/net/bluetooth/hci_sync.c
index 609b035e5c90..af2205c1a6d7 100644
--- a/net/bluetooth/hci_sync.c
+++ b/net/bluetooth/hci_sync.c
@@ -1603,8 +1603,17 @@ int hci_start_per_adv_sync(struct hci_dev *hdev, u8 instance, u8 data_len,
 
        if (instance) {
                adv = hci_find_adv_instance(hdev, instance);
-               /* Create an instance if that could not be found */
-               if (!adv) {
+               if (adv) {
+                       if (!adv->periodic) {
+                               /* Turn it into periodic advertising */
+                               adv->periodic = true;
+                               adv->per_adv_data_len = data_len;
+                               if (data)
+                                       memcpy(adv->per_adv_data, data,
+                                              data_len);
+                       }
+               } else if (!adv) {
+                       /* Create an instance if that could not be found */
                        adv = hci_add_per_instance(hdev, instance, flags,
                                                   data_len, data,
                                                   sync_interval,
```

ValeZAA, 2025-04-01:
AFAIK, Google Pixel 9 uses TMAP profile of LE audio for phone calls in LC3.

So this is not implemented? Galaxy devices did not implement it yet, I think.

michael-kong754, 2025-04-03:
> [@iulia-tanasescu](https://github.com/iulia-tanasescu) we might need to do something like:
>
> ```
> diff --git a/net/bluetooth/hci_sync.c b/net/bluetooth/hci_sync.c
> index 609b035e5c90..af2205c1a6d7 [...]

Hi @Vudentz ,
Thank you for your suggestion, i have tested this patch, and i think the 
**adv->flags = flags;** 
should be added, or hci_set_per_adv_params_sync will return -22.

```
diff --git a/net/bluetooth/hci_sync.c b/net/bluetooth/hci_sync.c
index 609b035..64ad7fd 100644
--- a/net/bluetooth/hci_sync.c
+++ b/net/bluetooth/hci_sync.c
@@ -1603,8 +1603,17 @@ int hci_start_per_adv_sync(struct hci_dev *hdev, u8 instance, u8 data_len,
 
 	if (instance) {
 		adv = hci_find_adv_instance(hdev, instance);
-		/* Create an instance if that could not be found */
-		if (!adv) {
+		if (adv) {
+			if (!adv->periodic) {
+				/* Turn it into periodic advertising */
+				adv->periodic = true;
+				adv->per_adv_data_len = data_len;
+				if (data)
+					memcpy(adv->per_adv_data, data, data_len);
+				adv->flags = flags;
+			}
+		} else if (!adv) {
+			/* Create an instance if that could not be found */
 			adv = hci_add_per_instance(hdev, instance, flags,
 						   data_len, data,
 						   sync_interval,
```

Vudentz, 2025-06-10:
> Hi [@Vudentz](https://github.com/Vudentz) , For PBP test steps in **https://github.com/bluez/bluez/blob/master/client/scripts/broadcast-source-pbp.bt** , it can run successfully use the latest kernel. But for original broadcast-source in **https://github.com/bluez/bluez/blob/master/client/scripts/broadcast-source.bt** , it will cause the kernel panic.
> 
> I have found the root cause. It was because that in hci_adv_bcast_annoucement the size of array ad was set to HCI_MAX_EXT_AD_LENGTH, and since this array hasn't set the EIR flags, so in eir_create_adv_data add this flags which cause the final size of adv_data_len exceeds the HCI_MAX_EXT_AD_LENGTH.
> 
> So i think we should change the array size of ad in hci_adv_bcast_annoucement .
> 
> Best regards

Weird I don't seem to be able to reproduce it, and there are existing tests such as:

tools/iso-tester -s "ISO Broadcaster - Success"
```
...
< HCI Command: LE Set Ext.. (0x08|0x0037) plen 255
        Handle: 0x01
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0xfb
        Service Data: Broadcast Audio Announcement (0x1852)
        Broadcast ID: 15601328 (0xee0eb0)

```

That said the length is indeed wrong and perhaps there is some discoverable flag that the above flag don't set as broadcast role normally shall not contain flags, are you perhaps testing with an old bluetoothd?

ValeZAA, 2025-06-26:
I bought Sony XM6 so will be ready to test this tomorrow. As well as other issues